mavlink/mavros #1482: Get no gps fix warning using RTK GPS

mzwtju, 2020-09-11:
### Issue details

I’m using [CUAV RT](https://docs.px4.io/v1.9.0/en/gps_compass/rtk_gps_cuav_c-rtk.html)K for more accurate global positioning.
I subscribe to the /mavros/global_position/global topic to calculate ralative distance between drones. However, I keep getting warning GP: no gps fix .

But when I plug out base module  from my PC then using M8N as GPS, the terminal stop sending warning info.

In the same time in the same time distance between drones calculated by global position is far from accurate. I think it’s because No GPS fix warning.

Could you please tell me which part should I check? 

Thank you in advance!
### MAVROS version and platform

Mavros: ?1.2.0?
ROS: ?Kinetic?
Ubuntu mate 16.04

### Autopilot type and version

PX4 

Version: 1.10.1

vooon, 2020-09-11:
Need to check gps and global position plugins, as i remember when RTK is on fix type > 3, need to check that conditions.

mzwtju, 2020-09-13:
@vooon  thanks for your advice
I'm not sure what do you mean by check gps and gp plugins, but I do make sure that they are loaded and initialized.

I check [GPS fix type ](https://mavlink.io/en/messages/common.html#GPS_FIX_TYPE),according to [fly logs](https://review.px4.io/plot_app?log=86f8ed15-c888-4bbe-a4df-1d668b0a3782), GPS fix value was 6 most of time, which means RTK FIXED. So I'm pretty sure RTK GPS is fixed. But the bad positioning performance and No GPS Fix warning  make me confused.  
[image] 

Could you please help  with the warning issue?
Thank you again!

mzwtju, 2020-09-13:
BTW I noticed in https://docs.px4.io/master/en/gps_compass/rtk_gps.html#rtk-gps 
_`QGroundControl can theoretically enable RTK GPS for multiple vehicles (provided they each have a rover module). At time of writing this use case has not been tested.`_
I use one base module and multiple rover modules. I don't know if this has been tested yet.

closed 2020-09-18 by mzwtju