mavlink/mavros #1483: catkin build error << mavros:make /home/huanyu/catkin_ws/logs/mavros/build.make.001.log

huanyu-zhao, 2020-09-11:
This is only bug and feature tracker, please use it
to report bugs or request features.

----
I just installed mavlink/mavros. After i finished the steps 1-5 according the official tutorial, I catkin build, then it occured an error.
### Issue details

_Please describe the problem, or desired feature_
In file included from /home/huanyu/catkin_ws/src/mavros/mavros/src/lib/enum_to_string.cpp:19:0:
/home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:70:40: error: ‘std::__cxx11::string mavros::utils::to_string’ redeclared as different kind of symbol
 std::string to_string(mavlink::common::MAV_AUTOPILOT e);
                                        ^
/home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:68:13: note: previous declaration ‘std::__cxx11::string mavros::utils::to_string(mavlink::common::MAV_SENSOR_ORIENTATION)’
 std::string to_string(mavlink::common::MAV_SENSOR_ORIENTATION e);
             ^
/home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:70:23: error: ‘MAV_AUTOPILOT’ is not a member of ‘mavlink::common’
 std::string to_string(mavlink::common::MAV_AUTOPILOT e);
                       ^
/home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:70:23: note: suggested alternative:
In file included from /home/huanyu/catkin_ws/devel/include/mavlink/v2.0/common/common.hpp:2254:0,
                 from /home/huanyu/catkin_ws/src/mavros/libmavconn/include/mavconn/mavlink_dialect.h:26,
                 from /home/huanyu/catkin_ws/src/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.h:18,
                 from /home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:22,
                 from /home/huanyu/catkin_ws/src/mavros/mavros/src/lib/enum_to_string.cpp:19:
/home/huanyu/catkin_ws/devel/include/mavlink/v2.0/common/../minimal/minimal.hpp:34:12: note:   ‘mavlink::minimal::MAV_AUTOPILOT’
 enum class MAV_AUTOPILOT : uint8_t
            ^
In file included from /home/huanyu/catkin_ws/src/mavros/mavros/src/lib/enum_to_string.cpp:19:0:
/home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:71:40: error: ‘std::__cxx11::string mavros::utils::to_string’ redeclared as different kind of symbol
 std::string to_string(mavlink::common::MAV_TYPE e);
                                        ^
/home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:68:13: note: previous declaration ‘std::__cxx11::string mavros::utils::to_string(mavlink::common::MAV_SENSOR_ORIENTATION)’
 std::string to_string(mavlink::common::MAV_SENSOR_ORIENTATION e);
             ^
/home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:71:23: error: ‘MAV_TYPE’ is not a member of ‘mavlink::common’
 std::string to_string(mavlink::common::MAV_TYPE e);
                       ^
/home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:71:23: note: suggested alternative:
In file included from /home/huanyu/catkin_ws/devel/include/mavlink/v2.0/common/common.hpp:2254:0,
                 from /home/huanyu/catkin_ws/src/mavros/libmavconn/include/mavconn/mavlink_dialect.h:26,
                 from /home/huanyu/catkin_ws/src/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.h:18,
                 from /home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:22,
                 from /home/huanyu/catkin_ws/src/mavros/mavros/src/lib/enum_to_string.cpp:19:
/home/huanyu/catkin_ws/devel/include/mavlink/v2.0/common/../minimal/minimal.hpp:62:12: note:   ‘mavlink::minimal::MAV_TYPE’
 enum class MAV_TYPE : uint8_t
            ^
In file included from /home/huanyu/catkin_ws/src/mavros/mavros/src/lib/enum_to_string.cpp:19:0:
/home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:72:40: error: ‘std::__cxx11::string mavros::utils::to_string’ redeclared as different kind of symbol
 std::string to_string(mavlink::common::MAV_STATE e);
                                        ^
/home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:68:13: note: previous declaration ‘std::__cxx11::string mavros::utils::to_string(mavlink::common::MAV_SENSOR_ORIENTATION)’
 std::string to_string(mavlink::common::MAV_SENSOR_ORIENTATION e);
             ^
/home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:72:23: error: ‘MAV_STATE’ is not a member of ‘mavlink::common’
 std::string to_string(mavlink::common::MAV_STATE e);
                       ^
/home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:72:23: note: suggested alternative:
In file included from /home/huanyu/catkin_ws/devel/include/mavlink/v2.0/common/common.hpp:2254:0,
                 from /home/huanyu/catkin_ws/src/mavros/libmavconn/include/mavconn/mavlink_dialect.h:26,
                 from /home/huanyu/catkin_ws/src/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.h:18,
                 from /home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:22,
                 from /home/huanyu/catkin_ws/src/mavros/mavros/src/lib/enum_to_string.cpp:19:
/home/huanyu/catkin_ws/devel/include/mavlink/v2.0/common/../minimal/minimal.hpp:137:12: note:   ‘mavlink::minimal::MAV_STATE’
 enum class MAV_STATE : uint8_t
            ^
In file included from /home/huanyu/catkin_ws/src/mavros/mavros/src/lib/enum_to_string.cpp:19:0:
/home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:73:40: error: ‘std::__cxx11::string mavros::utils::to_string’ redeclared as different kind of symbol
 std::string to_string(mavlink::common::MAV_COMPONENT e);
                                        ^
/home/huanyu/catkin_ws/src/mavros/mavros/include/mavros/utils.h:68:13: note: previous declaration ‘std::__cxx11::string mavros::utils::to_string(mavlink::common::MAV_SENSOR_ORIENTATION)’
[... 502 lines cut ...]
                                          ^
make[2]: *** [CMakeFiles/mavros.dir/src/lib/uas_data.cpp.o] Error 1
make[2]: *** [CMakeFiles/mavros.dir/src/lib/mavros.cpp.o] Error 1
make[1]: *** [CMakeFiles/mavros.dir/all] Error 2
make: *** [all] Error 2
cd /home/huanyu/catkin_ws/build/mavros; catkin build --get-env mavros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................................................................................................................
Failed     << mavros:make                  [ Exited with code 2 ]                                                                                                              
Failed    <<< mavros                       [ 15.7 seconds ]                                                                                                                    
Abandoned <<< mavros_extras                [ Unrelated job failed ]                                                                                                            
Abandoned <<< test_mavros                  [ Unrelated job failed ]                                                                                                            
[build] Summary: 3 of 6 packages succeeded.                                                                                                                                    
[build]   Ignored:   None.                                                                                                                                                     
[build]   Warnings:  None.                                                                                                                                                     
[build]   Abandoned: 2 packages were abandoned.                                                                                                                                
[build]   Failed:    1 packages failed.                                                                                                                                        
[build] Runtime: 19.6 seconds total.                          

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

vooon, 2020-09-11:
Duplicates: https://github.com/mavlink/mavlink/issues/1474
Need to change namespaces. Mavlink become incompatible.

closed 2020-09-11 by vooon

Coderx7, 2021-05-25:
I'm facing the very same error. but I believe I'm using the latest version. should I be using 1.4 exclusively for kinetic ?

vooon, 2021-05-25:
@Coderx7 1.6 should work for Kinetic.

commit 2702c04d (2020-09-11): mavros: use mavlink::minimal:: after incompatible changes in mavlink package

Incompatible change: https://github.com/mavlink/mavlink/pull/1463

Fix: #1483, https://github.com/mavlink/mavlink/issues/1474