mavlink/mavros #1500: The doc of mavros_msgs about coordinate needs better clarification.

petertheprocess, 2020-10-18:
http://docs.ros.org/en/api/mavros_msgs/html/msg/PositionTarget.html
https://github.com/mavlink/mavros/issues/1084#issuecomment-428177363
Just like the @FaboNo said
> the FCU is working with the NED frame and ROS is working with the ENU frame.
Mavros takes care of the transforms between the two frames.
So if you want to send a setpoint to the FCU, it must be defined in ENU frame (X forward and Z up), Mavros will do the transform in NED coordinates.
Conversely when you want to read the current pose of the quadrotor, you can use mavros/local_position/pose, mavros will do the transformation so that you will read the pose in ENU as well 

While in the documentation in http://docs.ros.org/en/api/mavros_msgs/html/msg/PositionTarget.html, FRAME_LOCAL_NED and FRAME_BODY_NED are still used. It can misguide many beginners of mavros like me.
So I suggest to rename these FRAM CONSTANT like this:
| old|new |
|-|-|
|FRAME_LOCAL_NED|FRAME_LOCAL_ENU|
|FRAME_LOCAL_OFFSET_NED|FRAME_LOCAL_OFFSET_ENU|
|FRAME_BODY_NED|FRAME_BODY_FLU|
|FRAME_BODY_OFFSET_NED|FRAME_BODY_OFFSET_FLU|
https://github.com/mavlink/mavros/blob/c9adcfbe5ae76b0f1f05c6745c038c1cd79afed9/mavros_msgs/msg/PositionTarget.msg#L8-L12

amilcarlucas, 2020-10-29:
Good point, can you please provide a github pull-request?

shupx, 2021-03-12:
i don't think it is a good idea to revise the frame constants in 'PositionTarget.msg'. Because these frame constants are also used in some mavros plugin cpp files such as[ 'setpoint_raw.cpp'](https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_raw.cpp), which means you should revise all the mavros plugins that involves the  'PositionTarget.msg' in the meantime.  so it is not necessary. 
For beginners, this is confusing. However the[ README.md file in mavros](https://github.com/mavlink/mavros/blob/master/mavros/README.md) has explained this very carefully.

vooon, 2021-03-12:
I do not think that it's good change because it's copy of [MAV_FRAME](https://mavlink.io/en/messages/common.html#MAV_FRAME) (old one...).
And also mavros is beyond 1.0.0, so i would rather keep API stable. It's ok to add new items, add comments, that won't make it incompatible with existing user's code.

ViniciusAbrao, 2022-09-10:
It took me some hours until I find this post and to see that the README explain that.

closed 2026-11-05 by github-actions[bot]