mavlink/mavros #1503: [libmavconn] tcp client/server illegal thread join from io_thread itself

guilhermelawless, 2020-10-28:
This is an issue when the server closes with an established connection, it causes a terminate on the client.

`io_thread` runs the `io_service` which runs things like `do_recv`, `do_send` (for the TCPClient, but the Server has a similar issue).

In these async methods, any failure (including EOF i.e. the server going offline) calls `close()` on the client, which joins the `io_thread`. That means the `io_thread` is getting joined from the thread itself, which throws (see error conditions in https://en.cppreference.com/w/cpp/thread/thread/join)

Moving the `join()` to the destructors of Client/Server seems to solve the issue.

This then allows the main program to get the `port_closed_cb` and do a reconnection routine, without terminating.

amilcarlucas, 2020-10-29:
Please create a github pull-request

guilhermelawless, 2020-11-03:
I'm unable to properly test at the moment so would leave that to you, but this is the patch from master

```diff
diff --git a/libmavconn/src/tcp.cpp b/libmavconn/src/tcp.cpp
index 609a9bd..3e88ebd 100644
--- a/libmavconn/src/tcp.cpp
+++ b/libmavconn/src/tcp.cpp
@@ -134,6 +134,8 @@ MAVConnTCPClient::~MAVConnTCPClient()
 {
        is_destroying = true;
        close();
+       if (io_thread.joinable())
+               io_thread.join();
 }
 
 void MAVConnTCPClient::close()
@@ -148,10 +150,6 @@ void MAVConnTCPClient::close()
 
        io_work.reset();
        io_service.stop();
-
-       if (io_thread.joinable())
-               io_thread.join();
-
        io_service.reset();
 
        if (port_closed_cb)
@@ -320,6 +318,8 @@ MAVConnTCPServer::~MAVConnTCPServer()
 {
        is_destroying = true;
        close();
+       if (io_thread.joinable())
+               io_thread.join();
 }
 
 void MAVConnTCPServer::close()
@@ -334,9 +334,6 @@ void MAVConnTCPServer::close()
        io_service.stop();
        acceptor.close();
 
-       if (io_thread.joinable())
-               io_thread.join();
-
        if (port_closed_cb)
                port_closed_cb();
 }
```

closed 2026-11-05 by github-actions[bot]