mavlink/mavros #1505: Wrong movement using ArduRover and ROS navigation stack

sammar80, 2020-11-03:
### Issue details

I am using ArduRover with ROS navigation stack and although I can see a correct trajectory in rviz map once I send the goal to move base but the rover is not moving in the correct direction I checked and the conversion from ENU to NED is done by the setpoint_velocity.cpp plugin in mavros 

### MAVROS version and platform

Mavros: master
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[ X ] ArduPilot
[  ] PX4

Version: ArduRover V4.0.0 (0e52bafa)

### Node logs

roslaunch mavros apm.launch fcu_url:=/dev/ttyACM0:921600
... logging to /home/optimus/.ros/log/802f23ec-1d9a-11eb-87fe-d03745187cad/roslaunch-Autobot-32312.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Autobot:39447/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/lidar0/id: 0
 * /mavros/distance_sensor/lidar0/orientation: PITCH_270
 * /mavros/distance_sensor/lidar0/subscriber: False
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_id: 4
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/horiz_accuracy: 0.5
 * /mavros/fake_gps/mocap_transform: False
 * /mavros/fake_gps/mocap_withcovariance: False
 * /mavros/fake_gps/satellites_visible: 6
 * /mavros/fake_gps/speed_accuracy: 0.0
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_hil_gps: True
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fake_gps/vert_accuracy: 0.5
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyACM0:921600
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
[... 397 lines cut ...]
        value: "-1.0"
  - 
    level: 1
    name: "mavros: Time Sync"
    message: "Frequency too high."
    hardware_id: "/dev/ttyACM0:921600"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "359"
      - 
        key: "Frequency (Hz)"
        value: "18.141095"
      - 
        key: "Last RTT (ms)"
        value: "607.003685"
      - 
        key: "Mean RTT (ms)"
        value: "4469004945.422476"
      - 
        key: "Last remote time (s)"
        value: "1604391653.295881033"
      - 
        key: "Estimated time offset (s)"
        value: "1604372388.825948715"

Alex-Beh, 2021-01-19:
Hello, may I know the board that you using?

sammar80, 2021-01-19:
@Alex-Beh for the companion computer we are using NVIDIA Jetson nano integeated with pixracer for ardupilot

chiragmakwana0296, 2021-02-13:
Thats Probably due to setpoint_velocity mav_frame is set to LOCAL_NED 
Set it to BODY_NED in apm_config.yaml or change it using:
```bash
rosservice call /mavros/setpoint_velocity/mav_frame "mav_frame: 8"
```

sammar80, 2021-05-23:
@chiragmakwana0296  thanks a lot for your reply and sorry for delay . I changed the mav_frame to BODY_NED and the problem was solved

closed 2026-11-05 by github-actions[bot]