mavlink/mavros #1506: Running on Docker - 'Operation not permitted'

psj900918, 2020-11-05:
### Issue details

Hi, i'm trying to run MAVROS node at Docker container.
In the container running bare-bone ubuntu 18.04 image, i installed

 - ROS Melodic ('desktop' package)
 - mavros* packages
 - install_geographic_.sh 
 - usermod -aG dialout $USER

as a root.

I attached a externally-powered Pixhawk, which appears to ```/dev/ttyACM0``` at the host.
Normally this setup worked, as i run as
``` roslaunch mavros px4.launch fcu_url:=/dev/ttyACM0:921600 ```
at the host.

However, this dev does not appear at the container, so i added option
``` -v /dev/ttyACM0:/dev/ttyACM0 ```
at creating container.

Then, this fatal log comes
```[FATAL] []: FCU: DeviceError:serial:open: Operation not permitted ```

I read [this](https://dev.px4.io/v1.9.0/en/test_and_ci/docker.html), although not exactly the same what i trying to,
but added this option
``` -e LOCAL_USER_ID="($id -u)" ```
but same error occurs.

Does anyone have faced this case?

### MAVROS version and platform

Mavros: latest
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[    ] ArduPilot
[ v ] PX4

Version: latest

vooon, 2020-11-06:
I suggest to run something like mavlink-router or ser2net on the host and use network connections.

mhallak, 2020-11-11:
May be you should:
sudo usermod -a -G docker $USER  in your host (and LOGOUT/LOGIN!!!)
I am curious to know if you solved this problem.

closed 2026-11-05 by github-actions[bot]