mavlink/mavros #1510: Request failed. check MAVROS logs

abhiramdapke, 2020-11-24:
### Issue details
I have a pixhawk 2.1 and Tx2i with a carrier board and have connected them with the TX,RX, GND pins and I know that the pins are correct. When running the roslaunch mavros apm.launch fcu_url:=/dev/ttyACM0:57600
command, MAVROS starts. I am not sure if the /dev/ttyACMO:57600 is correct but it starts the MAVROS. After that when I try to run any MAVROS commands like rosrun mavros mavsafety arm, I get this error - request failed. check MAVROS logs. i tried running rostopic echo -n1 /diagnostics | tee /tmp/diag.yml, and the message says FCU connection not connected. I don't understand why the FCU conection is not connected. Is it because I don't have a GPS module, RC and receiver plugged to the pixhawk? or is it the pins issue because I am sure the pins(between Tx2i and pixhawk) are correct and the pixhawk is powered by a battery properly. Also, how do I connect to the mission planner on the localhost? Can it be done without USB?

I am not sure what is going wrong here. If I use any other roslaunch command like roslaunch mavros apm.launch fcu_url:=/dev/ttyACM0:14855, I get this error -
[FATAL] [1606235473.482246655]: FCU: DeviceError:serial:set_option: Invalid argument.

So, how to get the FCU connection established? what is missing?
How to run Mission planner on the localhost without USB? Is it required to run with MAVROS to check the flight route?

**This is the log when MAVROS doesn't start i.e. roslaunch command doesn't run:**

```
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://{TX2i name}-desktop:35237/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_id: 4
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/horiz_accuracy: 0.5
 * /mavros/fake_gps/mocap_transform: False
 * /mavros/fake_gps/mocap_withcovariance: False
 * /mavros/fake_gps/satellites_visible: 6
 * /mavros/fake_gps/speed_accuracy: 0.0
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_hil_gps: True
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fake_gps/vert_accuracy: 0.5
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyACM0:14855
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
[... 535 lines cut ...]
    level: 2
    name: "mavros: Time Sync"
    message: "No events recorded."
    hardware_id: "/dev/ttyACM0:57600"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "0.000000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"
---
```

wrmichaelmak, 2020-11-25:
/dev/ttyACM0:57600 refers to USB connection, aka the Pixhawk will communicate to your TX2i via USB at a baud rate of 57600. However, based on your current setup you are trying to communicate to the Pixhawk via UART which requires /dev/ttyS0 or whatever your UART port is

abhiramdapke, 2020-11-25:
Thanks for the quick response. So, if I connect the pixhawk to the localhost via USB and on the other end, pixhawk is connected to the TX2i, do I have to use /dev/ttyACM0:57600?
Also, how is MAVROS starting if the fcu connection is not proper?

wrmichaelmak, 2020-11-25:
> Thanks for the quick response. So, if I connect the pixhawk to the localhost via USB and on the other end, pixhawk is connected to the TX2i, do I have to use /dev/ttyACM0:57600?
> Also, how is MAVROS starting if the fcu connection is not proper?

For the first question, i do not understand why do you want to connect two computers to the Pixhawk and to answer if it will work, it depends on where you are running mavros and whether the usb port of your localhost (PC) is actually defined as ACM0. Additionally, data will not be shared between two computers. For the second question, if you notice your log you are not getting any heartbeat communication from Pixhawk. Yes mavros did run because essentially the launch command you used is a script but you can see it is not connected to the Pixhawk.

abhiramdapke, 2020-11-29:
Let me clear the air a bit. I am running MAVROS commands on the Tx2i. It is connected to a monitor. Whenever I use the localhost, I use an SSH connection to the Tx2i and run the commands on the Tx2i only.
Regarding the USB port on the computer, I will figure out what the name of the port is. I wanted to ask how do I check the baud rate which in this case is 57600 as I am not sure whether that is correct? Also, just wanted to confirm that I don't need a connection between the localhost and the pixhawk right? as all the commands are going to be sent to the Tx2i and tx2i will in turn send them to pixhawk.
One more question I had was if I need all the components like GPS, Rc, receiver connected to the pixhawk while making this connection to the Tx2i, or will it work if I have the pixhawk without these components just for testing? and will it recognize the MAVROS commands?

Thanks.

wrmichaelmak, 2020-11-30:
So basically in that setup you described above, what you should do is, run mavros with fcu_url as the serial port of the TX2i instead of usb port from localhost to pixhawk (assuming you are powering it via that although the serial port of the TX2i should give it power too). Yes, you do not need a connection between Pixhawk and localhost cause mavros is running on TX2i and commands are given by TX2i to Pixhawk (the localhost is merely just a client to your TX2i to show you what is going on wiht your TX2i). The baud rate is the baud rate that is set for Pixhawk, you can check it on Mission Planner (since you are using APM). To merely connect to Pixhawk via MAVROS, you do not need the peripherals. However, for flight you will need them. In SITL (cause APM does not support HITL), you do not need them to run your code.

abhiramdapke, 2020-12-03:
Thanks a lot. That helped me a lot. The fcu connection is done now and the tx2i is communicating with the pixhawk. I ran the commands on https://ardupilot.org/dev/docs/ros-connecting.html and the fcu is responding to those, however, I am not sure what are the next steps. I read this page(http://wiki.ros.org/mavros#Utility_commands) completely but it doesn't give a good description of how to use the mavlink commands. Now, my aim is to run them onto the tx2i(connected to the pixhawk of the actual drone) but could you guide me on the safety measures I should take before doing that and how do I run the commands and check if they are working properly? The link also doesn't mention HOW TO USE these commands.
Also, how do I run these commands on the Arducopter (for sitl/gazebo)? because when I try to simulate it on gazebo, it opens the MAVProxy terminal which would mean I need to type the Mavproxy commands but I want to write these commands.

wrmichaelmak, 2020-12-03:
@abhiramdapke I think it is best if you go through ROS tutorials (do this first) and also look at PX4 offboard mode example to see how to actually send take-off arming commands and etc from TX2i to Pixhawk.

closed 2026-11-05 by github-actions[bot]