mavlink/mavros #1511: Not able to simulate a custom drone with MAVROS and ardupilot

abhiramdapke, 2020-11-24:
### Issue details

Is it possible to simulate a custom drone by using MAVROS and Ardupilot in Gazebo that uses an onboard computer? I found huge documentation on PX4 on MAVROS but not on ardupilot. Using an onboard computer makes it tedious. I didn't find any documentation on ardupilot that allows uploading the parameters of a custom drone to connect it? Is it only possible to simulate it by connecting the pixhawk to the localhost or is there any other way?

I read the whole documentation of ardupilot and Gazebo with SITL but that doesn't include simulating with a companion computer and how to connect a custom drone that uses a pixhawk and ardupilot but is not one in the documentation.

### MAVROS version and platform

Mavros: latest
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

ArduPilot

Version: 4.0.3

linuxsocke, 2020-11-25:
So you want to use an companion computer (which is offboard) with sitl instead of the pixhawk (onboard)? I think the recipe is then using sitl with the [ardupilot_gazebo plugin](https://github.com/khancyr/ardupilot_gazebo). 

There you launch: 

- a sitl simulation 
- an instance of the drone in gazebo (with the plugin loaded in the robot description, tutorial [here](https://github.com/Intelligent-Quads))
- mavros in case you use ros (this provides the mavros topics)
- the you can start offboard code to send commands via the mavros topics

wrmichaelmak, 2020-11-25:
I think it also depends on what the OP wants to do with his drone. If he wants to implement computer vision, then it will be a different setup.

linuxsocke, 2020-11-26:
Ok how can it look like for computer vision tasks? Would be nice to know an example for PX4.

wrmichaelmak, 2020-11-26:
https://discuss.px4.io/t/how-to-add-a-ros-camera-to-iris-for-gazebo-simulation/5118

I will imagine it to be something like this. I am working with APM currently so yea, but am using AirSim instead of Gazebo.

abhiramdapke, 2020-11-29:
Thank you so much for the quick reply. 
@linuxsocke Yes, I am using an offboard computer to send the commands, and as the custom drone has pixhawk/cube in it, the flight controller will execute them on the drone. I am able to perform the sitl simulation as well as gazebo simulation with plane/rover/copter but I am not sure how to do the same with the custom drone i.e. to simulate the drone like in a real environment and send commands in the simulation to test whether it is working properly. Is it possible to do that?

Also, for the simulation on the localhost, do I need to make hardware connections always with the companion computer and pixhawk and both of them connected to the localhost, or is there a way to simulate without doing the hardware connections?

Thanks.

abhiramdapke, 2020-11-29:
Thank you for the quick reply.
@wrmichaelmak Initially, I would like to fly the drone and test it, and later on, maybe I can write scripts for Path planning and CV. I am working with Ardupilot and MAVROS. How different would the setup be for CV? Could you please shed some light on this? I will try to run the MAVROS commands on AirSim too. I am wondering what difference would it make if I use gazebo. Is it possible to simulate a custom drone in AirSim?
Thanks.

wrmichaelmak, 2020-11-30:
> Thank you for the quick reply.
> @wrmichaelmak Initially, I would like to fly the drone and test it, and later on, maybe I can write scripts for Path planning and CV. I am working with Ardupilot and [...]

To be really honest, I am still working with the CV setup. To answer some of your questions, gazebo is much more simplistic as compared to AirSim, but AirSim allows you to have more much accurate environments. To ease your setup, I will recommend PX4 instead maybe. Your definition of custom drone is quite loose. What do you mean by custom drone? Do you have a frame that is slightly different from APM's standard configuration or ...?

linuxsocke, 2020-11-30:
@abhiramdapke You will see the connection between sitl and gazebo in the tutorials. 

In short on the one hand you have the ardupilot gazebo plugin which is called in the sdf file of the model. On the other hand you start mavproxy (sitl) with extra gazebo arguments. 

You can also easily add a camera plugin (from gazebo) in the sdf file so you can get a camera stream out of your simulation. I recommend doing some gazebo tutorials to see how this works.

abhiramdapke, 2020-12-03:
@wrmichaelmak do you have any links that you are referring to for the CV setup? It would be great if you could share those. By custom drone, I mean that I am not sure how to upload the sdf file of the drone into the gazebo. The simulations performed were on Arducopter.

abhiramdapke, 2020-12-03:
@linuxsocke I did simulations with the ArduCopter on sitl, gazebo with MAVProxy commands but not sure how to upload the sdf model of my drone into gazebo. Can you tell me the procedure to do that and share any links that might be useful? Thanks. Also, while using MAVROS commands on the companion computer during simulation, we do not need to open the MAVproxy terminal, is that correct?

wrmichaelmak, 2020-12-03:
@abhiramdapke i think as of now you should focus on getting like a generic drone working with gazebo I think that what @linuxsocke posted on earlier is good. For your convenience, look at this link: https://github.com/Intelligent-Quads/iq_sim. Have a look at their camera drone files and study it.

abhiramdapke, 2020-12-03:
Thanks. I'll study that.

abhiramdapke, 2020-12-13:
@wrmichaelmak @linuxsocke 
I have an important question to ask. I did all the tutorials from the IQ guys and have setup the environment and ran a simulation of arducopter on that, too. It is working properly now. 
Now, according to the tutorials, if I run these commands in separate terminals:
1. roslaunch iq_sim runway.launch
2. ./startsitl.h
3. roslaunch iq_sim apm.launch
4. rosrun iq_gnc square

It is running the square.cpp file on the simulated drone when I switch to mode guided in MAVProxy terminal (which is of arducopter).
So, what I want to do is, I want to send these commands to a real drone that has an offboard computer and uses ardupilot and a pixhawk.
I understand that I need the fcu url and baud rate set for that (while writing the roslaunch mavros/iq_sim apm.launch command) which I already have but the question is how do I type the command ./startsitl.sh because that starts the arducopter mavproxy terminal.
My aim is to have the Mavproxy terminal so that I can type the command ''mode guided'' and then pass the script to it but ./startsitl.sh starts the Mavproxy terminal for the arducopter and not the real drone. So how can I solve this problem when trying to fly a real drone?

Also, I don't have a .sdf file of the drone with me.

It would be really helpful if you could provide some solution to this. Thanks a ton.

wrmichaelmak, 2020-12-13:
So basically if I understand you correctly you want to take the square.cpp file and use it on your drone? Or in simpler turns to do what exactly you are doing on simulation but in real life?

abhiramdapke, 2020-12-13:
Yes, that is exactly what I want to do.

wrmichaelmak, 2020-12-13:
> Yes, that is exactly what I want to do.

To answer your original question, I have a suspicion that when you run `./startsitl.sh` you are running a SITL... not an actual run on your drone. Hence, the MAVProxy is connected to SITL.

Why don't you use ROS to set your Pixhawk to `GUIDED` mode (totally recommend this less hassle)? Another way I will suggest is to use a RC to switch Pixhawk to `GUIDED` mode. 

If you really want to use MAVProxy to actually switch it to `GUDIED`, I think you will need to adjust the .sh script to allow real hardware connection from computer to MAVProxy. https://ardupilot.org/mavproxy/docs/getting_started/starting.html

abhiramdapke, 2020-12-13:
@wrmichaelmak Yes, that's true. When I run ./startsitl.sh, it is running sitl from the arducopter folder where the model is arducopter and not the real drone model yet running this command helps to type the MAVproxy commands like mode guided, etc in the same terminal. Is there any way to do that when working with the actual drone. I am thinking since I don't have the SDF file of the drone, it would not be possible to use MAVProxy to switch to guided mode. Let me know your views on this.

Could you tell me how to use ROS to switch it to the guided mode? As far as my understanding goes and according to the tutorials, it is written in the square.cpp file to wait until the mode is guided and then run the scripts.

That is interesting. I'll try it doing with the RC. That should work.

Also, if not MAVPorxy, is it possible to upload the square.cpp via QGC or Mission Planner on the real drone? 
So, in separate terminals, I'll need to type only these commands:
1. roslaunch mavros apm.launch (with the right fcu url, of course)
2. Connect the Mission Planner to the pixhawk via udp and upload the scripts and run them from there. (in the GUI)
That would mean there will be no simulation during the actual flight which I am fine with. Let me know what you think about this and if this can be done? or anything else that needs to be done.

Thank you for the link. I'll try to do that as well.

wrmichaelmak, 2020-12-13:
@abhiramdapke Having a sdf file merely helps in simulation to obtain a more accurate behaviour and representation in SIMULATION not in real life. Since MAVProxy is connected to SITL obviously it will not be able to send actual commands to the real Pixhawk. 

If you look at PX4 example it will tell you how to actually setup your drone for flight and for SITL too. Although you must take care that you are running APM not PX4 but in terms of ROS most of the code are recyclable. 

For your last question, afaik I don't think Mission Planner or QGroundControl allow you to upload and run code on a computer. Besides, I don't see much reasoning behind doing so. If you want to run the code immediately you can easily just write a bash script for it? And I don't quite understand by "no simulation during actual flight"

abhiramdapke, 2020-12-13:
Oh. Okay okay. 

Could you please share the link to that page? I am not quite familiar with the PX4 Dev docs.

Oh. Could you also please send the important links as to how to write the bash script? Thanks.
By no simulation, I mean no gazebo simulation during the actual flight of the real drone because the simulation that I am doing was for the arducopter model.

wrmichaelmak, 2020-12-13:
@abhiramdapke

PX4: https://dev.px4.io/v1.9.0/en/ros/mavros_offboard.html

For bash scripts there are multiple links online which you can easily google for. 

I still quite understand the last point. Why would you want to run a gazebo simulation of the drone flying while during an actual flight?

abhiramdapke, 2020-12-14:
Thanks a lot. I'll check that out.

Yes, you're right. I don't need a simulation during the flight. I was kind of trying something.

abhiramdapke, 2020-12-14:
@wrmichaelmak Hi Michael, just a quick question.
While I was playing around with the arducopter simulation in gazebo using the following commands:
1. gazebo --verbose ~/ardupilot_gazebo/worlds/iris_arducopter_runway.world
2. cd ~/ardupilot/ArduCopter/ &&
sim_vehicle.py -v ArduCopter -f gazebo-iris --console

and ran the commands into the MAVProxy terminal
1. mode guided
2. arm throttle
3. takeoff 3
4. position 20 0 0
5. mode rtl

The drone is crashing automatically during steps 4 and 5. It is going out of bounds I guess. During step 4, when it reaches 20m in the x-direction, it stays there for a while, drifts a lot with the roll and pitch values changing, and crashes, and during step 5, while landing, it lands a bit and then crashes on the side very speedily.

Do you know why this might be happening? Is there any glitch in the arducopter code? When I checked on the internet, some people had similar problems but didn't find any solution.

Thanks.

wrmichaelmak, 2020-12-14:
@abhiramdapke try it with position 20 0 3

abhiramdapke, 2020-12-15:
Oh yes. That was silly. Thanks.

abhiramdapke, 2020-12-21:
@wrmichaelmak 
Hi Michael,

I had a couple of questions regarding the same. I went through all the tutorials and have set up my environment properly, however, while I am running the simulation with square.cpp, the drone is crashing sometimes while reaching the second or third waypoint. Sometimes, it crashes abruptly without even reaching the waypoint. Do you know why this might be happening? The drone crashes abruptly at any point in time. I performed at least 10 simulations with this.

The error I am getting in the console is APM: GPS glitch in red and keep getting yaw re-aligned in yellow.

I wanted to run the scripts on a real drone after I get the simulation correctly.
For doing it on the real drone, I need to run only these two commands in two separate terminals, is that correct? or do I need to run anything else?

1. roslaunch iq_sim apm.launch  (which is the mavros node communication)
2. rosrun iq_gnc square (The script)
I am a bit concerned about running the script on a real drone because there are crashes during the simulation. Do you have any idea why this might be happening?  Also, would it be safe to run them on a real drone ignoring the GPS glitch because this is arducopter but the drone will have a stable GPS? Any suggestions would be highly appreciated.

One other thing is that the manual controller doesn't have a GUIDED mode in it. So will I need QGC every time to switch to GUIDED mode? Would it work?

Thank you.

wrmichaelmak, 2020-12-21:
@abhiramdapke I am not sure what will be the cause of the problem. I have never encountered such problems before. Additionally, it is your call to decide if it is safe enough to fly the drone. Another thing is I have previously mentioned before you can ask your Pixhawk to enter into GUIDED mode via ROS commands if you want. Not sure if you read that. However, to answer your question you can ask QGC to ask the Pixhawk to enter GUIDED mode.

abhiramdapke, 2020-12-21:
Yes, I have read that comment. I am still working on that. Thanks for your reply.

closed 2026-11-04 by github-actions[bot]