mavlink/mavros #1514: RC Override Crazy maneuvers

HarshilNaikk, 2020-12-08:
----

### Issue details

I am Using the /mavros/rc/override topic to publish input values to an sitl simulation of the iris drone. 
I followed all the proper steps, and sent the following channel inputs:
channels: [65535, 65535, 1500, 65535, 65535, 65535, 65535, 65535]"
Instead of the drone going up, it yaws a lot and spins out of control, ascending up with higher and higher speed. 
If i now do the same for roll and pitch instead of throttle, it does the same crazy maneuver. 

I used the following command as it is:

rostopic pub -r 10 /mavros/rc/override mavros_msgs/OverrideRCIn "channels: [65535, 65535, 1500, 65535, 65535, 65535, 65535, 65535]"

Please Help!

### MAVROS version and platform

Mavros: ?Latest one afaik?
ROS: ?Melodic?
Ubuntu: ?18.04?

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: ?3.7.1?

### Node logs

```
started roslaunch server http://harshil-Lenovo-ideapad-320-15IKB:40989/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
[... 562 lines cut ...]
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.081203"
      - 
        key: "Last remote time (s)"
        value: "266.564000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.336107715"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 5719 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 24, 30, 31, 32, 33, 36, 74, 331, 76, 83, 340, 85, 230, 105, 111, 241, 242, 245

```

nm47, 2021-04-19:
Did you ever figure this out?

HarshilNaikk, 2021-04-23:
@nm47 Yeah I ditched the use of the topic altogether. It was just super unstable to use. So I used another topic - manual_control
I'd really suggest you try the same.

closed 2026-11-04 by github-actions[bot]