mavlink/mavros #1515: İnstalling the older version of mavros?

negriye, 2020-12-11:
Hello all,

I would like to install mavros version 0.29.2 to resolve some issues in my project. I did my research and found this link https://github.com/mavlink/mavros/issues/941 , and learned that you can specify the version you want to install. My question is .. how? I've read the documentation of rosinstall nothing exactly shows the usage. I've tried

rosinstall_generator --upstream mavros=0.29.2 --rosdistro kinetic | tee -a /tmp/mavros.rosinstall

rosinstall_generator --upstream mavros/releases/tag/0.29.2 --rosdistro kinetic | tee -a /tmp/mavros.rosinstall

rosinstall_generator --from-path *path_to_downloaded_folder_mavros_0.29.2* --rosdistro kinetic| tee -a /tmp/mavros.rosinstall

rosinstall_generator --tar *mavros_0.29.2.tar.gz* --rosdistro kinetic | tee -a /tmp/mavros.rosinstall

and some more and all returned the same

The following not released packages/stacks will be ignored: *input_by_me*

No packages/stacks left after ignoring not released

Would please someone help me out, I feel like this should not be this complicated.

Best regards,

p.s. : my enviorement is ubuntu 16.04 with ros-kinetic

and all installations neceserray are done (lke wstool etc.)

vooon, 2020-12-11:
Probably easiest solution:
1. run rosinstall_generator
2. manually edit `version` tag.

See for example https://github.com/mavlink/mavros/blob/master/dependencies.rosinstall

negriye, 2020-12-11:
Thank you for your reply, I've already tried that, it just installs the updated version, namely 1.5.0. 
Isn't there any way to specify the version of the mavros to be installed?

vooon, 2020-12-12:
Did you try to edit version in the yaml file before calling wstool?

vooon, 2020-12-12:
Wstool searches for branch and tags, so `version: 0.29.2` i think should work. But you also need older version of mavlink.

closed 2026-11-04 by github-actions[bot]