mavlink/mavros #1516: Error building mavros from source

eldewany, 2020-12-14:
### Issue details

I'm trying to build mavros from source to create some custom plugins, i'm following the installation instructions but at the time of build i always get this error at command

Errors     << mavros:make /home/eldewany/catkin_ws2/logs/mavros/build.make.006.log
/usr/bin/ld: /usr/local/lib/libGeographic.a(Geoid.o): relocation R_X86_64_PC32 against symbol `_ZN13GeographicLib5Geoid4c3n_E' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [/home/eldewany/catkin_ws2/devel/.private/mavros/lib/libmavros.so] Error 1
make[1]: *** [CMakeFiles/mavros.dir/all] Error 2
make: *** [all] Error 2
cd /home/eldewany/catkin_ws2/build/mavros; catkin build --get-env mavros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << mavros:make                  [ Exited with code 2 ]              
Failed    <<< mavros                       [ 2.2 seconds ]                     
Abandoned <<< mavros_extras                [ Unrelated job failed ]            
Abandoned <<< test_mavros                  [ Unrelated job failed ]            
[build] Summary: 3 of 6 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 2 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 5.9 seconds total. 

### MAVROS version and platform

Mavros: upstream release
ROS: Melodic
Ubuntu: ?18.04?

I tried to install Geographics lib from source and still the same issue. what can i do ?

vooon, 2020-12-16:
What distro you are trying to use?
What hardware you are aiming to?

eldewany, 2020-12-17:
@vooon Ubuntu 18.05 , i'm planning to use it on an nvidia jetson nano with  pixhawk 1 (fmu v3)

vooon, 2020-12-17:
For Jetson you have to use ARMv8 and linker says something about x86_64. Where are you trying to compile?

eldewany, 2020-12-17:
@vooon I'm trying to compile on my main intel pc at the moment

vooon, 2020-12-17:
How you built libgeo? Probably you really need to add -fPIC.

eldewany, 2020-12-17:
@vooon i used the one that auto installs from mavros setup script.. I tried to rebuild it from source but i didn't know where to add the fPIC argument since I'm not very good with linux.

vooon, 2020-12-17:
Usual setup should just use deb package, but i'm not quite sure where you get `.a` lib instead of `.so`.
Probably you'll better to use ros docker image to start working with mavros and return to the problem later.

eldewany, 2020-12-20:
@vooon So i solved the issue by adding -fPIC to CXX_FLAGS in the file geographiclib source folder/Tools/Makefile.mk , and then rebuild the libraries and it worked

vooon, 2020-12-21:
Great! Closing as solved.

closed 2020-12-21 by vooon

ninja-hu, 2021-07-06:
I saw a similar error for ROS Noetic and have added full resolution steps for future references.

MAVROS version and platform
Mavros: upstream release
ROS: Noetic
Ubuntu: 20.04

The first error is https://github.com/ethz-asl/rotors_simulator/issues/520 and using the solution does not work for libgeographic-dev .  After downloading the Geographic lib from https://geographiclib.sourceforge.io/html/install.html#autoconf , the next error seen is 

========== Trace =======================
```
I'm trying to build mavros from source to create some custom plugins, i'm following the installation instructions but at the time of build i always get this error at command

Errors << mavros:make /home/eldewany/catkin_ws2/logs/mavros/build.make.006.log
/usr/bin/ld: /usr/local/lib/libGeographic.a(Geoid.o): relocation R_X86_64_PC32 against symbol `_ZN13GeographicLib5Geoid4c3n_E' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [/home/eldewany/catkin_ws2/devel/.private/mavros/lib/libmavros.so] Error 1
make[1]: *** [CMakeFiles/mavros.dir/all] Error 2
make: *** [all] Error 2
```

To resolve this , edit the following files
```
src/Makefile.mk
tools/Makefile.mk
```
and add  -fPIC to CXX_FLAGS 

After this change, build again. Mavros compilation is now successful