mavlink/mavros #1517: Mavros version incompatible between two ubuntu systems?

Shiyu-Liu, 2020-12-17:
### Issue details
Hello guys, I am trying to run controller from my PC to control a Pixhawk-based quadrotor, with Ubuntu 18.04 OS installed both on my PC and Raspberry Pi. Everything was well configured but when I run the node, it shows an error on the Mavros node of the drone: 

**[ERROR] [1608214083.907738946]: Client [/autopilot_drone] wants topic /crazy2fly5/mavros/state to have datatype/md5sum [mavros_msgs/State/ce783f756cab1193cb71ba9e90fece50], but our version has [mavros_msgs/State/65cd0a9fff993b062b91e354554ec7e9]. Dropping connection.**

After searching on line, I found that it might be a mavros version problem. However, I run the same operating system on both two sides with Ros Melodic installed. And the Mavros was installed also by the same procedure, the binary installation. So I am wondering if anyone has an idea about this?

I checked the mavros version, and provide the detailed information below. 
 
### MAVROS version and platform

Mavros: 1.5.0
ROS: Melodic
Ubuntu: 18.04

The results when I checked both sides about the mavros version:

Raspberry Pi:
pi@drone5:~$ rosversion mavros               ->     1.5.0
pi@drone5:~$ rosversion mavros_msgs     ->    1.5.0
pi@drone5:~$ rosversion mavlink               ->     2020.11.11
pi@drone5:~$ apt list ros-melodic-mavros
Listing... Done
**ros-melodic-mavros/bionic,now 1.5.0-1bionic.20201113.223737 arm64 [installed]**

PC:
user@po-5400:~$ rosversion mavros               ->     1.5.0
user@po-5400:~$ rosversion mavros_msgs     ->    1.5.0
user@po-5400:~$ rosversion mavlink               ->     2020.11.11
user@po-5400:~$ apt list ros-melodic-mavros
Listing... Done
**ros-melodic-mavros/bionic,now 1.5.0-1bionic.20201113.182456 amd64  [installed]**

I am thinking now if this is because the mavros I installed on the Raspberry Pi has a more recent version ? (1.5.0-1bionic.20201113.223737 compared to xxxxxx.182456)
I tried to upgrade that of my PC, but it said that it is already the latest version.

I will appreciate it if anyone has the similar experience and could share their ideas. Thanks !

vooon, 2020-12-17:
That is weird. But maybe something else also different?
Possibly some of the base messages are different (at compile time) so you got that.

Anyway that problem is lower level to mavros - somewhere at ros messaging.
I suggest that if you update both sides it will start working.

Shiyu-Liu, 2020-12-17:
> That is weird. But maybe something else also different?
> Possibly some of the base messages are different (at compile time) so you got that.
> 
> Anyway that problem is lower level to mavros - somewhere at ros messaging.
> I suggest that if you update both sides it will start working.

Hi vooon, thanks for your reply. Actually I've updated the mavros on both sides, and they are already of the latest version.
I tried to install a previous version(mavros 1.1.0) though on the Raspberry Pi, via source installation method. 
It seemed to start working, cause I don't have the error about data type and version stuff anymore. However, I got a lot of errors on plugin loading exception. An example of that is given here:
**Plugin landing_target load exception: Could not find library corresponding to plugin landing_target. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.**
I've gotten dozen plugins failed to load while the others are successfully loaded. 

I saw a similar topic about that plugin error. Gonna keep on doing my investigation on this issue.

vooon, 2020-12-18:
Ensure that you sourced setup.bash file from your environment (workspace or install directory).
Then check that you have `libmavros_extras.so`. Error tells me that plugin lib sees extras plugin xml file, but could not find shared library.

Shiyu-Liu, 2020-12-22:
> Ensure that you sourced setup.bash file from your environment (workspace or install directory).
> Then check that you have `libmavros_extras.so`. Error tells me that pluginlib sees extras plugin xml file, but could not find shared library.

Yes, you are right. I forgot to also compile the mavros_extras from source. But after that, I got the same error as before. 
**[ERROR] [1608633280.123556557]: Client  [/autopilot_drone] wants topic /crazy2fly5/mavros/state to have datatype/md5sum [mavros_msgs/State/65cd0a9fff993b062b91e354554ec7e9], but our version has [mavros_msgs/State/ce783f756cab1193cb71ba9e90fece50]. Dropping connection.**
Really don't know why it's like that.

closed 2021-01-19 by Shiyu-Liu

KhuongDiep911, 2022-11-16:
----

### Issue details
[image]

Hello Shiyu-Liu,

Can you give me some advice on how to solve this problem? I am facing the problem that has some similarity with you and I don't know how to solve it. I have tried to replace the line "65cd0a9fff993b062b91e354554ec7e9" by the line "9e3d873fae342c8f48a8bd64c53d991e" in "/opt/ros/noetic/mavros_msgs/State.h" file but it didn't work. I also tried to uninstall and reinstall the mavros and mavlink too. Do you know how to solve it? I really need help!

Thank you in advance for your help!

### Platform
MAVROS: 1.14.0
Mavros_msgs: 1.14.0
Mavlink: 2022.8.8
ROS: Noetic
Ubuntu: 20.04

### Autopilot type

[  ] ArduPilot
[ x ] PX4

```

Shiyu-Liu, 2022-11-16:
> ### Issue details
> [image]
>
> Hello Shiyu-Liu,
>
> Can you give me some advice on how to solve this problem? I am facing the problem that has some similarity with you and I don't know how to solve it. I [...]

Hello,
I don't remember what exactly was the issue, but after reading my own answers to that issue almost two years ago, your problem might be solved by reinstalling on both sides (running your subscriber and publisher) the mavros and mavros_msg from binary:
**$ sudo apt-get install ros-<your-ros-distro>-mavros ros-<your-ros-distro>-mavros-extras**.

KhuongDiep911, 2022-11-16:
Hi,

I have tried this solution but it hasn't worked. 
In addition, I haven't run my code on a real drone yet. I am now doing my study on sitl and gazebo simulator. Unluckily, I am stuck with this problem.

Shiyu-Liu, 2022-11-16:
> Hi,
> 
> I have tried this solution but it hasn't worked. In addition, I haven't run my code on a real drone yet. I am now doing my study on sitl and gazebo simulator. Unluckily, I am stuck with this problem.

In this case, as you're running nodes on the same PC, there shouldn't be any issue related to the different versions. Maybe check your gazebo simulation environment (for simulating the mavros) and probably it is using mavros_msgs in an incompatible way. 
I cannot provide any more helps since I have totally moved to ros2 without using  mavros any more. I hope you can find out the solution.

KhuongDiep911, 2022-11-16:
Okay, I will try your suggestion. Thank you so much for your help!