mavlink/mavros #1518: mavros version 1.5.0 doesn't publish /mavros/state topic

bigsuperZZZX, 2020-12-24:
Current stable version 1.5.0 doesn't publish "/mavros/state" topic.
I installed mavros 1.5.0 following binary installation method and launched the px4.launch to connect to a pixhawk flight controller.
Various topics are publishing normally except the "/mavros/state" topic
[image].

Following tests are done:
1. Three different FCUs: Pixracer, CUAV V5+, Holybro pix32 v5. None of which make the "/mavros/state" published.
2. An older version of mavros is tested: a source installed mavros 1.0.0. That topic published normaly.
3. Two different ubuntu version 16.04 and 18.04 are tested, On none of them that topic published.

### MAVROS version and platform
Mavros: 1.5.0
ROS: Kinetic and Melodic
Ubuntu: 16.04 and 18.04

### Autopilot type and version
PX4
Version: 1.11.2

### Node logs
Everything seems normal.
```
process[mavros-2]: started with pid [65618]
[ INFO] [1608831037.858531027]: FCU URL: /dev/ttyACM0:921600
[ INFO] [1608831037.860076302]: serial0: device: /dev/ttyACM0 @ 921600 bps
[ INFO] [1608831037.862222815]: GCS bridge disabled
[ INFO] [1608831037.867997562]: Plugin 3dr_radio loaded
[ INFO] [1608831037.869556745]: Plugin 3dr_radio initialized
[ INFO] [1608831037.869631167]: Plugin actuator_control loaded
[ INFO] [1608831037.871592256]: Plugin actuator_control initialized
[ INFO] [1608831037.873414478]: Plugin adsb loaded
[ INFO] [1608831037.875581494]: Plugin adsb initialized
[ INFO] [1608831037.875661597]: Plugin altitude loaded
[ INFO] [1608831037.876261475]: Plugin altitude initialized
[ INFO] [1608831037.876336522]: Plugin cam_imu_sync loaded
[ INFO] [1608831037.876785436]: Plugin cam_imu_sync initialized
[ INFO] [1608831037.876860874]: Plugin command loaded
[ INFO] [1608831037.883336253]: Plugin command initialized
[ INFO] [1608831037.883428523]: Plugin companion_process_status loaded
[ INFO] [1608831037.885258836]: Plugin companion_process_status initialized
[ INFO] [1608831037.885327278]: Plugin debug_value loaded
[ INFO] [1608831037.888549823]: Plugin debug_value initialized
[ INFO] [1608831037.888614045]: Plugin distance_sensor blacklisted
[ INFO] [1608831037.888684083]: Plugin esc_status loaded
[ INFO] [1608831037.889434952]: Plugin esc_status initialized
[ INFO] [1608831037.889522007]: Plugin fake_gps loaded
[ INFO] [1608831037.899830490]: Plugin fake_gps initialized
[ INFO] [1608831037.899933350]: Plugin ftp loaded
[ INFO] [1608831037.903550333]: Plugin ftp initialized
[ INFO] [1608831037.903649064]: Plugin global_position loaded
[ INFO] [1608831037.912167910]: Plugin global_position initialized
[ INFO] [1608831037.912258160]: Plugin gps_rtk loaded
[ INFO] [1608831037.913790923]: Plugin gps_rtk initialized
[ INFO] [1608831037.913865842]: Plugin gps_status loaded
[ INFO] [1608831037.914997976]: Plugin gps_status initialized
[ INFO] [1608831037.915082159]: Plugin hil loaded
[ INFO] [1608831037.923142321]: Plugin hil initialized
[ INFO] [1608831037.923234188]: Plugin home_position loaded
[ INFO] [1608831037.925944138]: Plugin home_position initialized
[ INFO] [1608831037.926035251]: Plugin imu loaded
[ INFO] [1608831037.930134223]: Plugin imu initialized
[ INFO] [1608831037.930226081]: Plugin landing_target loaded
[ INFO] [1608831037.938383393]: Plugin landing_target initialized
[ INFO] [1608831037.938489432]: Plugin local_position loaded
[ INFO] [1608831037.942574825]: Plugin local_position initialized
[ INFO] [1608831037.942674874]: Plugin log_transfer loaded
[ INFO] [1608831037.944080686]: Plugin log_transfer initialized
[ INFO] [1608831037.944165446]: Plugin manual_control loaded
[ INFO] [1608831037.945715494]: Plugin manual_control initialized
[ INFO] [1608831037.945872853]: Plugin mocap_pose_estimate loaded
[... 217 lines cut ...]
        key: "Last RTT (ms)"
        value: "2.111450"
      - 
        key: "Mean RTT (ms)"
        value: "1.922639"
      - 
        key: "Last remote time (s)"
        value: "1255.658136000"
      - 
        key: "Estimated time offset (s)"
        value: "1608829703.759995222"
---

```

### Check ID
```
ERROR. I got 0 addresses, but not your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages
```

Thank you for your reply!

vooon, 2020-12-24:
That looks like you didn't get HEARTBEAT's. I'm not quite sure where to look because there are almost no changes in libmavconn since 1.0.
Although one big change comes from mavlink itself with moves of some messages from common.

Could you try to get latest mavros and mavlink 2020.12.12 and try it with freshest PX4?
Also please look into `/mavlink/from` topic, it should have some stream of messages. From `check id` it looks like completely silent.

bigsuperZZZX, 2020-12-25:
Following the given hints, I think I've found the solution, which seems really wired but helpful.

After various trials and comparisons, I found that the problem comes from the SD card in FCU. All tested FCUs have no SD card installed since last firmware burning. As soon as I insert a SD card and reboot the FCU, that problem disappeared and the required topic published normally. After that, I removed the SD card again and rebooted the FCU, the mavros can still work well. 
So I think the solution to solve this problem is just inserting a SD card especially after the first firmware burning or parameter changing.

Thank you for your hints!

closed 2020-12-25 by bigsuperZZZX