mavlink/mavros #1519: apt install unable to locate package

lweingart, 2020-12-29:
Hello,

I'm trying to install mavros on Ubuntu 20.04 using apt install as described [here](https://github.com/mavlink/mavros/tree/master/mavros#installation) but I get
```
% sudo apt install ros-kinetic-mavros ros-kinetic-mavros-extras
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-mavros
E: Unable to locate package ros-kinetic-mavros-extras
```

Are there repositories to add?
I tried adding http://packages.ros.org/ros/ubuntu in the sources list but that didn't help

Any help would be greatly appreciated.

Best Regards,
Laurent

TSC21, 2020-12-29:
ROS Kinetic is not supported in Ubuntu 20.04. Only Noetic.  So use the `ros-noetic-` prefix. And I suppose you already followed the tutorial on how to install the ROS base components in http://wiki.ros.org/noetic/Installation/Ubuntu.

closed 2020-12-29 by TSC21

lweingart, 2020-12-29:
Hello,
Actually I arrived at that point following the PX4 documentation and I missed that tutorial.
Thank you very much for your help and your quick response.

Cheers

TSC21, 2020-12-29:
If you are new to ROS I recommend you you go first over the ROS tutorials before jumping into using MAVROS..