mavlink/mavros #1520: LLA to mavros coordinates?

niciBume, 2021-01-07:
### MAVROS version and platform

Mavros: 1.4.0
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

[  ] ArduPilot
[ X] PX4

### Question

Hi guys,

I would like to know if there's an option to to get the earth -> map transformation from mavros i.e I want to know how to convert an arbitrary LLA coordinate into my base_link frame? 

I haven't found a straight forward function for this. Is there one, or could you consider adding this as a feature?
Reason behind all of this is that I want to compare the PX4 and fix ground anchors to an onboard RTK. Anchors and PX4 are in map frame coordinates while RTK is in LLA, so I'd need a way to either convert LLA in map or map to LLA. So I feel this feature would be valuable to others as well.

Thanks  a lot!

vooon, 2021-01-08:
I think you can use geographic lib: https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/global_position.cpp#L317-L351

niciBume, 2021-01-08:
@vooon 
Thanks for the reply! Yeah that'd be an option. But I feel like a clean integrated solution to this would benefit quite some others as well. 

I circumvent this problem by including a custom message and injecting a publisher that publishes the origin that mavros uses at the codelines you stated. Now I use geographiclib  (as you suggested) and perform the transformation separately.

TSC21, 2021-01-08:
> I circumvent this problem by including a custom message and injecting a publisher that publishes the origin that mavros uses at the codelines you stated. Now I use geographiclib (as you suggested) and perform the transformation separately.

You can use https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/global_position.cpp#L495 and then get it from the FMU with https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/global_position.cpp#L222.

niciBume, 2021-01-08:
@TSC21 
Hmm... whats the difference from `g_origin` to `map_origin`? 
Because the geographic lib transformations from `g_origin`:
`
GeographicLib::Geocentric earth(GeographicLib::Constants::WGS84_a(),
				GeographicLib::Constants::WGS84_f());

earth.Forward(g_origin->position.latitude, g_origin->position.longitude, g_origin->position.altitude,
g_origin->position.latitude, g_origin->position.longitude, g_origin->position.altitude);
`

and the one from `map_origin` look pretty similar?
`
GeographicLib::Geocentric map(GeographicLib::Constants::WGS84_a(),
						GeographicLib::Constants::WGS84_f());

/**
* @brief Checks if the "map" origin is set.
* - If not, and the home position is also not received, it sets the current fix as the origin;
* - If the home position is received, it sets the "map" origin;
* - If the "map" origin is set, then it applies the rotations to the offset between the origin
* and the current local geocentric coordinates.
*/
// Current fix to ECEF
map.Forward(fix->latitude, fix->longitude, fix->altitude,
		map_point.x(), map_point.y(), map_point.z());

// Set the current fix as the "map" origin if it's not set
if (!is_map_init && fix->status.status >= sensor_msgs::NavSatStatus::STATUS_FIX) {
  map_origin.x() = fix->latitude;
  map_origin.y() = fix->longitude;
  map_origin.z() = fix->altitude;
  
  ecef_origin = map_point; // Local position is zero
  is_map_init = true;
}
`

So whats the difference between `map` and `earth` transformations?

Cheers!

closed 2026-11-04 by github-actions[bot]