mavlink/mavros #1524: Odometry plugin odom_cb missing estimator_type

ashwinvk94, 2021-01-18:
### Issue details
The Odometry plugin [does not set](https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/odom.cpp#L228) the [estimator_type](https://mavlink.io/en/messages/common.html#MAV_ESTIMATOR_TYPE) in the [Odometry message](https://mavlink.io/en/messages/common.html#ODOMETRY). This prevents the mavlink Odometry message from being published at the PX4 autoplilot since they [have a check](https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L1392-L1400) for this field.

### MAVROS version and platform
Mavros: master
ROS: Melodic
Ubuntu: 18.04
### Autopilot type and version
[  ] ArduPilot
[ x] PX4

TSC21, 2021-01-19:
Please issue a PR. Thank you!

ashwinvk94, 2021-01-19:
PR issued #1525

closed 2021-01-19 by ashwinvk94