mavlink/mavros #1526: Odometry Data from Odroid VIO not being sent to Pixhawk

nabatta, 2021-01-28:
### Issue details

Hello! I am attempting to use the [Auterion VIO node](https://docs.px4.io/master/en/computer_vision/visual_inertial_odometry.html) on my Odroid H2+. The VIO package seems to be working fine as I can echo /mavros/odometry/out and I can see the pose change as I move the camera around.

However, when I boot up QGC and go to the Mavlink Inspector, I am not getting any messages from the Odroid. I can see an ODOMETRY message but it is only being generated by the IMU's on the Pixhawk. However, if I run "rosrun mavros mavsafety arm" from the terminal, QGC responds to the command. So some messages must be getting through. I can also echo the heartbeat topic (/mavros/state) from the terminal and get a response. But there is no heartbeat message showing up in QGC from the Odroid (only from the Pixhawk).

I am also getting a message saying "HP: requesting home position" so I am not sure if this is causing the issue but I do not think so. I am relatively new to all this so hopefully this is a simple question. Thank you for the help!

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Melodic?
Ubuntu: ?18.04?

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: ?3.7.1?

### Node logs

```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 62
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: t265
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: True
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
[... 550 lines cut ...]
      - 
        key: "Last RTT (ms)"
        value: "5.119273"
      - 
        key: "Mean RTT (ms)"
        value: "4.342858"
      - 
        key: "Last remote time (s)"
        value: "388.537465000"
      - 
        key: "Estimated time offset (s)"
        value: "1611847902.391566277"
```

### Check ID

```
OK. I got messages from 1:1.

---
Received 5274 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 30, 31, 32, 36, 74, 331, 76, 77, 83, 340, 230, 105, 111, 241, 245

```

antonio-sc66, 2021-02-04:
Hi @nabatta I think the problem is that you might be using a MAVROS version in which the odom plugin doesn't send the proper estimator_type and PX4 discards that message. This was solved here #1524. If you can try to test the current master branch of MAVROS.

nabatta, 2021-02-05:
@antonio-sc66 I have MAVROS 1.5.1. Would this issue apply to that version? It seems like the problem described in #1524 is definitely what I am experiencing. However, when I looked for the odom.cpp file, it does not exist for some reason.

antonio-sc66, 2021-02-05:
@nabatta yes, that problem applies to 1.5.1. It has not been released and tagged yet, you have to download this repository and compile it yourself if you want to use it for the moment. The odom.cpp file should be in this path (mavros/mavros_extras/src/plugins/odom.cpp). 
Another option if you need to get it working now is to downgrade PX4 to the latest 1.10 version. This problem started in PX4 1.11 as it now checks the estimator_type and before 1.11 it didn't.

nabatta, 2021-02-05:
@antonio-sc66 I am currently using PX4 1.9.2. Do you know if this version also had the issue?

antonio-sc66, 2021-02-05:
@nabatta I don't know. I've been using the odom plugin and the vision_pose_estimate plugin and they both worked with PX4 1.10.1 and 1.11.3, but I haven't used them with Auterion VIO, I was sending the data from other algorithms.
Make sure you have MAV_ODOM_LP=1 so you can see the messages in QGroundControl and recheck the EKF2_AID_MASK

dagar, 2021-02-06:
Here's a fix for PX4. https://github.com/PX4/PX4-Autopilot/pull/16774

nabatta, 2021-02-08:
@antonio-sc66 Just downloaded the newest version of mavros from source but am still having the same issue. I doublechecked and the 2 lines in odom.cpp are definitely there. Not sure what the issue is now

nabatta, 2021-02-09:
@antonio-sc66 Just got it working! I simply updated to PX4 1.11.3 and that fixed the issue. The older firmware was still causing some issues even with the new fix to the mavros code. Thank you for the help!

closed 2026-11-04 by github-actions[bot]