mavlink/mavros #1528: Using fake_gps for custom GPS data

NishanthARao, 2021-02-05:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
I am trying to fly a drone in a gps-denied environment. For the time being, let's assume that I somehow know the GPS data in the latitude and longitude format, but the GPS sensor is unable to find any signal. Is there a way to send the custom GPS data at every time instant to the onboard estimator via the ```fake_gps``` rostopic? Please not that I am not using any motion capture system / visual localization systems for this project.

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Melodic?
Ubuntu: ?18.04?

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

Version: ?3.7.1?

### Node logs

```
... logging to /home/airl010/.ros/log/e6049f6c-675f-11eb-8fe4-b42e99b01d85/roslaunch-airl010-4253.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://airl010:41341/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
[... 517 lines cut ...]
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000320"
      - 
        key: "Last remote time (s)"
        value: "3741.084000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.315999935"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 5857 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 24, 30, 31, 32, 33, 36, 42, 74, 331, 76, 83, 340, 85, 230, 105, 111, 241, 242, 245
```

DogukanAltay, 2021-03-01:
if you are using a pixhawk with px4:
1. enable HIL_GPS parameter from qgroundcontrol. 
2. Publish your GPS data to /mavros/hil/gps topic
3. Confirm it through checking gpsraw topic from mavros or from mavlink inspector.

NishanthARao, 2021-03-05:
@DogukanAltay Thanks a lot! I was able to publish fake_gps to the drone, and the QGC is also detecting it.

closed 2021-03-05 by NishanthARao