mavlink/mavros #1532: FCU not connected and no packets received

FinnM4444, 2021-02-10:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

We are attempting to test our connection between a Pixhawk 1 running Ardupilot and a TX1 by trying to receive diagnostic data from the pixhawk. Using the command _rostopic echo -n1 /diagnostics_ tells us that FCU is not connected and we have received no packets.

We have tried this on serial port THS2 and multiple BAUD rates (57600, 921600, 115200). 

### MAVROS version and platform

Mavros: 1.5.2
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[ ArduRover ] ArduPilot
[ - ] PX4

Version: 4.0.0

### Node logs

```
... logging to /home/ugv1/.ros/log/19e8e930-6b3b-11eb-b0e8-00044b57fbc7/roslaunch-ugv1-desktop-17965.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ugv1-desktop:40327/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_id: 4
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/horiz_accuracy: 0.5
 * /mavros/fake_gps/mocap_transform: False
 * /mavros/fake_gps/mocap_withcovariance: False
 * /mavros/fake_gps/satellites_visible: 6
 * /mavros/fake_gps/speed_accuracy: 0.0
[... 370 lines cut ...]
      - 
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "0.000000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"

```

### Check ID

```
ERROR. I got 0 addresses, but not your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages

```

dimaxano, 2021-02-11:
Hi! Do you see any symbols appearing when run `cat /dev/ttyTHS2` on your TX1? Pixhawk should be switched on for this test

FinnM4444, 2021-02-16:
We've managed to fix the issue now, thank you for your help dimaxano, much appreciated.

FinnM4444, 2021-02-16:
It is all solved

closed 2021-02-16 by FinnM4444

dimaxano, 2021-02-16:
Really glad to hear that! :)
And what was the source of the problem?

FinnM4444, 2021-02-16:
It was in fact a hardware issue. Some pinouts were interpreted incorrectly
and so the wiring was done incorrectly.