mavlink/mavros #1533: Hil GPS data subscription

2ashishmohan, 2021-02-10:
Hello

I was writing a simple subscriber to subscribe to the gps data on a SITL quadcopter using mavros. This is my code

#!/usr/bin/env python
import rospy  
#from mavros_msgs.msg import HilGPS

def callback(gps):
    print (gps.geo.latitude)
    print (gps.geo.longitude)
    print (gps.geo.altitude)

rospy.init_node('gps_location', anonymous=True)
sub=rospy.Subscriber('/mavros/hil/gps','HilGPS',callback)
rospy.spin()

but when i run the code i am getting a error 
Traceback (most recent call last):
  File "./gpstest.py", line 11, in <module>
    sub=rospy.Subscriber('/mavros/hil/gps','HilGPS',callback)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 563, in __init__
    super(Subscriber, self).__init__(name, data_class, Registration.SUB)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 142, in __init__
    raise ValueError("data_class [%s] is not a class"%data_class) 
ValueError: data_class [HilGPS] is not a class

can anybody please help me with this, i am new to ROS programming

Niknu, 2021-03-02:
Your data type shall not be commented out at the top and the data type shouldn't be like that ' x ' .

This should work:
sub=rospy.Subscriber('/mavros/hil/gps',HilGPS,callback)

2ashishmohan, 2021-03-02:
Thank you for the reply . The code is working perfectly now

closed 2021-03-02 by 2ashishmohan