mavlink/mavros #1536: /mavros/state topic is silent when using companion computer

leocencetti, 2021-02-11:
### Issue details
Hi, I am trying to put a multi-copter in `OFFBOARD` mode, without success.
My setup is the following:
- PixHawk 4 running PX4 1.11.2
- NVidia Jetson Nano as companion computers set up according to the PX4 documentation [here](https://docs.px4.io/master/en/companion_computer/pixhawk_companion.html), running the `px4.launch` as follows:
    ```
    roslaunch mavros px4.launch fcu_url:=/dev/ttyTHS1:921600 gcs_url:=udp://:14555@192.168.0.105:14550
    ``` 
    where `/dev/ttyTHS1` is the PixHawk interface and `192.168.0.105` is the ground station computer (ROS master).
- OptiTrack MoCap system feeding live positional data through ROS (`POSITION` control mode works fine).
- Ground station computer running QGroundControl and `roscore`.

I have tried:
1. Manually publishing a position setpoint on the `/drone_1/setpoint_position/local` topic and then switching to `OFFBOARD` mode via  mapped RC switch. The mode is rejected and the following message is printed:
    ```
    [ERROR] [1511517243.676111562]: MODE: Unsupported FCU
    ```
2. Using the `offb_node` example from [here](https://docs.px4.io/master/en/ros/mavros_offboard.html) (changing the topics accordingly, of course). The node gets stuck in the following while loop, since  `current_state.connected` never evaluates to `true`:
    ```cpp
    while(ros::ok() && !current_state.connected){
        ros::spinOnce();
        rate.sleep();
    }
    ```
    The example works fine in SITL.
    
From further inspection, the problem seems to be with the `/drone_1/mavros/state` topic being silent:
```
rostopic hz /drone_1/mavros/state 
subscribed to [/drone_1/mavros/state]
no new messages
no new messages
no new messages
```
If echoed, the topic returns a single (I guess default) message:
```
rostopic echo /drone_1/mavros/state 
header: 
  seq: 0
  stamp: 
    secs: 1612964386
    nsecs:  32524857
  frame_id: ''
connected: False
armed: False
guided: False
manual_input: False
mode: ''
system_status: 0
```
I do receive the `HEARTBEAT` (from MAVLink Inspector in QGroundControl) as well as all other MAVROS topics (IMU, Position, Attitude, etc...).

### MAVROS version and platform

Mavros: 1.5.2
ROS: melodic
Ubuntu: 16.04 and 18.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: 1.11.2

### Node logs
<details>
  <summary>[click to expand]</summary>

```
[... 543 lines cut ...]
            key: "RSSI (dBm)"
            value: "-91.7"
          - 
            key: "Remote RSSI"
            value: "0"
          - 
            key: "Remote RSSI (dBm)"
            value: "-127.0"
          - 
            key: "Tx buffer (%)"
            value: "100"
          - 
            key: "Noice level"
            value: "69"
          - 
            key: "Remote noice level"
            value: "0"
          - 
            key: "Rx errors"
            value: "0"
          - 
            key: "Fixed"
            value: "0"
    ```
    </details>

antonio-sc66, 2021-02-11:
Hi @leocencetti  If you don't get /mavros/state msgs is because you are not connected to the FMU or there's a further problem with your connection.

The setup you describe with the Jetson Nano and Pixhawk is compatible and works, so it has to be a configuration problem.

I would try a few things, first make sure that you're connecting the telemetry cables to the correct pins on the Jetson (https://www.jetsonhacks.com/nvidia-jetson-nano-j41-header-pinout/) board TX is pin 8 and board RX is pin 10. You also need to connect the GND to one of the GND pins of the Jetson.

By default the Nano is publishing console data on ttyTHS1, which will interfere with what you want to do. Make sure you disable the running service with:
`systemctl stop nvgetty`
`systemctl disable nvgetty`
`udevadm trigger`

Then validate that ttyTHS1 is working well, you can follow this tutorial (https://www.jetsonhacks.com/2019/10/10/jetson-nano-uart/)

Then I would try to launch MAVROS with the roscore running on the Jetson without the gcs_url and check if you have a valid connection. (ex: rostopic echo /mavros/state and check that you receive connected: True)

You don't describe how you're using the roscore, but I think that in order to connect the Jetson with the other computer you are going to have to configure the ROS_MASTER_URI and ROS_IP. I'm not sure if that can be part of the problem.

leocencetti, 2021-02-13:
Hi @antonio-sc66, thank you for the reply.
As you suggested, I have verified the telemetry connection between the Jetson and the `TELEM2` port of the PixHawk, which does indeed work (tested successfully with the tutorial you linked).

> Then I would try to launch MAVROS with the roscore running on the Jetson without the gcs_url and check if you have a valid connection. (ex: rostopic echo /mavros/state and check that you receive connected: True)

The `nvgetty` service was enabled, thus I have disabled it. 
Nevertheless, launching MAVROS with `roscore` running on the Jetson and without the `gcs_url` option (as I had already done previously) still gives me the same outputas before: `/diagnostics` says FCU is disconnected, and `/mavros/state` is silent. 
The same goes with the `gcs_url` option. 
I am still able to read data from other telemetry topics (e.g. `/drone_1/mavros/imu/data`), thus I would rule out  hardware connection problems...
I have also tried to connect the Jetson to both `TELEM1` and `TELEM2` ports, with the same results. As a reference, here is my configuration of the MavLink parameters (from QGC):
[image]

> You don't describe how you're using the roscore, but I think that in order to connect the Jetson with the other computer you are going to have to configure the ROS_MASTER_URI and ROS_IP. I'm not sure if that can be part of the problem.

Yes, I have the following ENV configuration: 

- Jetson:
    ```bash
    ROS_MASTER_URI=http://192.168.0.105:11311
    ROS_IP=192.168.0.223
    ```
- GSC:
    ```bash
    ROS_MASTER_URI=http://localhost:11311
    ROS_IP=192.168.0.105
    ```
where `192.168.0.105` is the IP of the ground station and `192.168.0.223` the IP of the Jetson.

antonio-sc66, 2021-02-14:
@leocencetti You have a really low MAV_1_RATE try a higher value like 921600. The parameters described here (https://docs.px4.io/v1.9.0/en/peripherals/mavlink_peripherals.html#example) work for me.

I've seen that you use a custom MAV_SYS_ID, MAVROS has parameters for the sys_id and comp_id in the launch file, they're tgt_system and tgt_component. Make sure you have the same values in MAVROS and PX4.

leocencetti, 2021-02-18:
@antonio-sc66 sorry for the delay...

> You have a really low MAV_1_RATE try a higher value like 921600. The parameters described here (https://docs.px4.io/v1.9.0/en/peripherals/mavlink_peripherals.html#example) work for me.

changed them to `57600` and `921600` respectively, as said in the guide

> I've seen that you use a custom MAV_SYS_ID, MAVROS has parameters for the sys_id and comp_id in the launch file, they're tgt_system and tgt_component. Make sure you have the same values in MAVROS and PX4.
Yes, we have the defaults of the launch file matching the "custom" ID
```xml
<launch>
        <!-- vim: set ft=xml noet : -->
        <!-- example launch script for PX4 based FCU's -->

        <arg name="fcu_url" default="/dev/ttyACM0:57600" />
        <arg name="gcs_url" default="" />
        <arg name="tgt_system" default="9" />
        <arg name="tgt_component" default="9" />
        <arg name="log_output" default="screen" />
        <arg name="fcu_protocol" default="v2.0" />
        <arg name="respawn_mavros" default="false" />

        <include file="$(find mavros)/launch/node.launch">
                <arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
                <arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />

                <arg name="fcu_url" value="$(arg fcu_url)" />
                <arg name="gcs_url" value="$(arg gcs_url)" />
                <arg name="tgt_system" value="$(arg tgt_system)" />
                <arg name="tgt_component" value="$(arg tgt_component)" />
                <arg name="log_output" value="$(arg log_output)" />
                <arg name="fcu_protocol" value="$(arg fcu_protocol)" />
                <arg name="respawn_mavros" default="$(arg respawn_mavros)" />
        </include>
</launch>
```

Unfortunately, still the same results,` /diagnostics` says FCU is disconnected, and `/mavros/state` is silent.

antonio-sc66, 2021-02-19:
@leocencetti. Well, that seems strange, I'm not sure what else you can try

rob-clarke, 2021-06-11:
@leocencetti May be a little late for you but: In the launch file you've got `tgt_component` set to 9, but your param screenshot shows `MAV_COMP_ID` set to 1. That might be the problem.

closed 2026-11-03 by github-actions[bot]