mavlink/mavros #1540: obstacle_distance plugin [ERROR]DS: no mapping for sensor id: 13

siddharthcb, 2021-02-19:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

Hi, I have added an [obstacle_distance](https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/obstacle_distance.cpp) plugin in mavros and have written an extra script to accommodate 72 laser points.
when i run roslaunch mavros`($ roslaunch mavros apm.launch fcu_url:=/dev/ttyUSB1:921600)` without running this extra script everything works fine but when i run this extra script for laser.
i get this error on the terminal

```
[ERROR] [1613710136.068156132]: DS: no mapping for sensor id: 15, type: 0, orientation: 5
[ERROR] [1613710136.069381900]: DS: no mapping for sensor id: 16, type: 0, orientation: 6
[ERROR] [1613710136.069772193]: DS: no mapping for sensor id: 17, type: 0, orientation: 7
[ERROR] [1613710136.168928011]: DS: no mapping for sensor id: 10, type: 0, orientation: 0
[ERROR] [1613710136.170112119]: DS: no mapping for sensor id: 11, type: 0, orientation: 1
[ERROR] [1613710136.170890259]: DS: no mapping for sensor id: 12, type: 0, orientation: 2
[ERROR] [1613710136.171694176]: DS: no mapping for sensor id: 13, type: 0, orientation: 3
[ERROR] [1613710136.172040311]: DS: no mapping for sensor id: 14, type: 0, orientation: 4
[ERROR] [1613710136.172359888]: DS: no mapping for sensor id: 15, type: 0, orientation: 5
[ERROR] [1613710136.172740809]: DS: no mapping for sensor id: 16, type: 0, orientation: 6

```
ROS: melodic
Ubuntu: 18.04
APM software

as @TSC21 mentioned in this [thread](https://github.com/mavlink/mavros/issues/1071), i am supposed to change the config in apm_config.yaml, though i have added config for obstacle_distance in this file i am still finding the same error.
my apm_config.yaml looks like this:

```
# Common configuration for APM2 autopilot
#
# node:
startup_px4_usb_quirk: false
[... 237 lines cut ...]
  twist_cov: true       # enable listen to twist with covariance topic

# vibration
vibration:
  frame_id: "base_link"

# wheel_odometry
wheel_odometry:
  count: 2           # number of wheels to compute odometry
  use_rpm: false     # use wheel's RPM instead of cumulative distance to compute odometry
  wheel0: {x: 0.0, y: -0.15, radius: 0.05} # x-, y-offset (m,NED) and radius (m)
  wheel1: {x: 0.0, y:  0.15, radius: 0.05} # x-, y-offset (m,NED) and radius (m)
  send_raw: true              # send wheel's RPM and cumulative distance (~/wheel_odometry/rpm, ~/wheel_odometry/distance)
  send_twist: false           # send geometry_msgs/TwistWithCovarianceStamped instead of nav_msgs/Odometry
  frame_id: "map"             # origin frame
  child_frame_id: "base_link" # body-fixed frame
  vel_error: 0.1              # wheel velocity measurement error 1-std (m/s)
  tf:
    send: true
    frame_id: "map"
    child_frame_id: "base_link"

# vim:set ts=2 sw=2 et:

```

vooon, 2021-02-19:
Are you sure that you are publishing to `/mavros/obstacle/send`?

Because log message comes from `distance_sensor`, which requires to configure a mapping.

siddharthcb, 2021-02-19:
@vooon I am publishing to `/mavros/obstacle/send`
Here is my node.launch file
```
<launch>
	<!-- vim: set ft=xml noet : -->
	<!-- base node launch file-->

	<arg name="fcu_url" />
	<arg name="gcs_url" />
	<arg name="tgt_system" />
	<arg name="tgt_component" />
	<arg name="pluginlists_yaml" />
	<arg name="config_yaml" />
	<arg name="log_output" default="screen" />
	<arg name="fcu_protocol" default="v2.0" />
	<arg name="respawn_mavros" default="false" />

	<node pkg="mavros" type="mavros_node" name="mavros" required="$(eval not respawn_mavros)" clear_params="true" output="$(arg log_output)" respawn="$(arg respawn_mavros)">
		<param name="fcu_url" value="$(arg fcu_url)" />
		<param name="gcs_url" value="$(arg gcs_url)" />
		<param name="target_system_id" value="$(arg tgt_system)" />
		<param name="target_component_id" value="$(arg tgt_component)" />
		<param name="fcu_protocol" value="$(arg fcu_protocol)" />

		<!-- load blacklist, config -->
		<rosparam command="load" file="$(arg pluginlists_yaml)" />
		<rosparam command="load" file="$(arg config_yaml)" />
                <!-- remap from="/mavros/setpoint_velocity/cmd_vel_unstamped" to="/cmd_vel" /-->
                <remap from="/mavros/vision_pose/pose" to="/robot_pose" />
                <remap from="/mavros/obstacle/send" to="/revised_scan" />
	</node>
</launch>
```

> log messages comes from distance_sensor directly
so i do not have to configure obstacle_distance separately in the apm_config.yaml file?
and what are rangefinder_pub and rangefinder_sub topics? where is it being published?

siddharthcb, 2021-02-19:
@vooon is there a necessity to change apm_pluginlist.yaml?
my apm_pluginlist.yaml looks like this:
```
plugin_blacklist:
# common
- actuator_control
- ftp
- safety_area
- hil
# extras
- altitude
- debug_value
- image_pub
- px4flow
- vibration
- vision_speed_estimate
- wheel_odometry

plugin_whitelist: []
#- 'sys_*'

```

siddharthcb, 2021-02-19:
UPDATE:
when i added "distance_sensor" to apm_pluginlist.yaml(under plugin_blacklist), the error disappeared and I can still see the laser data in APMs  "proximity" and "obstacle_distance" plugin in mavlink inspector BUT IS THIS THE CORRECT WAY?

vooon, 2021-02-20:
It seems that APM sends back DISTANCE_SENSOR messages when you're using OBSTACLE_DISTANCE.
Excluding plugin from loading should be fine.

siddharthcb, 2021-02-20:
@voon Thank you

closed 2021-02-20 by siddharthcb