mavlink/mavros #1541: mavros/setpoint_raw/local not working for velocity z

NishanthARao, 2021-02-24:
This is only a bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hello. I want to control the drone by providing desired position (x, y) and the height using velocity z (position z should be ignored). I have set the flags for the PositionTarget message as follows:
```
mavros_msgs::PositionTarget mv_setpoint;
mv_setpoint.type_mask = 476; //(000111011100)
mv_setpoint.coordinate_frame = 1; //(LOCAL_NED)

//I want the drone to move upward at a speed of 1.0m/s along the z-axis.
mv_setpoint.position.x = 0.0f;
mv_setpoint.position.y = 0.0f;
mv_setpoint.velocity.z = 1.0f;
```
But when I do this, the drone just arms, but doesn't fly. I am new to mavros, so I would really like to know if I am making any mistakes / missing out on something. Thanks in advance.
### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Melodic?
Ubuntu: ?18.04?

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: ?3.7.1?

### Node logs

```
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://airl010:43871/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
[... 539 lines cut ...]
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "11.724000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.176000000"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 5965 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 8, 140, 141, 147, 24, 30, 31, 32, 33, 36, 42, 74, 331, 76, 83, 340, 85, 230, 105, 111, 241, 242, 245
```

wrmichaelmak, 2021-02-24:
You will need to takeoff first. You have to do via cmd/takeoff

NishanthARao, 2021-02-26:
@wrmichaelmak Thank you for the suggestion. However, this doesn't work for me. As soon as the drone takes off, I try switching the mode from ```AUTO.TAKEOFF``` to ```OFFBOARD``` so that I can publish the desired setpoints, but I get ```Unexpected command 176```, and the drone lands. 

Any idea where I might be doing any mistakes?

Thanks in advance.

wrmichaelmak, 2021-02-26:
@NishanthARao The way to do it is to set to offboard mode first and then send takeoff command via offboard mode.

NishanthARao, 2021-03-01:
@wrmichaelmak Thanks a lot! The drone now flies accordingly.

closed 2021-03-01 by NishanthARao