mavlink/mavros #1542: Run MAVROS without SITL

spellftw, 2021-02-24:
How to run mavros without sitl. I just want to use mavros with gazebo without it connecting to fcu. I have my own robot and I want to command it by using mavros. 

Thanks

Scarf195, 2021-03-03:
Hi, I have the same problem, did you find any solutions?

dimaxano, 2021-03-03:
gazebo is a virtual replacement of your robot. You can either connect it to gazebo or real robot.
@spellftw do you have any kind of FCU on your robot?

spellftw, 2021-03-03:
@dimaxano, Hi I am pretty new to this topic, my robot is a ship with 2 thrusters. I want to control it using ArduPilot in gazebo without using SITL. Is that possible? I want to check whether we can trick the arduPilot in reading "fake" GPS and IMU values otherwise it will give readings as 0.
Thanks

dimaxano, 2021-03-04:
Hi!
If you want to control the robot in real life you need to put hardware [FCU](https://pixhawk.org/products/) on your ship and connect your trusters to it. Gazebo is not used here.
If you have a model of your ship (or similar one) in gazebo you may run Ardupilot SITL and connect it to gazebo ([instruction](https://ardupilot.org/dev/docs/using-gazebo-simulator-with-sitl.html)). Then you will be able to control simulated ship
Tell me if something is still no clear

spellftw, 2021-03-04:
Hi, 
So to make everything clear:
1) If I have a ship model with me I need to convert it accordingly to ardupilot model file. 
2)This is the only way to communicate with MAVROS, without SITL it is not possible? If I want to simulate my ship with ardupilot in gazebo I will have to use SITL, I can't by pass it?

Thanks a Lot!

Scarf195, 2021-03-04:
@spellftw but are you using any FCU hardware? I also have my own ship delevoped in ROS and I would only to send gps data to QGrounControl through mavros. Do you know if this is possible?

spellftw, 2021-03-04:
@Scarf195 , I have Navio2. But I am trying to do the same thing as you. I also would like to know if it is possible. Are you also trying to run a mavros node without connecting to FCU Url. So that you can control your ship through mavros without using SITL?

vooon, 2021-03-04:
I could not understand what you are trying to achieve.
Mavros is a bridge between Flight Control software (a firmware) which runs somewhere, on a Flight Control Unit, such as Pixhawk autopilot, or on some computer in virtual environments.

Software in Loop or Hardware in Loop not matters. You're talking to control software that's it.

spellftw, 2021-03-04:
I am trying to develop a pilot for my ship which can command its thrusters. The pilot is a ros node which should connect to ardupilot controller.  Trying to establish a communication between Gazebo and ArduPilot to deliver commands to the ArduPilot via the pilot node.

vooon, 2021-03-04:
Then you first need to connect all the hardware, like thrusters ESC's, RC receiver (at least to have kill-switch) etc, to the FCU which runs Ardupilot (i suppose that you need ArduRover variant for boat).

Then make a connection from APM to Mavros by setting suitable fcu_url.
At that stage your node would be able to use mavros topics to control APM.

Then use whatever software you want to control your node.

Scarf195, 2021-03-04:
@vooon so MAVROS is a bridge for FCU-ROS but not for ROS-GCS right? I need to send GPS data from my vehicle (fully controlled by ROS running on board) to a ground control station (like QGC). Do you have any suggestions? Thank you very much for your help.

vooon, 2021-03-04:
@Scarf195 yes, i only have plans to make possible to write GCS-plugins for ROS2. For mavros 1 it's unlikely to make any new big changes.
My suggestion - use pymavlink to write prototype node which would implement some minimal set of autopilot messages (see PX4 and APM code for examples).
Then if you have issues with performance, you could use libmavconn to write C++ node.

spellftw, 2021-03-04:
Thanks a lot for the information and patience!

spellftw, 2021-03-04:
@vooon Thanks a lot for your patience and your help.

vooon, 2021-03-04:
Probably confusion here caused by GCS proxy feature, which simply forwards FCU messages to GCS.

For @Scarf195 's objective, he needs to act as FCU, so GCS could talk to it.

For @spellftw - connect existing APM firmware to mavros, then use it to control FC. And it's possible to use MissionPlanner trough proxy.
For Gazebo i suppose you have to write custom node. But i do not know how, i only use it as a simulation environment.

natetoombs, 2021-06-21:
@Scarf195 Have you made any progress with sending GPS data to a ground control station?

closed 2026-11-03 by github-actions[bot]