mavlink/mavros #1552: Issue with /mavros/fake_gps/mocap/tf topic

NishanthARao, 2021-03-08:
----

### Issue details

Hi. I was experimenting with the fake_gps topic that mavros publishes. For now, I have disabled the GPS plugin that the iris drone uses. To ensure that fake_gps is working properly, I subscribed to ```/gazebo/model_states``` to get the exact position and orientation of the iris drone (which is very accurate as compared to data received from the IMU). I am using this data to publish to the ```/mavros/fake_gps/mocap/tf``` topic, and I am receiving the GPS data on the ```/mavros/global_position/global```, and everything works fine till here.

Now, when I try to fly the drone using ```OFFBOARD``` mode via the topic ```/mavros/setpoint_raw/local``` the drone just flies to a huge height, and I get barometer time out error on the console. The drone just flies to some point at a high altitude and crashes. I am unable to understand why this is happening. 

I know that the topic ```/mavros/fake_gps/mocap/tf``` is intended for motion capture systems, but they would essentially do the same right? i.e., give out the position and orientation of the drone to a good degree of accuracy.

Can someone please help me out with this issue? Thanks in advance.

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Melodic?
Ubuntu: ?18.04?

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: ?3.7.1?

### Node logs

```
... logging to /home/airl010/.ros/log/ad0726f4-7fbe-11eb-9a0f-b42e99b01d85/roslaunch-airl010-18803.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://airl010:39109/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
[... 509 lines cut ...]
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.100671"
      - 
        key: "Last remote time (s)"
        value: "59.804000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.096047217"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 5366 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 8, 140, 141, 147, 30, 31, 32, 36, 42, 74, 331, 76, 77, 83, 230, 105, 111, 241, 245

```

NishanthARao, 2021-03-24:
@TSC21 Could you please help me out in figuring this out? Thanks in advance.

DogukanAltay, 2021-05-14:
The same error occured for me too. Generally when I publish absolute ground truth(gazebo model state) UAV ossilates for a while and lose control and gives preflight check errors. It is either plugin broken or we are using it wrong.

NishanthARao, 2021-05-25:
@DogukanAltay I see. I haven't yet figured the issue. If you have, you could help me out by telling what you did. Otherwise, we might have to wait for the developer's answers.

@TSC21, It would be great to hear your comments on this.

Thanks in advance.

closed 2026-11-02 by github-actions[bot]