mavlink/mavros #1553: Multiple issues navigating to waypoints using mavros_msgs::globalpositiontarget

abhiramdapke, 2021-03-11:
Mavros: 1.5
ROS: Melodic
Ubuntu: 18.04

[ Y ] ArduPilot

Hi guys, 
I am trying to implement a GPS function by passing multiple waypoints to the drone using mavros and would like to implement it with a real drone with a companion computer. Currently, I am trying to accomplish it in the simulation space(Gazebo). I am using mavros_msgs::globalpositiontarget msg to pass the waypoints and sensor_msgs::NatSatFix to get the location of the drone.

The subscriber callback for getting the global position is -
```
void globalPosition_cb(const sensor_msgs::NavSatFix::ConstPtr& msg) {
    global_position = *msg;
    gps_position_received = true;
    ROS_INFO("Received global position: [%f, %f, %f]", global_position.latitude, global_position.longitude, global_position.altitude);
}
```
For this, the subscriber is - 
```
ros::Subscriber global_pos_sub = waypoint_node.subscribe<sensor_msgs::NavSatFix>("mavros/global_position/global", 10, globalPosition_cb);
```
The function for the waypoint is -
```
void set_waypoint(float lat, float lon, float alt)
{
	mavros_msgs::GlobalPositionTarget goal_position;
	goal_position.coordinate_frame = goal_position.FRAME_GLOBAL_TERRAIN_ALT;
	goal_position.type_mask = goal_position.IGNORE_VX | goal_position.IGNORE_VY | goal_position.IGNORE_VZ | goal_position.IGNORE_AFX | goal_position.IGNORE_AFY | goal_position.IGNORE_AFZ | goal_position.IGNORE_YAW | goal_position.IGNORE_YAW_RATE;
        goal_position.latitude = lat;
        goal_position.longitude = lon;
        goal_position.altitude = alt;
	global_pos_pub_gps.publish(goal_position);
}
```
For this, the Publisher is - 
```
ros::Publisher global_pos_pub_gps = waypoint_node.advertise<mavros_msgs::GlobalPositionTarget>("/mavros/setpoint_raw/global",10);
```

A snippet of the script that I am using is like this -
```
struct waypoint
{
	float lat; 
	float lon; 
	float alt; 
};
```
 ---------------------------
```
    ros::Rate rate(2.0);
    wait4connect();
    wait4start();
    

    takeoff(5);
    sleep(4);
    
    std::vector<waypoint> waypointlist;
    waypoint nextwp;
    nextwp.lat = 0;
    nextwp.lon = 0;
    nextwp.alt = 5;
    waypointlist.push_back(nextwp);

    nextwp.lat = -35.3613525;
    nextwp.lon = 149.1579742;
    nextwp.alt = 5;
    waypointlist.push_back(nextwp);

[... 328 lines cut ...]
  frame_id: ''
coordinate_frame: 11
type_mask: 3576
latitude: 0.0
longitude: 0.0
altitude: 5.0
velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
acceleration_or_force: 
  x: 0.0
  y: 0.0
  z: 0.0
yaw: 0.0
yaw_rate: 0.0
---

```
Image of rqt graph

[image]

I am working on these for the past few weeks and not yet able to accomplish a working GPS function so any insights or guidance would be highly appreciated. Please feel free to share a snippet of your code as that becomes easier to understand.
Thanks in advance!

burak-yildizoz, 2021-03-19:
Try using `mavros_msgs::Altitude::amsl` as home altitude.

``` c++
double latitude, longitude;
bool gps_received = false;
void get_global_pose (const sensor_msgs::NavSatFix::ConstPtr& msg) {
	latitude = msg->latitude;
	longitude = msg->longitude;
	gps_received = true;
}

double altitude;
void get_altitude(const mavros_msgs::Altitude::ConstPtr &msg)
{
    altitude = msg->amsl;
}
```

Set `home_alt` to `altitude` when `gps_received` becomes true, and publish to `setpoint_position/global` using `geographic_msgs::GeoPoseStamped`.

murphym18, 2021-03-19:
I've also been trying to control the drone with the same type of message, but with python. I've noticed issues too. But much of what I do is quite different.

1. I takeoff by putting the drone in takeoff mode (`AUTO.TAKEOFF`). 
1. I look at the `mavros/extended_state` and `mavros/global_position/rel_alt` messages. I don't move on until `landed_state == 2` and I've reached a high enough relative altitude (in my case I need to be at least 2 meters up).
2. I send 20 setpoint messages at a rate of 10 per second. Then I switch to offboard mode. I continue to send set point messages at the same rate after that.
3. I look at  `mavros/global_position/global` messages. I check to see how many meters away my target position is from my current position. And this is where I get stuck.

The drone flies to the right latitude and longitude, but it's altitude is wrong. Here are the details from my log message:
```
distance: 47.21 meters 

                  LATITUDE           LONGITUDE          ALTITUDE
current position: 47.3978429,        8.545869,          597.5515792976987
target position:  47.39784293752265, 8.545869057172045, 550.3417375283871
```

The altitude accounts for all the distance.

@burak-yildizoz I've also tried sending `GeoPoseStamped` messages and I get the same problem where the drone flies 47.2 meters too high.

By the way, I calculate the distance by converting both my current position and target position to X, Y, Z values in WGS84 coordinates (each component is in meters), sometimes this is called ECEF. Then I use the Pythagorean theorem to find the distance. Getting the X, Y, and Z values is tricky and I can share some python with you, if you'd like. But I use the tricks from [the n-vector page](https://www.navlab.net/nvector/) for the heavy lifting.

murphym18, 2021-03-19:
@burak-yildizoz I'm not sure what you mean by:

>Try using mavros_msgs::Altitude::amsl as home altitude.

and

>Set home_alt to altitude when gps_received becomes true

Does the drone's home altitude have something to do with how `GeoPoseStamped` messages are interpreted? When I run `rostopic echo mavros/home_position/home` Here's what I see
```
geo: 
  latitude: 47.3978435
  longitude: 8.5458706
  altitude: 535.3505773470537
position: 
  x: 19.823331832885742
  y: 10.315902709960938
  z: 0.02930159494280815
```

Should I change `position.z` to `535`?

burak-yildizoz, 2021-03-19:
@murphym18 I have pointed out this problem in [GPS altitude is different in subscriber and receiver](https://github.com/mavlink/mavros/issues/1414). It seems like [someone needs to update the documentation](https://github.com/mavlink/mavros/issues/1411). For now, issues are the documentation.

<details>
<summary>Full code</summary>
<p>
``` c++
#include <ros/ros.h>
#include <mavros_msgs/State.h>
#include <mavros_msgs/CommandBool.h>
#include <mavros_msgs/SetMode.h>
#include <sensor_msgs/NavSatFix.h>
#include <geographic_msgs/GeoPoseStamped.h>
#include <mavros_msgs/Altitude.h>

#define ALT 30  // fly altitude in meters

mavros_msgs::State current_state;
void state_cb(const mavros_msgs::State::ConstPtr& msg) {
    current_state = *msg;
}

double latitude, longitude;
bool gps_received = false;
void get_global_pose (const sensor_msgs::NavSatFix::ConstPtr& msg) {
    latitude = msg->latitude;
    longitude = msg->longitude;
    gps_received = true;
}

double altitude;
void get_altitude(const mavros_msgs::Altitude::ConstPtr &msg)
{
    altitude = msg->amsl;
}

int main(int argc, char **argv)
{
    ros::init(argc, argv, "offb_node");
    ros::NodeHandle nh;

    ros::Subscriber state_sub = nh.subscribe<mavros_msgs::State>
        ("mavros/state", 10, state_cb);
    ros::ServiceClient arming_client = nh.serviceClient<mavros_msgs::CommandBool>
        ("mavros/cmd/arming");
    ros::ServiceClient set_mode_client = nh.serviceClient<mavros_msgs::SetMode>
        ("mavros/set_mode");
    ros::Subscriber sub_global_pose = nh.subscribe<sensor_msgs::NavSatFix>
        ("mavros/global_position/global", 10, get_global_pose);
    ros::Subscriber sub_altitude = nh.subscribe<mavros_msgs::Altitude>
        ("mavros/altitude", 10, get_altitude);
    ros::Publisher pub_global = nh.advertise<geographic_msgs::GeoPoseStamped>
        ("mavros/setpoint_position/global", 10);

    //the setpoint publishing rate MUST be faster than 2Hz
    ros::Rate rate(20.0);

    // wait for FCU connection
    while(ros::ok() && !current_state.connected) {
        ros::spinOnce();
        rate.sleep();
    }
    ROS_INFO("Connected to PX4!");

    // wait for GPS connection
    while(ros::ok() && !gps_received) {
        ros::spinOnce();
        rate.sleep();
    }

    // home position
    const double home_lat = latitude;
    const double home_lon = longitude;
    const double home_alt = altitude;
    ROS_INFO_ONCE("GPS Received!\nlat:%f\tlon:%f\talt:%f", home_lat, home_lon, home_alt);

    geographic_msgs::GeoPoseStamped gsp;
    gsp.pose.position.latitude = home_lat;
    gsp.pose.position.longitude = home_lon;
    gsp.pose.position.altitude = home_alt;

    // send a few setpoints before starting
    for(int i = 100; ros::ok() && i > 0; --i) {
        gsp.header.stamp = ros::Time::now();
        pub_global.publish(gsp);
        ros::spinOnce();
        rate.sleep();
    }

    // simulation requirements
    mavros_msgs::SetMode offb_set_mode;
    offb_set_mode.request.custom_mode = "OFFBOARD";
    mavros_msgs::CommandBool arm_cmd;
    arm_cmd.request.value = true;

    // DO NOT FORGET TO CHANGE THIS BEFORE REAL FLIGHT
    const bool simulation = false;
    if (simulation) {
        if (current_state.mode != "OFFBOARD") {
            set_mode_client.call(offb_set_mode);
            if (offb_set_mode.response.mode_sent)
                ROS_INFO("Offboard enabled");
            else
                ROS_WARN("Offboard failed");
        }
        if (!current_state.armed) {
            arming_client.call(arm_cmd);
            if (arm_cmd.response.success)
                ROS_INFO("Vehicle armed");
            else
                ROS_WARN("Arm failed");
        }
    }

    while (ros::ok()) {
        gsp.pose.position.latitude = home_lat;
        gsp.pose.position.longitude = home_lon;
        gsp.pose.position.altitude = home_alt + ALT;

        gsp.header.stamp = ros::Time::now();
        pub_global.publish(gsp);

        ROS_INFO_THROTTLE(2, "\n"
            "altitude\t: %.2f\n"
            "target_alt\t: %.2f",
            altitude, gsp.pose.position.altitude
        );

        ros::spinOnce();
        rate.sleep();
    }

    return 0;
}
```
</p>
</details>

abhiramdapke, 2021-03-22:
Hi @burak-yildizoz,
Thank you so much for the quick response and the code. I have a couple of doubts though. I am trying to do the same in ardupilot by changing the mode to guided but the drone is not taking off to the desired altitude. I can do that with a takeoff client but have you implemented any logic for navigating it to multiple waypoints? Also, do we need to write a logic for checking if the waypoint is reached, or can we directly pass in waypoints one by one by using geographic_msgs::geoposestamped?

abhiramdapke, 2021-03-22:
Hi murphym18,
Thank you for your comment. I am also having similar issues with mavros/global_position/global topic. Are you trying to implement a multiple waypoint mission in the WGS84? Could you share the python code for checking the distance between the current position and target position? I am trying to check that with the waypoint_reached function using the Haversine formula but it's not working.

abhiramdapke, 2021-03-22:
> @murphym18 I have pointed out this problem in [GPS altitude is different in subscriber and receiver](https://github.com/mavlink/mavros/issues/1414). It seems like [someone needs to update the documentation](https://github.com/mavlink/mavros/issues/1411). For now, issues are the documentation.
> 
> Full code

@burak-yildizoz I checked out the #1414. Do you think mavros_msgs::GlobalPositiontarget has a problem or it can work if the altitude issues are solved by using another subscriber callback for the altitude like you did? Or using geographic_msgs::Geoposestamped is the only option?

murphym18, 2021-03-23:
> Hi murphym18,
> Thank you for your comment. I am also having similar issues with mavros/global_position/global topic. Are you trying to implement a multiple waypoint mission in the WGS84? Could you [...]

I can share the code we use to check the distance.  We created a math lib in python. [Here is the repo](https://github.com/DroneResponse/python-droneresponse-mathtools). To find the distance from this example:

```
                  LATITUDE           LONGITUDE          ALTITUDE
current position: 47.3978429,        8.545869,          597.5515792976987
target position:  47.39784293752265, 8.545869057172045, 550.3417375283871
```

Here is the python:
```python
from droneresponse_mathtools import Lla

current_pos = Lla(47.3978429,        8.545869,          597.5515792976987)
target_pos  = Lla(47.39784293752265, 8.545869057172045, 550.3417375283871)

distance = current_pos.distance(target_pos)
assert 47.21 - distance < 0.001
```

[The complete source code for the lib is here (don't worry it's not too long)](https://github.com/DroneResponse/python-droneresponse-mathtools/blob/78807233c8ef84d7ebf1ea34ea6ca4a1ea45c2f0/src/droneresponse_mathtools/__init__.py#L25). And here is the most relevant excerpt :

```python
    def distance(self, other):
        p1 = self.to_pvector().as_array()
        p2 = other.to_pvector().as_array()

        resid = p1 - p2
        resid_sq = resid ** 2
        resid_sum_sq = resid_sq.sum()
        dist = np.sqrt(resid_sum_sq)

        return dist
```

The code converts both points to p-vectors. You can think of a p-vector as an X, Y, and Z value that specifies a position relative to Earth (the "p" in p-vector is for position). Here are the details about the coordinate system:

- The origin of this coordinate system is at the center of mass of Earth.
- The X-axis goes through latitude 0, longitude 0,
- The Z-axis goes through the North pole (the axis of rotation)
- The Y-axis is derived using right-hand rule (it goes through latitude 0, longitude 90). 

The line of code that says `p1 - p2` finds the vector that points from `p1` to `p2`.  Then we find the magnitude of that vector by squaring all the elements, adding them up and finding the square root. If you've never seen this before, we're implementing the [euclidean distance](https://en.wikipedia.org/wiki/Euclidean_distance).

In case it's helpful, `current_pos.to_pvector()` creates something like a vec3 with these values:
```
(4277613.251663816, 642794.9997351523, 4672256.91285744)
```

And `target_pos.to_pvector()` creates something like a vec3 with these values:
```
(4277581.646276147, 642790.2547726352, 4672222.165857614)
```

The units are in meters.

If you calculated the euclidean distance by hand you'd find the points are about 47.2 meters apart.

If you want something similar in C++ [the n-vector page](https://www.navlab.net/nvector/) has a download link for a C++ lib. Given a latitude, longitude, and altitude, you can calculate a p-vector in C++ using the lib. You can use function 1 on the n-vector page to get an n-vector. Once you have the n-vector you can then calculate a p-vector using function 5. If you're not sure, you can see how it's done in the python code I linked. Or you can study the n-vector page if you want the complete details. 

>Are you trying to implement a multiple waypoint mission in the WGS84? 

I'm working on some code that has a drone search along a path of way-points until it finds someone, then it will follow the person. Another program actually generates the search path and sends it to us. So when we get the list of way points we start off by fly to each one. In that sense it's like we're flying a mission. But we need to use set points because at any time, we might need to change course.

I think I'm going to solve the altitude problem by undoing the ASML to ellipsoid height conversion found [here](https://github.com/mavlink/mavros/blob/48152f2d09e07de54423a7ec734d28cee2ca5cdd/mavros/src/plugins/global_position.cpp#L147) and [here](https://github.com/mavlink/mavros/blob/48152f2d09e07de54423a7ec734d28cee2ca5cdd/mavros/src/plugins/global_position.cpp#L232). I looked at how MAVROS calculates the ellipsoid height, and found the code [here](https://github.com/mavlink/mavros/blob/7a883b7ab6bcb6f2845c186998fa622d9ee1609c/mavros/include/mavros/mavros_uas.h#L250-L256). I've not decided how I'm going to do the EGM96 conversion yet. [This gist](https://gist.github.com/murphym18/872e9ed8b371acc9d34e49db27f3f881) looks like a promising approach. 

But I think I want to avoid getting my position from more than one message. I don't want to get my latitude and longitude from `mavros/global_position/global` while my altitude comes from another message. The reason is I can't be sure the messages will be in sync. I'm 99% sure it will be fine. But I don't want to piece together my position by looking at the time stamp of each message and combining the messages with the nearest time stamp. To do that well, I'd need to re-create much of the estimator code and that's a pretty big undertaking.

I think it would be better for everyone if MAVROS just added an `AMSL` field to the `mavros/global_position/global` message. I'm even considering forking the project just for this one tiny change that is essential for my use case (and from what I gather on GitHub, it looks like I'm not the only one who needs this).

Anyway, I hope that helps.

abhiramdapke, 2021-03-23:
Thanks a lot for the detailed explanation, @murphym18. I'll get back to you after reading, researching, and implementing the stuff you just mentioned. In the meantime, let me know if you figure out navigating to multiple waypoints. I'll also try to find a workaround.

I also think sensor_msgs::Natsatfix should provide all the 3 fields according to the drone's position in real-time as it seems quite noncoherent writing another subscriber callback just for the altitude.

IbonDLucas, 2021-07-07:
Hi everyone,

I am currently facing similar issues as described above. I am not able to send any right `/mavaros/setpoin_raw/target_global` I always get `lat=0, long=0 and alt=0` except at auto.takeoff, where I also get the altitude mismatch. 

Currently I tried sending `geographic_msgs::GeoPoseStamped` message through  `/mavaros/setpoin_position/global` and `mavros_msgs::GlobalPositionTarget` through `/mavaros/setpoin_raw/global` with same results... I have tried ROS Melodic with Ubuntu 18 and mavros/bionic 1.8.0-1bionic.20210505.151420 and ROS Noetic with Ubuntu 20 and mavros/focal,1.8.0-1focal.20210505.164222. With ARM64 and AMD64 architectures, in simulation and on flights... 

Is there any results from your side? 

Thanks in advance!

murphym18, 2021-07-07:
> Currently I tried sending `geographic_msgs::GeoPoseStamped` message through `/mavaros/setpoin_position/global` and `mavros_msgs::GlobalPositionTarget` through `/mavaros/setpoin_raw/global` with same results

I'm not sure this helps but I noticed a typo in `/mavaros/setpoin_position/global`. I recognize that it could be correct in your code but I thought I'd point that out just in case. 

In case it helps, the rest of this comment explains how we do what you're describing. We use ubuntu 20.04 and ROS Noetic.

We're able to use `geographic_msgs::GeoPoseStamped` messages with mavros. We publish those messages to `/mavros/setpoint_position/global`. Here are some excerpts:

### How we build `geographic_msgs::GeoPoseStamped` messages
```python
from geographic_msgs.msg import GeoPoseStamped
from tf.transformations import quaternion_from_euler

def _build_global_setpoint2(latitude, longitude, altitude, yaw=0.0):
    """Builds a message for the /mavros/setpoint_position/global topic
    """
    geo_pose_setpoint = GeoPoseStamped()
    geo_pose_setpoint.pose.position.latitude = latitude
    geo_pose_setpoint.pose.position.longitude = longitude
    geo_pose_setpoint.pose.position.altitude = altitude

    roll = 0.0
    pitch = 0.0
    q = quaternion_from_euler(roll, pitch, yaw)
    geo_pose_setpoint.pose.orientation.x = q[0]
    geo_pose_setpoint.pose.orientation.y = q[1]
    geo_pose_setpoint.pose.orientation.z = q[2]
    geo_pose_setpoint.pose.orientation.w = q[3]

    return geo_pose_setpoint
```
### How we publish the messages:
We publish those messages with `rospy` using this `Publisher`:
```python
import rospy

_setpoint_global_pub = rospy.Publisher(uav_name + "/mavros/setpoint_position/global", GeoPoseStamped, queue_size=1)

setpoint = _build_global_setpoint2(1, 2, 3) # just an example setpoint
_setpoint_global_pub.publish(setpoint)
```

We namespace mavros nodes by UAV name. But if you're flying one drone, your code might be simpler.

The way we repeatedly send setpoint messages is beyond the scope of this comment. But hopefully that helps.

vooon, 2021-07-07:
@murphym18 do not forget to fill `header.stamp` for any `*Stamped` message.

IbonDLucas, 2021-07-08:
@murphym18 thanks! it's a typo while I was writing here, my bad. I have it like that in my code -in C++, but is equivalent- and I am publishing at 10Hz rate. Is there a reason why you don't add the header of the GeoPoseStamped? Thanks again :)

murphym18, 2021-07-16:
> do not forget to fill header.stamp for any *Stamped message.

@vooon Thanks! I will make that change. 

> Is there a reason why you don't add the header of the GeoPoseStamped?

@IbonDLucas I think that was just an unintentional omission

murphym18, 2021-07-16:
@vooon Thanks again! I edited my earlier GitHub comment and I changed the code in my project. Turns out, this thread is showing up when I run web searches related to this topic. I figured it's better to replace the incorrect code with better code so that others will have an easier time finding examples. Thanks for the help.

closed 2026-11-02 by github-actions[bot]