mavlink/mavros #1560: Error building release 1.7.0 in ROS Noetic Ubuntu Focal

NoImportance, 2021-04-06:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I could not build the new MAVROS release 1.7.0 following the provided instructions in the readme.

### MAVROS version and platform

Mavros: Release 1.7.0 
ROS: Noetic (1.15.10)
Ubuntu: Focal 20.04

### Autopilot type and version

[  ] ArduPilot
[ ] PX4

Version: ?3.7.1?

### Build Log

```
In file included from /home/noimportance/catkin_ws/src/mavros/mavros/src/plugins/mission_protocol_base.cpp:17:
/home/noimportance/catkin_ws/src/mavros/mavros/include/mavros/mission_protocol_base.h: In function ‘double mavros::plugin::waypoint_encode_factor(const uint8_t&)’:
/home/noimportance/catkin_ws/src/mavros/mavros/include/mavros/mission_protocol_base.h:77:29: error: ‘BODY_FLU’ is not a member of ‘mavlink::common::MAV_FRAME’; did you mean ‘BODY_FRD’?
   77 |  case enum_value(MAV_FRAME::BODY_FLU):
      |                             ^~~~~~~~
      |                             BODY_FRD
/home/noimportance/catkin_ws/src/mavros/mavros/include/mavros/mission_protocol_base.h:78:29: error: ‘MOCAP_NED’ is not a member of ‘mavlink::common::MAV_FRAME’; did you mean ‘LOCAL_NED’?
   78 |  case enum_value(MAV_FRAME::MOCAP_NED):
      |                             ^~~~~~~~~
      |                             LOCAL_NED
/home/noimportance/catkin_ws/src/mavros/mavros/include/mavros/mission_protocol_base.h:79:29: error: ‘MOCAP_ENU’ is not a member of ‘mavlink::common::MAV_FRAME’; did you mean ‘LOCAL_ENU’?
   79 |  case enum_value(MAV_FRAME::MOCAP_ENU):
      |                             ^~~~~~~~~
      |                             LOCAL_ENU
/home/noimportance/catkin_ws/src/mavros/mavros/include/mavros/mission_protocol_base.h:80:29: error: ‘VISION_NED’ is not a member of ‘mavlink::common::MAV_FRAME’
   80 |  case enum_value(MAV_FRAME::VISION_NED):
      |                             ^~~~~~~~~~
/home/noimportance/catkin_ws/src/mavros/mavros/include/mavros/mission_protocol_base.h:81:29: error: ‘VISION_ENU’ is not a member of ‘mavlink::common::MAV_FRAME’
   81 |  case enum_value(MAV_FRAME::VISION_ENU):
      |                             ^~~~~~~~~~
/home/noimportance/catkin_ws/src/mavros/mavros/include/mavros/mission_protocol_base.h:82:29: error: ‘ESTIM_NED’ is not a member of ‘mavlink::common::MAV_FRAME’
   82 |  case enum_value(MAV_FRAME::ESTIM_NED):
      |                             ^~~~~~~~~
/home/noimportance/catkin_ws/src/mavros/mavros/include/mavros/mission_protocol_base.h:83:29: error: ‘ESTIM_ENU’ is not a member of ‘mavlink::common::MAV_FRAME’
   83 |  case enum_value(MAV_FRAME::ESTIM_ENU):
```

vooon, 2021-04-06:
Hello, hotfix 1.7.1 released. Problem were in outdated pymavlink.dialects.v20.common, which is used to generate some code.

https://github.com/ros/rosdistro/pull/28959
https://github.com/ros/rosdistro/pull/28965

NoImportance, 2021-04-07:
Thanks for the fast response, that sorted it out on my end.

closed 2021-04-07 by NoImportance