mavlink/mavros #1561: Unable to Issue Mavros mavcmd takeoff command (Pixhawk 4 mini, Ardupilot, Odroid companion computer, and Optitrack system)

FioreViola, 2021-04-06:
### Trying to fly autonomously using pixhawk 4 min FCU, ardupilot, companion computer (ODROID C4) and Optitrack mo cap system. Able to Arm but not able to issue take off command. 

I am currently an d undergrad working on robotics research and have become stuck and am hoping someone in this community could help point me in the right direction. I currently have a pixhawk4 mini flashed with ardupilot (apm flight stack). That is hooked up to a companion computer currently by the debugging usb port to usb port (I do have an FTDI adapter wired that I will use eventually). The companion computer is running ros and receives optitrack pose data. The steps I take to arm the drone are as follows:

1) Plug in drone battery and ODROID 

2) open up several terminals and source them ( source ~/catkin_ws/devel/setup.bash and source /opt/ros/melodic/setup.bash)

3) run mavproxy.py 

4) start roscore

5) run roslaunch mavros apm.launch

6) run rosrun mavros mavsys rate --all 10

7) run roslaunch vrpn_client_ros sample.launch server:=IP ADDRESS OF OPTITRACK

8) rostopic echo /vrpn_client_node/trackerID/pose (this confirms that I do get pose information)

9) remap topics rosrun topic_tools relay /vrpn_client_node/trackerID/pose /mavros/mocap/pose

10) rosrun mavros mavsafety arm (this works and motors spin then stop)

11) rosrun mavros mavsys mode -c GUIDED

12) rosrun mavros mavsafety arm (This fails now, sometimes it works sometimes it fails)

I am still very new and have read a lot of information but could you help me figure out how to get the drone to take off to about .5 or 1 m off the ground using ros (or via the terminal mavcmd takeoff) attached are the logs, and explain if I am missing any steps or crucial stuff. Thank you so much in advance for any help! 

### MAVROS version and platform

Mavros: ?0.18.ardupilot optitrack guided mode takeoff4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[ X] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs from roslaunch mavros apm.launch

```
... logging to /home/fioreviola/.ros/log/df2a9e5e-9715-11eb-80f4-dc1ba1f14c2b/roslaunch-fioreviola-25480.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://fioreviola:42225/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
[... 1027 lines cut ...]
#### data from echoing rostopic echo /vrpn_client_node/MiniDrone1/pose

---
header: 
  seq: 2054
  stamp: 
    secs: 1617740954
    nsecs: 194667032
  frame_id: "world"
pose: 
  position: 
    x: -0.287670522928
    y: 0.0826278924942
    z: 0.103268995881
  orientation: 
    x: 0.551197648048
    y: -0.0056214989163
    z: 0.794052064419
    w: -0.256185412407
---

#### the remap command line I am using to remap topics 

rosrun topic_tools relay /vrpn_client_node/MiniDrone1/pose /mavros/mocap/pose
[ INFO] [1617744393.977017240]: advertised as /mavros/mocap/pose

rob-clarke, 2021-04-13:
Looks like you need to do some configuration of ArduPilot first as the EKF is still expecting a GPS fix. The [ArduPilot docs](https://ardupilot.org/copter/docs/common-optitrack.html#configuration-the-drone) have a section on this. You should be able to fly the vehicle in Loiter mode based off the OptiTrack data if the vehicle is configured properly. The docs also have some text on how to test the setup before you move on to software-based control.

ZhongmouLi, 2024-04-08:
I think you need to set the EKF origin before flights.

closed 2024-04-08 by vooon