mavlink/mavros #1563: ROS2: port mavros scripts

vooon, 2021-04-12:
Need to port scripts to python3 and ros2.
Also would use single entry point.

akvavit01, 2021-05-07:
Hi, is there any progress in this task? I would like to contribute on porting MAVROS to ROS2

vooon, 2021-05-07:
Hi. I;m currently working on new python client library. Perhaps would get first scripts soon.

See: https://github.com/mavlink/mavros/tree/ros2/mavros/mavros

Also i found out that current param plugin doesn't work well - i have to redo service interface by myself and not rely on rclcpp provided one.

akvavit01, 2021-05-07:
Can I trouble you to explain about the param plugin? Is the one that's not working the parser?

akvavit01, 2021-05-07:
By the way, will there be any plan to port the code into c++?
As far as I know, c++ is better in terms of performance

vooon, 2021-05-11:
No, param is working, just having some bugs which i need to fix...

Mavros node is written in C++, for helper cli's like mavparam/mavwp performance is not an issue, so i do not see any reason to rewrite.

akvavit01, 2021-05-11:
Okay
Is there anything that can be done in parallel?

vooon, 2021-05-11:
Actually you could start porting of mavros_extras, that's entirely untouched ground.

akvavit01, 2021-05-11:
Okay, will do it after my laptop has beem repaired
Best of luck to you, sir

vooon, 2021-05-28:
Done.

closed 2021-05-28 by vooon