mavlink/mavros #1568: (maybe) Mavros hangs for some time during

dimaxano, 2021-04-20:
### Issue details

I noticed a strange pause in the publishing topic from different mavros nodes. [Here](https://ibb.co/pZHw0XB) is a screenshot of the recorder bag from `rqt_bag`.  After a pause, you may see, it sends all the messages that should be sent during the pause. Not sure where to look for the source of this behavior.

### MAVROS version and platform

Mavros: 1.7.0
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[ +] ArduPilot
[  ] PX4

Version: 4.0

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
header
	seq:  160
	stamp
		secs:  1618410233
		nsecs:  243612976
	frame_id: 
status
	[0]
		level:  0
		name: mavros: FCU connection
		message: connected
		hardware_id: /dev/ttyTHS2:921600
		values
			[0]
				key: Received packets:
				value: 63482
			[1]
				key: Dropped packets:
				value: 0
			[2]
				key: Buffer overruns:
				value: 0
			[3]
				key: Parse errors:
				value: 0
			[4]
				key: Rx sequence number:
				value: 113
			[5]
				key: Tx sequence number:
				value: 0
			[6]
				key: Rx total bytes:
				value: 2212919
			[7]
				key: Tx total bytes:
				value: 212141
			[8]
				key: Rx speed:
				value: 11987.000000
			[9]
				key: Tx speed:
				value: 1464.000000
	[1]
		level:  0
		name: mavros: GPS
		message: 3D fix
		hardware_id: /dev/ttyTHS2:921600
		values
			[0]
				key: Satellites visible
				value: 20
			[1]
				key: Fix type
				value: 4
			[2]
				key: EPH (m)
				value: 0.54
			[3]
				key: EPV (m)
				value: 0.81
	[2]
		level:  0
		name: mavros: Heartbeat
		message: Normal
		hardware_id: /dev/ttyTHS2:921600
		values
			[0]
				key: Heartbeats since startup
				value: 217
			[1]
				key: Frequency (Hz)
				value: 0.999980
			[2]
				key: Vehicle type
				value: Normal helicopter with tail rotor
			[3]
				key: Autopilot type
				value: ArduPilot
			[4]
				key: Mode
				value: CMODE(3)
			[5]
				key: System status
				value: Active
	[3]
		level:  2
		name: mavros: System
		message: Sensor health
		hardware_id: /dev/ttyTHS2:921600
		values
			[ 0]
				key: Sensor present
				value: 0x1321FDEF
			[ 1]
				key: Sensor enabled
				value: 0x1121FDEF
			[ 2]
				key: Sensor health
				value: 0x1131FDAF
			[ 3]
				key: 3D gyro
				value: Ok
			[ 4]
				key: 3D accelerometer
				value: Ok
			[ 5]
				key: 3D magnetometer
				value: Ok
			[ 6]
				key: absolute pressure
				value: Ok
			[ 7]
				key: GPS
				value: Ok
			[ 8]
				key: optical flow
				value: Fail
			[ 9]
				key: computer vision position
				value: Ok
			[10]
				key: laser based position
				value: Ok
			[11]
				key: 3D angular rate control
				value: Ok
			[12]
				key: attitude stabilization
				value: Ok
			[13]
				key: yaw position
				value: Ok
			[14]
				key: z/altitude control
				value: Ok
			[15]
				key: x/y position control
				value: Ok
			[16]
				key: motor outputs / control
				value: Ok
			[17]
				key: rc receiver
				value: Ok
			[18]
				key: AHRS subsystem health
				value: Ok
			[19]
				key: Logging
				value: Ok
			[20]
				key: CPU Load (%)
				value: 52.6
			[21]
				key: Drop rate (%)
				value: 0.0
			[22]
				key: Errors comm
				value: 0
			[23]
				key: Errors count #1
				value: 0
			[24]
				key: Errors count #2
				value: 0
			[25]
				key: Errors count #3
				value: 0
			[26]
				key: Errors count #4
				value: 0
	[4]
		level:  1
		name: mavros: Battery
		message: Low voltage
		hardware_id: /dev/ttyTHS2:921600
		values
			[0]
				key: Voltage
				value: 0.00
			[1]
				key: Current
				value: -0.0
			[2]
				key: Remaining
				value: -1.0
	[5]
		level:  0
		name: mavros: Time Sync
		message: Normal
		hardware_id: /dev/ttyTHS2:921600
		values
			[0]
				key: Timesyncs since startup
				value: 2140
			[1]
				key: Frequency (Hz)
				value: 9.642592
			[2]
				key: Last RTT (ms)
				value: 15.134688
			[3]
				key: Mean RTT (ms)
				value: 75.647564
			[4]
				key: Last remote time (s)
				value: 1330.041305000
			[5]
				key: Estimated time offset (s)
				value: 1618408902.619921446
```

closed 2021-05-02 by dimaxano

ashirsat, 2021-06-02:
I am facing the same or a similar issue. @dimaxano were you able to find a solution for this?

dimaxano, 2021-06-02:
Unfortunately, no. I didn't found a reproducible case yet: sometimes it repeats, sometimes not(

@ashirsat Can you please answer some questions, that may help with my investigations
1. What platform runs mavros in your case (OS, architecture)?
2. How does this platform connected to the Autopilot hardware (serial, USB cable, UDP\TCP)?
3. How do you perceive the issue? I mean did you try to record a bag and you see a picture similar to [this](https://ibb.co/pZHw0XB)? What is the lag time in your case?

ashirsat, 2021-06-02:
> Unfortunately, no. I didn't found a reproducible case yet: sometimes it repeats, sometimes not(
> 
@dimaxano 
[image]

> @ashirsat Can you please answer some questions, that may help with my investigations
> 
>     1. What platform runs mavros in your case (OS, architecture)?
I am using Jetson Nano running ubuntu18.04  mavros version 1.18.0
>     2. How does this platform connected to the Autopilot hardware (serial, USB cable, UDP\TCP)?
I connect it through serial */dev/ttyTHS1 at a baud rate of 921600
>     3. How do you perceive the issue? I mean did you try to record a bag and you see a picture similar to [this](https://ibb.co/pZHw0XB)? What is the lag time in your case?
I perceive this issue in the form of a mode change on the px4 side  from Offboard(equivalent to Guided in APM) to PosCtl. I havent recorded a rosbag because I cannot predict when it will happen. The one thing I did observe though was that the rostopic echo on any mavros node publisher topic also freezes for about 0.5 sec and then regains subscription.
like for example I am performing rostopic echo /mavros/local_position/pose.

So to give a visual representation of the stage till I have been able to root cause this is as follows:
(See the image attached for a visual [Flow](https://user-images.githubusercontent.com/6137894/120454330-a3e9c400-c348-11eb-9ca9-68689a1c3228.png)
So initially after mavros node inits I am able to set it into offboard mode by sending cmd_vel_setpoints and the corresponding message is displayed on QGC 

But sometimes this pipeline experiences an error ( I am inclining more towards an issue with mavros_node) for at least 0.5 sec due to which the offboard is rejected and it transitions to fail-safe mode for a 1s (minimum) then re-enters offboard mode. This erroneous behavior happens in a non deterministic fashion i.e. it can occur once or many times during a run but is also not repeatable to the same extent on every test run or it doesn't occur at all for some runs.

One thing i did observe is that the subscriber always hangs or freezes when the offboard mode fail-safe is triggered and   
So the flow is mavros_node starts publishing normally as desired.

Please let me know any further suggestions to debug.

Thanks

dimaxano, 2021-06-02:
Thank you very much for your feedback!
I am also using Jetson (but TX2, with this board) and connect it to the pixhawk using serial on 921600.
Do you use Jetson Nano with dev board or with some other carrier board?

My first assumption is that the cause of the problem is somewhere in the L4T internals. Need to search for `uart`, `serial` or some other keyword in [this](https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/70) branch of Nvidia forum, maybe someone already faced a similar problem.
If no success, I will create a post on that forum

ashirsat, 2021-06-02:
@dimaxano 
> Thank you very much for your feedback!
> I am also using Jetson (but TX2, with this board) and connect it to the pixhawk using serial on 921600.
> Do you use Jetson Nano with dev board or with some other carrier board?
Yes I am using jetson nano as a companion board along with pixhawk.
Yes I am using jetson nano as a companion computer
 
> My first assumption is that the cause of the problem is somewhere in the L4T internals. Need to search for `uart`, `serial` or some other keyword in [...]
Can you state why do you think it is an issue with the L4T?  Yeah I did try to find a similar issue but did not get a result in that forum.  
> If no success, I will create a post on that forum
Please do and let me know if you find any possible solutions.