mavlink/mavros #1572: Reporting progress in response result value for long running commands

jkeshav-bvignesh, 2021-04-28:
Hi,

I have noticed this with multiple long running commands but let's take one command:  `/mavros/cmd/takeoff `

After the necessary mode and arm checks, I call the service. The autopilot (ArduCopter SITL) accepts it and the drone takes off. 

The response I get is:
```
success: True
result: 0
```
But shouldn't the `result` value be 5 (`MAV_RESULT_IN_PROGRESS`) instead of 0(`MAV_RESULT_ACCEPTED`) according to the [Mavlink Long Running Commands Protocol](https://mavlink.io/en/services/command.html#long_running_commands) or am I missing something?

If this is the expected behavior, then what is the correct way to check if the command has been completed successfully?
Calling the service again after the drone has successfully reached the altitude gives the below response which doesn't seem right:
```
success: True
result: 4
```

### MAVROS version and platform

Mavros: 1.6.0-1bionic.20210303.153807
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[0] ArduPilot
[  ] PX4

Version: ArduCopter - V4.1.0-dev

P.S. I can't currently test the Px4 SITL to see it's response. Therefore I am not sure if the issue is with Ardupilot SITL or mavros.

vooon, 2021-04-28:
I haven't seen that protocol to be implemented anywhere. Perhaps PX4 supports it now, but not at the moment when i done command plugin.
Don't checked AP's sources for a while...

jkeshav-bvignesh, 2021-04-28:
Oh I see.

The problem with the current implementation is that we will have to use a combination of sleep/wait commands and checks to decide if the command has completed successfully.

Since the command returns as soon as the autopilot accepts it, I don't see any other way to track the completion of the command.

Is that the right way?

closed 2026-11-02 by github-actions[bot]