mavlink/mavros #1575: Desired trajectory is not publishing in HITL mode

eldewany, 2021-05-18:
----

### Issue details

The desired trajectory which is published on the topic /mavros/trajectory/desired is publishing fine when using SITL but whenever I switch to HITL the topic doesn't publish any data and I can't trace why is this happening.

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Melodic
Ubuntu: ?18.04?

### Autopilot type and version

[  ] ArduPilot
[ . ] PX4

Version: ?1.11.0?

### Node logs

```
[ INFO] [1621372650.390428359]: FCU URL: udp://:14540@localhost:14557
[ INFO] [1621372650.392848160]: udp0: Bind address: 0.0.0.0:14540
[ INFO] [1621372650.393001793]: udp0: Remote address: 127.0.0.1:14557
[ INFO] [1621372650.393253510]: GCS bridge disabled
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
[ INFO] [1621372650.404324082]: Plugin 3dr_radio loaded
[ INFO] [1621372650.406934718]: Plugin 3dr_radio initialized
[ INFO] [1621372650.407213682]: Plugin actuator_control loaded
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
[ INFO] [1621372650.411216994]: Plugin actuator_control initialized
[ INFO] [1621372650.415247408]: Plugin adsb loaded
[ INFO] [1621372650.420089011]: Plugin adsb initialized
[ INFO] [1621372650.420408415]: Plugin altitude loaded
[ INFO] [1621372650.422571272]: Plugin altitude initialized
[ INFO] [1621372650.422828417]: Plugin cam_imu_sync loaded
[ INFO] [1621372650.423743965]: Plugin cam_imu_sync initialized
[ INFO] [1621372650.423945475]: Plugin command loaded
[ INFO] [1621372650.431803733]: Plugin command initialized
[ INFO] [1621372650.431975963]: Plugin companion_process_status loaded
[ INFO] [1621372650.434479879]: Plugin companion_process_status initialized
[ INFO] [1621372650.434596156]: Plugin debug_value loaded
[ INFO] [1621372650.440414865]: Plugin debug_value initialized
[ INFO] [1621372650.440458109]: Plugin distance_sensor blacklisted
[ INFO] [1621372650.440618913]: Plugin esc_status loaded
[ INFO] [1621372650.442221007]: Plugin esc_status initialized
[ INFO] [1621372650.442483874]: Plugin fake_gps loaded
[ INFO] [1621372650.453680677]: Plugin fake_gps initialized
[ INFO] [1621372650.453878922]: Plugin ftp loaded
[ INFO] [1621372650.463168157]: Plugin ftp initialized
[ INFO] [1621372650.463363717]: Plugin geofence loaded
[ INFO] [1621372650.467162422]: Plugin geofence initialized
[ INFO] [1621372650.467320375]: Plugin global_position loaded
[ INFO] [1621372650.482993911]: Plugin global_position initialized
[ INFO] [1621372650.483149641]: Plugin gps_rtk loaded
[ INFO] [1621372650.485365976]: Plugin gps_rtk initialized
[ INFO] [1621372650.485548650]: Plugin gps_status loaded
[ INFO] [1621372650.487442374]: Plugin gps_status initialized
[ INFO] [1621372650.487663877]: Plugin hil loaded
[ INFO] [1621372650.496255287]: Plugin hil initialized
[ INFO] [1621372650.496415504]: Plugin home_position loaded
[ INFO] [1621372650.500344729]: Plugin home_position initialized
[ INFO] [1621372650.500576595]: Plugin imu loaded
[ INFO] [1621372650.505248700]: Plugin imu initialized
[ INFO] [1621372650.505386629]: Plugin landing_target loaded
[ INFO] [1621372650.514345219]: Plugin landing_target initialized
[ INFO] [1621372650.514520317]: Plugin local_position loaded
[ INFO] [1621372650.518242318]: Plugin local_position initialized
[ INFO] [1621372650.518446354]: Plugin log_transfer loaded
[... 223 lines cut ...]
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Voltage"
        value: "65.54"
      - 
        key: "Current"
        value: "-0.0"
      - 
        key: "Remaining"
        value: "-1.0"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 5582 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 24, 132, 26, 30, 31, 32, 36, 42, 74, 331, 83, 230, 105, 111, 241, 242, 245
```

eldewany, 2021-05-18:
@TSC21

eldewany, 2021-05-23:
@vooon any help?

vooon, 2021-05-24:
Do you sure that you're using right topic name?
https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_trajectory.cpp#L41 -> `~/setpoint_trajectory/desired`.

eldewany, 2021-05-25:
Solved. The serial interface isn't delivering the full range of messages, use another way f connecting mavros like a serial or something.

closed 2021-05-25 by eldewany