mavlink/mavros #1577: How can I edit the set_point.cpp file such that I can change velocity along with the global position?

mariofaragalla, 2021-05-24:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I am working on a drone project I’m connecting ros to gazebo and using the mavros node of course. I would like to publish position messages to /mavros/setpoint_position/global topic and at the same time I’m trying to adjust the velocity with which the drone goes to the intended position using the /mavros/setpoint_velocity/cmd_vel, but it doesn’t work together. I think the problem is in the setpoint_position.cpp in this method:

void setpointg_cb(const geographic_msgs::GeoPoseStamped::ConstPtr &req)
	{
		using mavlink::common::POSITION_TARGET_TYPEMASK;

		uint16_t type_mask = uint16_t(POSITION_TARGET_TYPEMASK::VX_IGNORE)
					| uint16_t(POSITION_TARGET_TYPEMASK::VY_IGNORE)
					| uint16_t(POSITION_TARGET_TYPEMASK::VZ_IGNORE)
					| uint16_t(POSITION_TARGET_TYPEMASK::AX_IGNORE)
					| uint16_t(POSITION_TARGET_TYPEMASK::AY_IGNORE)
					| uint16_t(POSITION_TARGET_TYPEMASK::AZ_IGNORE);

		Eigen::Quaterniond attitude;
		tf::quaternionMsgToEigen(req->pose.orientation, attitude);
		Eigen::Quaterniond q = ftf::transform_orientation_enu_ned(
						ftf::transform_orientation_baselink_aircraft(attitude));

		set_position_target_global_int(
					req->header.stamp.toNSec() / 1000000,
					uint8_t(MAV_FRAME::GLOBAL_INT),
					type_mask,
					req->pose.position.latitude * 1e7,
					req->pose.position.longitude * 1e7,
					req->pose.position.altitude,
					Eigen::Vector3d::Zero(),
					Eigen::Vector3d::Zero(),
					ftf::quaternion_get_yaw(q),
					0);
}

As I believe that this bit masking ignores the velocity change effects.

Thank you in advance.

### MAVROS version and platform

Mavros: ?1.0.0?
ROS: ?noetic?
Ubuntu: ?20.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

vooon, 2021-05-29:
As i remember you cannot control at the same time position and velocity (or attitude?), in any case i suppose easier to use setpoint_raw for prototyping.
Then check your autopilot code to see which type masks it supports.

mariofaragalla, 2021-05-29:
What do you mean by prototyping? I tried to publish the desired position and velocity to the `mavros/setpoint_raw/global` topic once from my ros node and another time using `rostopic pub -r 50` but I didn't see any change with the drone in the gazebo simulation.

vooon, 2021-05-31:
Did you update header.stamp? Update stream also should be frequent, else OFFBOARD would disable (or don't enable).

By prototyping i mean that it's easier to use existing plugin than do new one gust to try different type masks.

closed 2026-11-01 by github-actions[bot]