mavlink/mavros #1582: ROS2 installation instructions

quentingllmt, 2021-06-03:
### Issue details

How can I test the mavros ROS2 port ?
Please provide installation instructions

### MAVROS version and platform

Mavros: 
ROS: Dashing / Foxy
Ubuntu: 18.04 / 20.04

vooon, 2021-06-03:
Hello,
Yes, we need to update that instruction: https://github.com/mavlink/mavros/blob/master/mavros/README.md#installation

But actually it's must be quite close, just instead `catkin` tool you need to use `colcon`.

Workspace should look like that:
```
ws
└── src
    ├── mavlink
    └── mavros
```

(so you still need to clone mavlink release and mavros).

Then you should use `rosdep install --from-paths src --ignore-src -y` to install dependencies.
Then use `colcon build`, which would create log, build and install dirs in your ws.
After build you should do `source ws/install/setup.bash` to update lookup path.

Then you could use mavros_node to run router and uas nodes, or you can load them into component node container.
Here is example:
```
# mavros_param.yaml
mavros:
  ros__parameters: {}

mavros_router:
  ros__parameters: {}

mavros_node:
  ros__parameters:
    fcu_url: udp://:14540@
    gcs_url: tcp-l://:5670
```
```
ros2 run mavros mavros_node --ros-args --params-file ws/mavros_param.yaml
```
```
ros2 run mavros mav --help
```

quentingllmt, 2021-06-03:
Hello vooon, thank you !

I'll try and give you some feedback.

Thanks for your work

vooon, 2021-06-03:
From architectural point of view mavlink protocol parser and message plugins were separated to different nodes:
1. MAVROS Router, `mavros::router::Router`, which makes it possible to add/remove connections on the fly.
2. MAVROS UAS, `mavros::uas::UAS`, a plugin container which loads all protocol plugins and manage UAS Link to the Router.

Router
--------
Makes possible to connect several drones and then provide single link to GCS. Or for example provide connections for several GCS'es.
It's not as advanced as `mavlink-router`, but should reduce headaches with multidrone setups.

Basically it have 3 link types:
1. FCU Link, a MAVLink connection provided by libmavconn
2. GCS Link, similar to FCU one
3. UAS Link, that special kind of endpoint translates message streams to ROS2 mavros_msgs::msg::Mavlink (similar to ROS1 `/mavlink/to|from` topics

Routing rule is dead simple:
1. retranslate message only to different link type, so crosstalk between FCU's not possible
2. if message have target ids, then retranslate to link(s) known to send some messages with that ids.
3. if there are no links for 2 then retranslate message to all links (broadcast)
4. messages without target would be retranslated to all other links (broadcast)

UAS Plugin
---------------
Is very close to ROS1 variant, but now each plugin is separate Node.

New thing introduced here - Filters, that is helper class that filter messages coming to handler functor.
I'm still not quite sure about design decision which do not allow make them plugin-state-dependant.
But for much of places it's enough to gert rid of source ids checks.

vooon, 2021-06-03:
Ah, also i'm not going to support anything older than Foxy. As i remember they all already EOLed.
For Galactic i need to fix some compilation issues and rewrite pram plugin (in any case it's not reliable in present state).

quentingllmt, 2021-06-03:
Yes I understand, I just think that supporting ROS2 Dashing might be a good idea (at least for a little bit of time) as it is the last ROS2 version intended to be used on Ubuntu bionic (18.04). Foxy and Galactic are made for focal (20.04). Supporting Ubuntu 18.04 might be nice as it is sometimes the last Ubuntu release supported by small embedded boards (like Nvidia jetson family (with cuda support), who plans Ubuntu focal support with Jetpack 5.0, planned for 2H-2022). Those boards are where mavros might be used a lot?. This would save users from having to install ROS2 from source if they want to use mavros with ROS2 on ubuntu 18.04. 
Anyway, nice work !

karishmapatnaik, 2021-06-23:
Hi @vooon @quentingllmt 
I am trying to install mavros for ROS2. My version is ROS dashing (due to the same reasons @quentingllmt  mentioned above). However when I try to run the following command:
$ rosdep install --from-paths src --ignore-src --rosdistro dashing -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 libopensplice69 rti-connext-dds-5.3.1 urdfdom_headers" I get the following:
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
mavros: Cannot locate rosdep definition for [mavlink]
libmavconn: Cannot locate rosdep definition for [mavlink]
px4: Cannot locate rosdep definition for [mav_msgs]
I have the master version of mavlink and ros2 version of mavros. 
How do I resolve it? Any help is appreciated, I have been stuck on this for a couple of days now :|

vooon, 2021-06-24:
Do that in your ws:
```
cd src
git clone https://github.com/mavlink/mavlink-gbp-release.git --branch release/foxy/mavlink mavlink
```

karishmapatnaik, 2021-06-24:
Hi @vooon 
Thanks, I tried that and now when I run the rosdep it says:
px4: Cannot locate rosdep definition for [mav_msgs]. 
I cloned the default version for mavros by:
`$ git clone https://github.com/mavlink/mavros.git --recursive ~/colcon_ws/src/mavros`
Is this the correct command, thanks again.

vooon, 2021-06-25:
As far as i can tell, now it's px4 dependency. As i see, it needs http://wiki.ros.org/mav_msgs, which do not have ROS2 build support.
Try to (temporary) remove px4 from that workspace, leave just `mavlink` and `mavros`.
Then ask PX4 team is it possible to run simulator in ros2 or not.

karishmapatnaik, 2021-06-25:
Hi Voon, 
Thank you, I could make some progress. It's worthy to note that we need to have an updated version of CMake too. Now, rosdep says all dependencies are installed. 
However, here is a new error:
`‘RCLCPP_ERROR_STREAM’ was not declared in this scope
   RCLCPP_ERROR_STREAM(
   ^~~~~~~~~~~~~~~~~~~
/home/kpatnaik/ros2_ws/src/mavros/mavros/src/lib/enum_to_string.cpp:791:3: note: suggested alternative: ‘RCLCPP_ERROR_ONCE’
   RCLCPP_ERROR_STREAM(
   ^~~~~~~~~~~~~~~~~~~
`
Any idea how to fix this, thank you!

vooon, 2021-06-26:
Hmm, Dashing rclcpp/logging.hpp do not have _STREAM and _THROTTLE variants, which is widely used in mavros.
Maybe it's worth to build Galactic from source?

karishmapatnaik, 2021-06-27:
Ah I see, thank you! I will try Galactic then.

karishmapatnaik, 2021-06-30:
I can verify that everything works perfectly fine with following command:
`ros2 run mavros mavros_node`
Ubuntu: 20.04 
ROS2: Foxy
@vooon Thank you!

dayjaby, 2021-07-06:
@vooon any idea how to let the mavros_extras plugins run?

Tried this without success:
```
mavros:
  ros__parameters:
    plugin_denylist: ["safety_area", "image_pub", "vibration", "rangefinder", "wheel_odometry"]
    plugin_allowlist: ["mount_control"]
mavros_node:
  ros__parameters:
    fcu_url: /dev/ttyACM0:2000000
```

vooon, 2021-07-06:
@dayjaby none of extras being ported yet.

Peter010103, 2021-08-24:
@vooon So does this mean I can't add the `mocap_pose_estimate` or the `vision_pose_estimate` plugins from mavros_extras?

vooon, 2021-08-24:
@Peter010103 do you sure, that you need ROS **2**? 

By now only few extras plugins have been ported. See: https://github.com/mavlink/mavros/blob/ros2/mavros_extras/mavros_plugins.xml

Peter010103, 2021-08-24:
@voon Yh I definitely need ROS2 as I would like to integrate our lab's previous work which uses ROS2 foxy. I've made a pull request with the some changes I've made to accomodate for the `vision_pose_estimate` plugin, which we'll use for sending motion capture data.

I followed what @dayjaby has done with the `landing_target` plugin. So far I can see the topic listed with `ros2 topic list`, but I get 2 warnings during compilation (colcon build) pertaining to `mavros::ftf::covariance_urt_to_mavlink` in line 197 of `mavros_extras/src/plugins/vision_pose_estimate`. I would really love it if the devs can have a look and give comments :)

vooon, 2021-08-25:
@aa-calvo wstool is not used anymore. For ros2 you should use `vcs` tool.
See: https://github.com/mavlink/mavros/blob/ros2/mavros/README.md#source-installation

vooon, 2021-08-28:
@Mark-Beaty if disconnection warning only happening once after mavros connection, i suppose you can ignore it as false positive.
If it appears again, then it might be a sign of communication problem.

Most of config options in px4.launch is close to defaults, so it should be ok to run without them set.
Please note that ros2 params file is different from old one, especially that params now has flat names.
Also note that new params could bee reconfigured in runtime (like old dynamic parameters).

vooon, 2021-08-28:
Regarding checkid, could you please check `ros2 topic list` && `ros2 topic echo /uas1/mavlink_source`?

Mark-Beaty, 2021-08-30:
@vooon Thanks for the quick response, its good to hear that everything should be fairly closely configured by default.  It might be worth noting that when I have QGroundControl running alongside the simulation I get the mission computer disconnect error echoed back to the mavros_node output as well as in QGroundControl, whereas without QGroundControl I didn't see any indication of the disconnect on the node output.

Here is the output of those commands:
`ros2 topic list`:

> /diagnostics
> /mavros/actuator_control
> /mavros/altitude
> /mavros/battery
> /mavros/estimator_status
> /mavros/extended_state
> /mavros/geofence/fences
> /mavros/global_position/compass_hdg
> /mavros/global_position/global
> /mavros/global_position/gp_lp_offset
> /mavros/global_position/gp_origin
> /mavros/global_position/local
> /mavros/global_position/raw/fix
> /mavros/global_position/raw/gps_vel
> /mavros/global_position/raw/satellites
> /mavros/global_position/rel_alt
> /mavros/global_position/set_gp_origin
> /mavros/home_position/home
> /mavros/home_position/set
> /mavros/imu/data
> /mavros/imu/data_raw
> /mavros/imu/diff_pressure
> /mavros/imu/mag
> /mavros/imu/static_pressure
> /mavros/imu/temperature_baro
> /mavros/imu/temperature_imu
> /mavros/local_position/accel
> /mavros/local_position/odom
> /mavros/local_position/pose
> /mavros/local_position/pose_cov
> /mavros/local_position/velocity_body
> /mavros/local_position/velocity_body_cov
> /mavros/local_position/velocity_local
> /mavros/manual_control/control
> /mavros/manual_control/send
> /mavros/mission/reached
> /mavros/mission/waypoints
> /mavros/param/event
> /mavros/rallypoint/rallypoints
> /mavros/rc/in
> /mavros/rc/out
> /mavros/rc/override
> /mavros/setpoint_accel/accel
> /mavros/setpoint_attitude/cmd_vel
> /mavros/setpoint_attitude/thrust
> /mavros/setpoint_position/global
> /mavros/setpoint_position/global_to_local
> /mavros/setpoint_position/local
> /mavros/setpoint_raw/attitude
> /mavros/setpoint_raw/global
> /mavros/setpoint_raw/local
> /mavros/setpoint_raw/target_attitude
> /mavros/setpoint_raw/target_global
> /mavros/setpoint_raw/target_local
> /mavros/setpoint_trajectory/desired
> /mavros/setpoint_trajectory/local
> /mavros/setpoint_velocity/cmd_vel
> /mavros/setpoint_velocity/cmd_vel_unstamped
> /mavros/state
> /mavros/statustext/recv
> /mavros/statustext/send
> /mavros/target_actuator_control
> /mavros/time_reference
> /mavros/timesync_status
> /mavros/wind_estimation
> /parameter_events
> /rosout
> /tf
> /tf_static
> /uas1/mavlink_sink
> /uas1/mavlink_source

`ros2 topic echo /uas1/mavlink_source` --This topic is spitting out data super fast, here is a sample of 2 messages, let me know if you need something specific/more
> header:
>   stamp:
>     sec: 1630343583
>     nanosec: 548014400
>   frame_id: ep:1000
> framing_status: 1
> magic: 253
> len: 28
> incompat_flags: 0
> compat_flags: 0
> seq: 8
> sysid: 1
> compid: 1
> msgid: 30
> checksum: 54712
> payload64:
> - 13447375164476031728
> - 13387070856780902823
> - 13466833860560430501
> - 4290600779823017929
> signature: []
> 
> header:
>   stamp:
>     sec: 1630343583
>     nanosec: 548063100
>   frame_id: ep:1000
> framing_status: 1
> magic: 253
> len: 28
> incompat_flags: 0
> compat_flags: 0
> seq: 9
> sysid: 1
> compid: 1
> msgid: 33
> checksum: 18211
> payload64:
> - 2035717509352588012
> - 2095995665511490
> - 18446744069414584322
> - 4290600781202915328
> signature: []
>

closed 2021-11-04 by vooon

standmit, 2022-02-18:
> New thing introduced here - Filters, that is helper class that filter messages coming to handler functor.

Hello, @vooon ! Could you tell me where can I find more information about the filters?

vooon, 2022-02-18:
@standmit https://github.com/mavlink/mavros/blob/ros2/mavros/include/mavros/plugin_filter.hpp

vooon, 2022-02-18:
@standmit also it's possible to make custom ones:

https://github.com/mavlink/mavros/blob/c035b8ff1ffb042db235648371f5d7d2e531fae4/mavros_extras/src/plugins/3dr_radio.cpp#L31-L46

vooon, 2022-07-07:
@Lemvos perhaps sudo drops path to ros binaries. You may try do that:
```
sudo src/mavros/mavros/scripts/install_geographiclib_datasets.sh
```

Lemvos, 2022-07-08:
Still getting the same error. I've also tried from source and this also doesn't seem to work for me. 
When building from source it fails here:
`vcs import src < /tmp/mavros.repos` 

responding with:
`-bash: /tmp/mavros.repos: No such file or directory`

From binary I also tried the "Alternative":
`wget https://raw.githubusercontent.com/mavlink/mavros/ros2/mavros/scripts/install_geographiclib_datasets.sh
./install_geographiclib_datasets.sh
`
Which just returns 
`sudo: ./install_geographiclib_datasets.sh: command not found
`
Any suggestion would be appreciated

vooon, 2022-07-08:
@Lemvos in case of wget, don't forget to add eXecutable flag, `chmod +x <file>`, or do `sudo bash <file>.sh`.

ZhongmouLi, 2023-11-12:
Hi, @vooon .   Thanks so much for your help in this thread.

My mavros2 cannot connect to UAV in simulation, could you please help me find the reason?

Env:
- Ubuntu 20.04
- ros2 foxy

Autopilot
- [ X] Arducopter 4.3
- [ ] PX4

I followed the installation steps at https://github.com/mavlink/mavros/blob/ros2/mavros/README.md.
1. `sudo apt install ros-foxy-mavros`
2.```
 wget https://raw.githubusercontent.com/mavlink/mavros/ros2/mavros/scripts/install_geographiclib_datasets.sh
./install_geographiclib_datasets.sh
```

But when I run` ros2 run mavros mavros_node --ros-args --params-file params.yaml`,  and I got an error 
`[ERROR] [1699753727.477090516] [rcl]: Failed to parse global arguments
terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
  what():  failed to initialize rcl: Couldn't parse params file: '--params-file params.yaml'. Error: Error opening YAML file, at /tmp/binarydeb/ros-foxy-rcl-yaml-param-parser-1.1.14/src/parser.c:198, at /tmp/binarydeb/ros-foxy-rcl-1.1.14/src/rcl/arguments.c:388`
  
If i run `os2 run mavros mavros_node --ros-args --param fcu_url:=udp://:14655@` and I get 

```
[INFO] [1699754012.139540837] [mavros_node]: Starting mavros_node container
[INFO] [1699754012.139609112] [mavros_node]: FCU URL: udp://:14655@
[INFO] [1699754012.139617780] [mavros_node]: GCS URL: 
[INFO] [1699754012.139623646] [mavros_node]: UAS Prefix: /uas1
[INFO] [1699754012.139628871] [mavros_node]: Starting mavros router node
[INFO] [1699754012.141625239] [mavros_router]: Built-in SIMD instructions: SSE, SSE2
[INFO] [1699754012.141641452] [mavros_router]: Built-in MAVLink package version: 2022.12.30
[INFO] [1699754012.141647647] [mavros_router]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all cubepilot development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[INFO] [1699754012.141653288] [mavros_router]: MAVROS Router started
[INFO] [1699754012.141671563] [mavros_router]: Requested to add endpoint: type: 0, url: udp://:14655@
[INFO] [1699754012.141696038] [mavros_router]: Endpoint link[1000] created
[INFO] [1699754012.141938889] [mavros_router]: link[1000] opened successfully
[INFO] [1699754012.141965866] [mavros_router]: Requested to add endpoint: type: 2, url: /uas1
[INFO] [1699754012.141979486] [mavros_router]: Endpoint link[1001] created
[INFO] [1699754012.142270496] [mavros_router]: link[1001] opened successfully
[INFO] [1699754012.142290667] [mavros_node]: Starting mavros uas node
[INFO] [1699754012.169408037] [mavros]: UAS Executor started, threads: 12
[INFO] [1699754012.180261623] [mavros]: Plugin actuator_control created
[INFO] [1699754012.180310005] [mavros]: Plugin actuator_control initialized
[INFO] [1699754012.181993762] [mavros]: Plugin altitude created
[INFO] [1699754012.182018219] [mavros]: Plugin altitude initialized
[INFO] [1699754012.185188117] [mavros]: Plugin command created
[INFO] [1699754012.185213167] [mavros]: Plugin command initialized
[INFO] [1699754012.189163610] [mavros]: Plugin ftp created
[INFO] [1699754012.189188322] [mavros]: Plugin ftp initialized
[INFO] [1699754012.191743154] [mavros]: Plugin geofence created
[INFO] [1699754012.191779229] [mavros]: Plugin geofence initialized
[INFO] [1699754012.195583868] [mavros]: Plugin global_position created
[INFO] [1699754012.195616051] [mavros]: Plugin global_position initialized
[INFO] [1699754012.198076873] [mavros]: Plugin home_position created
[INFO] [1699754012.198101286] [mavros]: Plugin home_position initialized
[INFO] [1699754012.201212290] [mavros]: Plugin imu created
[INFO] [1699754012.201257280] [mavros]: Plugin imu initialized
[INFO] [1699754012.204472688] [mavros]: Plugin local_position created
[INFO] [1699754012.204510853] [mavros]: Plugin local_position initialized
[INFO] [1699754012.207048295] [mavros]: Plugin manual_control created
[INFO] [1699754012.207074469] [mavros]: Plugin manual_control initialized
[INFO] [1699754012.209441744] [mavros]: Plugin nav_controller_output created
[INFO] [1699754012.209463463] [mavros]: Plugin nav_controller_output initialized
[INFO] [1699754012.212382947] [mavros]: Plugin param created
[INFO] [1699754012.212405975] [mavros]: Plugin param initialized
[INFO] [1699754012.215480343] [mavros]: Plugin rallypoint created
[INFO] [1699754012.215515061] [mavros]: Plugin rallypoint initialized
[... 11 lines cut ...]
[INFO] [1699754012.236238668] [mavros]: Plugin setpoint_trajectory initialized
[INFO] [1699754012.239873108] [mavros]: Plugin setpoint_velocity created
[INFO] [1699754012.239922882] [mavros]: Plugin setpoint_velocity initialized
[INFO] [1699754012.245723285] [mavros]: Plugin sys_status created
[INFO] [1699754012.245769502] [mavros]: Plugin sys_status initialized
[INFO] [1699754012.249050627] [mavros.time]: TM: Timesync mode: MAVLINK
[INFO] [1699754012.250239011] [mavros]: Plugin sys_time created
[INFO] [1699754012.250266889] [mavros]: Plugin sys_time initialized
[INFO] [1699754012.255441014] [mavros]: Plugin waypoint created
[INFO] [1699754012.255493840] [mavros]: Plugin waypoint initialized
[INFO] [1699754012.259139801] [mavros]: Plugin wind_estimation created
[INFO] [1699754012.259169366] [mavros]: Plugin wind_estimation initialized
[INFO] [1699754012.259698157] [mavros]: Built-in SIMD instructions: SSE, SSE2
[INFO] [1699754012.259714222] [mavros]: Built-in MAVLink package version: 2022.12.30
[... cut ...]
[WARN] [1699754014.249297463] [mavros.sys]: VER: broadcast request timeout, retries left 4
[WARN] [1699754015.247098230] [mavros.sys]: VER: broadcast request timeout, retries left 3
[WARN] [1699754016.248869506] [mavros.sys]: VER: unicast request timeout, retries left 2
[WARN] [1699754017.246886428] [mavros.sys]: VER: unicast request timeout, retries left 1
[WARN] [1699754018.247447305] [mavros.sys]: VER: unicast request timeout, retries left 0
[WARN] [1699754019.247368172] [mavros.sys]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities

```