mavlink/mavros #1583: ROS Autopilot messages to Groundstation

natetoombs, 2021-06-03:
### Issue details

I am working on integrating a non-mavlink autopilot (ROSflight) with QGroundControl. The autopilot uses ROS messages, so I would need to convert those to mavlink messages to send to QGC. Can mavros generate and broadcast mavlink messages to a ground station?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4
[X] Custom

vooon, 2021-06-03:
Short answer - No. Mavros is designed to act like a GCS but to provide API to control the drone.
All it's plugins works like a clients, while you need to implement servers.

But you can use libmavconn as not to reimplement all stub code.

vooon, 2021-06-03:
In ROS2 it's actually possible to make "gcs-plugin" as a side effect of router introduction though.

natetoombs, 2021-06-03:
Thanks for the quick response! I'm not familiar enough with ROS2, and I probably won't be switching over anytime soon. I'll look into libmavconn; if you have any other tips, feel free to share!

closed 2021-06-21 by natetoombs