mavlink/mavros #1584: ROS2 build farm failures

rob-clarke, 2021-06-03:
### Issue details

ROS2 builds of `mavros_msgs`  are failing on the build farm. I suspect it's due to a missing `rcl_interfaces` dependency in `mavros_msgs/package.xml`.

Both [amd64 build](https://build.ros2.org/view/Fbin_uF64/job/Fbin_uF64__mavros_msgs__ubuntu_focal_amd64__binary/7/console) and [arm64 build](https://build.ros2.org/job/Fbin_ubv8_uFv8__mavros_msgs__ubuntu_focal_arm64__binary/7/console) show the same error from CMake:

```
07:05:37   CMake Error at CMakeLists.txt:17 (find_package):
07:05:37   By not providing "Findrcl_interfaces.cmake" in CMAKE_MODULE_PATH this
07:05:37   project has asked CMake to find a package configuration file provided by
07:05:37   "rcl_interfaces", but CMake did not find one.
07:05:37 
07:05:37   Could not find a package configuration file provided by "rcl_interfaces"
07:05:37   with any of the following names:
07:05:37 
07:05:37     rcl_interfacesConfig.cmake
07:05:37     rcl_interfaces-config.cmake
07:05:37 
07:05:37   Add the installation prefix of "rcl_interfaces" to CMAKE_PREFIX_PATH or set
07:05:37   "rcl_interfaces_DIR" to a directory containing one of the above files.  If
07:05:37   "rcl_interfaces" provides a separate development package or SDK, be sure it
07:05:37   has been installed.
07:05:37 
07:05:37 
07:05:37 -- Configuring incomplete, errors occurred!
```
Building locally worked fine for me so it must be something to do with the build farm setup.

### MAVROS version and platform

Mavros: 2.0.0
ROS: Foxy

rob-clarke, 2021-06-08:
Fixed for both [amd64](https://build.ros2.org/view/Fbin_uF64/job/Fbin_uF64__mavros_msgs__ubuntu_focal_amd64__binary/12/) and [arm64](https://build.ros2.org/job/Fbin_ubv8_uFv8__mavros_msgs__ubuntu_focal_arm64__binary/12/).

Packages available on the testing repo. [See instructions](https://docs.ros.org/en/foxy/Installation/Prerelease-Testing.html)

closed 2021-06-08 by rob-clarke

vooon, 2021-06-09:
Great! Thanks @rob-clarke !