mavlink/mavros #1586: Coordinate transformation using quaternions incorrect

cgnarendiran, 2021-06-09:
### Issue details
Hi, I'm trying to command an attitude in the PX4 SITL (for a standard fixedwing plane) by setting `setpoint_raw/attitude`. I'm sending these commands in the ENU frame, hoping that mavros will take care of the conversion to the NED frame when sending it to the FCU. However, I couldn't achieve the desired behavior. I took a look under the hood and saw that the frame conversions using quaternions are wrong:
https://github.com/mavlink/mavros/blob/c13ceed94ce42f6c4ed961885d2b61466b5d1459/mavros/src/lib/ftf_frame_conversions.cpp#L78-L96

Here, in line numbers 86,
https://github.com/mavlink/mavros/blob/c13ceed94ce42f6c4ed961885d2b61466b5d1459/mavros/src/lib/ftf_frame_conversions.cpp#L86
the equation should have been:
```
return NED_ENU_Q * q * NED_ENU.inverse();
```
Same thing goes for line numbers 90 and 94. 
Essentially when you're trying to rotate a given orientation in quaternion format, say `q_1` by another quaternion `q_2`, it needs to be done like this:
```
q_1_new  =    q_2 * q_1 * q_2^{-1}
```
Reference for the above calculation is provided in this [stackoverflow answer](https://math.stackexchange.com/questions/396617/rotate-the-axis-of-rotation-of-a-quaternion-by-another-quaternion).

For further references:
https://www.weizmann.ac.il/sci-tea/benari/sites/sci-tea.benari/files/uploads/softwareAndLearningMaterials/quaternion-tutorial-2-0-1.pdf

### MAVROS version and platform
Mavros: 2.0.0
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version
PX4
Version: 1.11.3 (Stable release)

vooon, 2021-06-09:
@TSC21 could you please have a look?

TSC21, 2021-06-09:
@cgnarendiran if you are sending attitude commands, weren't you supposed to be sending these in the body frame instead?

cgnarendiran, 2021-06-11:
Hi, I'm sorry, the transformations using quaternions are correct. My bad.

@TSC21, I'm sending the desired orientation of the aircraft wrt to the inertial frame; so FLU wrt ENU.

However, it is highly confusing to see `aircraft` to `base_link` transformation. It's just that the aircraft frame is different in the ENU and NED frames. Maybe consider changing names to `aircraft_enu` and `aircraft_ned`? I've made the following diagram to make things clear:
[image]

TSC21, 2021-06-11:
There's not aircraft enu or aircraft ned. There's a body frame. And body frame orientation in ROS is FLU, matching `base_link`, while body frame orientation in the MAVLink convention is FRD, which matches the `airframe` frame.

cgnarendiran, 2021-06-11:
I see, that clears things up. Thanks for the explanation. I'll close this issue.

closed 2021-06-11 by cgnarendiran