mavlink/mavros #1590: MAVROS-APM2.6 : mavsafety arming issue

elonv, 2021-07-06:
@vooon  @TSC21 Hello.I have been receiving error "REQUEST FAILED CHECK MAVROS LOGS" while running command "rosrun mavros mavsafety arm".
I am using apm 2.6 as flight control unit and usb cable (serial 0) to connect raspberry pi with apm 2.6(dev/ttyACM0).
The roslaunch command (roslaunch mavros apm.launch) works fine and shows everything fine in diagnostics, service for setting the STABILIZE (rosrun mavros mavsys mode -c STABILIZE) mode also works well but only STABILIZE mode could be set using other modes such as LAND shows timed out error.
I have tried every possible solution : trying it on different single board computers raspberry pi, UPBOARD, trying different versions of mavros by building it from source.
Flashing apm2.6 firmware with different baud rates.And what i observed from experimenting with apm flashing is that changes that are applied to apm before flashing does not get imposed on mavros.This problem is related to mavros it has nothing to do anything with what values and prearm checks are enabled at APM firmware.
I am able to arm the apm2.6 with TX RX but with mavros arming it shows error and does not arm.
I think there might be some problem with tgt_system and tgt_component values which by default are set to 1.
With these values only mavsys -c STABILIZE works and when i try different values of above two params then mavsys stabilize also begins to show error of request failed.
Its been a long time working over this problem but could not find any solution.

### MAVROS version and platform

Mavros: 1.6.0
ROS: Kinetic
Ubuntu: ?16.04?

### Autopilot type and version

ArduPilot - apm2.6

Version: ?3.7.1?

### Node logs

started roslaunch server http://pi:36731/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_id: 4
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/horiz_accuracy: 0.5
 * /mavros/fake_gps/mocap_transform: False
 * /mavros/fake_gps/mocap_withcovariance: False
 * /mavros/fake_gps/satellites_visible: 6
 * /mavros/fake_gps/speed_accuracy: 0.0
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_hil_gps: True
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
[... 663 lines cut ...]
        value: "0.000000"
      - 
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "0.000000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"

### Check ID

```
rosrun mavros checkid

OK. I got messages from 1:1.

---
Received 452 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 35, 36, 165, 33, 42, 24, 163, 74, 30, 150, 152, 27, 29, 62

vooon, 2021-07-07:
From logs it looks like firmware do not accept incoming messages. Since apm2.6 quite old, i suppose that it do not support Mavlink 2.0, so try to set `fcu_protocol: v1.0`.

elonv, 2021-07-07:
@vooon i knew apm 2.6 works on v1 mavlink protocol..so immediately after installing mavros i looked for connection protocol and changed it to apm 2.6 supported one i.e v1..but did'nt worked.

vooon, 2021-07-07:
@elonv you probably done it wrong:
```
/mavros/fcu_protocol: v2.0
```

elonv, 2021-07-08:
@vooon This is my launch file:

<launch>
        
        

        <arg name="fcu_url" default="/dev/ttyACM0:57600" />
        <arg name="gcs_url" default="" />
        <arg name="tgt_system" default="1" />
        <arg name="tgt_component" default="1" />
        <arg name="log_output" default="screen" />
        <arg name="fcu_protocol" default="v1.0" />
        <arg name="respawn_mavros" default="false" />

        <include file="$(find mavros)/launch/node.launch">
                <arg name="pluginlists_yaml" value="$(find mavros)/launch/apm_pluginlists.yaml" />
                <arg name="config_yaml" value="$(find mavros)/launch/apm_config.yaml" />

                <arg name="fcu_url" value="$(arg fcu_url)" />
                <arg name="gcs_url" value="$(arg gcs_url)" />
                <arg name="tgt_system" value="$(arg tgt_system)" />
                <arg name="tgt_component" value="$(arg tgt_component)" />
                <arg name="log_output" value="$(arg log_output)" />
                <arg name="fcu_protocol" value="$(arg fcu_protocol)" />
                <arg name="respawn_mavros" default="$(arg respawn_mavros)" />
        </include>
</launch>

And this is what i get after roslaunch:

PARAMETERS

/mavros/cmd/use_comp_id_system_control: False
/mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
/mavros/conn/heartbeat_rate: 1.0
/mavros/conn/system_time_rate: 1.0
/mavros/conn/timeout: 10.0
/mavros/conn/timesync_rate: 10.0
/mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
/mavros/distance_sensor/rangefinder_pub/frame_id: lidar
/mavros/distance_sensor/rangefinder_pub/id: 0
/mavros/distance_sensor/rangefinder_pub/send_tf: False
/mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
/mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
/mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
/mavros/distance_sensor/rangefinder_sub/id: 1
/mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
/mavros/distance_sensor/rangefinder_sub/subscriber: True
/mavros/fake_gps/eph: 2.0
/mavros/fake_gps/epv: 2.0
/mavros/fake_gps/fix_type: 3
/mavros/fake_gps/geo_origin/alt: 408.0
/mavros/fake_gps/geo_origin/lat: 47.3667
/mavros/fake_gps/geo_origin/lon: 8.55
/mavros/fake_gps/gps_id: 4
/mavros/fake_gps/gps_rate: 5.0
/mavros/fake_gps/horiz_accuracy: 0.5
/mavros/fake_gps/mocap_transform: False
/mavros/fake_gps/mocap_withcovariance: False
/mavros/fake_gps/satellites_visible: 6
/mavros/fake_gps/speed_accuracy: 0.0
/mavros/fake_gps/tf/child_frame_id: fix
/mavros/fake_gps/tf/frame_id: map
/mavros/fake_gps/tf/listen: False
/mavros/fake_gps/tf/rate_limit: 10.0
/mavros/fake_gps/tf/send: False
/mavros/fake_gps/use_hil_gps: True
/mavros/fake_gps/use_mocap: True
/mavros/fake_gps/use_vision: False
/mavros/fake_gps/vert_accuracy: 0.5
/mavros/fcu_protocol: v1.0

i could not see anything wrong Roslaunch shows it the way told /mavros/fcu_protocol: v1.0. (v1 as apm 2.6 works on mavlink protocol version 1).

vooon, 2021-07-08:
@elonv , hmm, yes, just your previous log has v2.0. But you still can't do anything?
Most obvious problem if message CRC's do not match on both edges, you'll get no commands, parameters and mission.
But i do not remember any significant changes for common.xml.

elonv, 2021-07-08:
@vooon no still cant do anything.

But i can change mode by command mavros mavsys mode -c STABILIZE..And only stabilize mode works, other modes shows timed out..And with arming command request failed...i am getting data of imu from fcu by running rostopic echo.. but this arming thing does not seems to work...

Where could i find this common xml...and what it has to do with CRC...?

How this arming problem could be solved?..its been seven days and i am struck at this arming issue...I am able to arm the quadcopter with Tnamsmitter and receiver...even able to arm it via apm planner 2 gui via arm/disarm event...But with mavros unable to arm it..

vooon, 2021-07-08:
https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml , message CRC is an mavlink implementation detail how it prevent unknown/changed messages to come.
What mavros prints to it's console, when you're trying to call arming, mode -c AUTO?

elonv, 2021-07-08:
@vooon is it right to say CRC message is just like parameters on apm planner2 or mission planner which involves pre flight check etc etc..????

1] when i run ROSRUN MAVROS MAVSAFETY ARM - it shows request failed. check log messages.In log messages only error i found is /USE_SIM_TIME is not set.

2] And when i run ROSRUN MAVROS MAVSYS MODE -C AUTO - it shows timed out.

3] But when i run ROSRUN MAVROS MAVSYS MODE -C STABILIZE - This command works and it shows mode changed, But even with mode there is a problem which is, I am only able to set mode to stabilize....EVERY MODE EXCEPT STABILIZE SHOWS TIMED OUT...

elonv, 2021-07-08:
@vooon in https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml there is :

<entry value="268435456" name="MAV_SYS_STATUS_PREARM_CHECK">
        <description>0x10000000 pre-arm check status. Always healthy when armed</description>
      </entry>

May be changing entry value to "0" .. might help..because this could disable prearm check...am i right?...may be this could stopping it to arm..or is there any other way to disable to prearm check on mavros package...because on apm planner2 side i have disabled pre arm check.

elonv, 2021-07-08:
@ vooon log output for failing arm command:

[rospy.client][INFO] 2021-07-06 18:06:36,781: init_node, name[/mavsafety_6570_1625574996706], pid[6570]
[xmlrpc][INFO] 2021-07-06 18:06:36,781: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2021-07-06 18:06:36,781: Started XML-RPC server [http://vansh:46601/]
[rospy.init][INFO] 2021-07-06 18:06:36,782: ROS Slave URI: [http://vansh:46601/]
[rospy.impl.masterslave][INFO] 2021-07-06 18:06:36,782: _ready: http://vansh:46601/
[rospy.registration][INFO] 2021-07-06 18:06:36,783: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2021-07-06 18:06:36,783: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2021-07-06 18:06:36,882: registered with master
[rospy.rosout][INFO] 2021-07-06 18:06:36,883: initializing /rosout core topic
[rospy.rosout][INFO] 2021-07-06 18:06:36,892: connected to core topic /rosout
[rospy.simtime][INFO] 2021-07-06 18:06:36,896: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rospy.internal][INFO] 2021-07-06 18:06:37,098: topic[/rosout] adding connection to [/rosout], count 0
[rospy.core][INFO] 2021-07-06 18:06:41,907: signal_shutdown [atexit]
[rospy.internal][INFO] 2021-07-06 18:06:41,914: topic[/rosout] removing connection to /rosout
[rospy.impl.masterslave][INFO] 2021-07-06 18:06:41,914: atexit

vooon, 2021-07-08:
That CRC is a part of communication protocol, it's defined by message definition. Not some kind of parameter, just older version of messages could have some changes, resulting in crc changes. I think probably it's worth to try last available version for Indigo/Jade (you would need to follow source install and **use old version of mavlink**).

That is log is for roscore, i'm asking about mavros log, by default it's being printed to the console where you run roslaunch.
E.g.:
```
[ WARN] [1625588489.283214897]: CMD: Command 410 -- wait ack timeout
[ INFO] [1625588494.278008685]: HP: requesting home position
[ WARN] [1625588499.284022423]: CMD: Command 410 -- wait ack timeout
```

elonv, 2021-07-08:
@vooon i mentioned while starting the issue that i have tried different versions of both mavros and mavlink by installing them from source (versions from 2016-17-18-19) for ros kinetic.But what here you are suggesting is to try mavlink version for indigo..right?

And when i run arming service , the console output shows:
CMD: Command 400 -- wait ack timeout

vooon, 2021-07-09:
Yes, around 2015 i had have a plane with apm 2.6, and mavros worked good. So i suppose that might better to try old version of the mavlink first.
Or you might consider to upgrade to something newer, on 32-bit ARM. Because as i know Arducopter no longer support that old boards.

vooon, 2021-07-09:
https://github.com/mavlink/mavros/blob/f1dad6c21b767ef2d3e1bff57952defc318c49bf/mavros/launch/apm_config.yaml#L38

Try to set it `true`. Maybe that is the problem for COMMAND_LONG.

elonv, 2021-07-09:
@vooon setting use_comp_id_system_control: to TRUE did not worked . Still getting request failed. And on console command 400 wait ack time out.

I will try to download versions from 2015.

Consider to upgrade to something newer ..are you suggesting me to buy upgraded version of flight controller?..I will do that if nothing seems to work.But first i will try older versions.

elonv, 2021-07-09:
@vooon i installed mavros version 0.18.5-0 and mavlink version 2016.12.12-0 these versions i did not tried earlier...and it worked. These version were launched around november and december and i remember working on mavros jan 2017 and at that time they worked ..so i thought of giving try to these versions..and glad they worked.

But one thing is there that apm 2.6 only arms for few seconds...The solid red light which signifies it is armed starts blinking after few seconds of solid red light of arming.

vooon, 2021-07-09:
Maybe some preflight check didn't pass? Will you get the same if arm from receiver?

elonv, 2021-07-09:
Few days ago it happened with receiver also but then i did radio calibration and esc calibration then it got hold over arming.May be it is happening because of firmware as after that i did not tried it arming with receiver and installed firmware many time but did not go for initial setup.

Is there a way to bypass preflight check in mavros?

vooon, 2021-07-09:
It's on firmware side. Check parameters, some of them could disable that checks (but i do not remember which exactly).

elonv, 2021-07-09:
I think i have to reflash the firmware and restart initial setup.

@vooon I am trying to manually control it via logitech 710 ..without involving rc override ..is it possible..?
From hector quadrotor i have made some changes into joy teleop node and able to publish on /mavros/setpoint_velocity/cmd_vel..
do you think that would help me in controlling quadcopter manually without involving rc at all?

julienrineau, 2022-05-25:
Hi, I'm having the same problem. 

Apparently the problem is solved by installing mavros version **0.18.5-0** and mavlink version **2016.12.12-0**. I found on #941 that it can be done by editing the mavros.rosinstall file after running `rosinstall_generator --rosdistro kinetic mavlink | tee /tmp/mavros.rosinstall` and `rosinstall_generator --upstream mavros | tee -a /tmp/mavros.rosinstall`.

However, I do not find the how to edit the mavros.rosinstall file. I did not even find its location (nor that of the /tmp folder).

Please can you explain @elonv or @vooon how you've done it ?

closed 2026-11-01 by github-actions[bot]