mavlink/mavros #1591: Offboard unable to set Body_NED while using setpoint_raw/local

JethroPhuah, 2021-07-09:
Hi, 

I have been trying to command my drone's vx, position z and yaw angle in stimulator environment. I managed to get it running however, when i changed my target_frame to BODY_NED, I get this warning. 
WARN  [mavlink] SET_POSITION_TARGET_LOCAL_NED invalid
Below is my simplified code for easier reference. 

raw.coordinate_frame = 1;
raw.type_mask = 3043; (Ignore VX, Yaw and PZ)
raw.header.stamp = ros::Time::now();
raw.yaw = 3.14;
raw.velocity.x = 10;
raw.position.z = 30;
ROS_INFO("Testing");
local_raw_pub.publish(raw);

Ultimately, I would like the drone to move forward and backward base on the direction of the drone instead of its local position. Thank you so much for any help in advance!

vooon, 2021-07-09:
As i know, PX4 supports only some small set up type_mask combinations. Better check autopilot docs / code.

JethroPhuah, 2021-07-09:
Does that mean only a certain combination of the type_mask can only run with BODY_NED and the rest can only run with LOCAL_NED?

vooon, 2021-07-09:
https://github.com/PX4/PX4-Autopilot/blob/223ca11aed7a886889123afa4dc4a6593a330dc3/src/modules/mavlink/mavlink_receiver.cpp#L875

Gergiouszsh, 2023-03-13:
I also met the same problem. Have you solved it？

Gergiouszsh, 2023-03-13:
> > raw.type_mask = 3043; (Ignore VX, Yaw and PZ)
> 
> This is not a supported type mask
What should be the correct one? I also encountered the same problem

Jaeyoung-Lim, 2023-03-13:
@Gergiouszsh You can find what combinations of setpoints are supported for offboard mode for different vehicle types in https://docs.px4.io/main/en/flight_modes/offboard.html

levelmoon, 2023-03-17:
I find that "[MAV_FRAME_BODY_NED](https://mavlink.io/en/messages/common.html#MAV_FRAME_BODY_NED)
DEPRECATED: Replaced by [MAV_FRAME_BODY_FRD](https://mavlink.io/en/messages/common.html#MAV_FRAME_BODY_FRD) (2019-08)."
Maybe it is the answer.
And i want to change my LocalNED to BodyNED,but i did't find any mavros messages can help me with that.
If someone can help me,i will really appreciate you!

closed 2026-11-01 by github-actions[bot]