mavlink/mavros #1592: FCU not connected, given that TX/RX seems to be functioning

hanzheteng, 2021-07-17:
### Issue details

The mavros launch file got stuck at the line of  `MAVROS started. MY ID 1.240, TARGET ID 1.1`.
In the diagnostic message, it says that FCU is not connected.

Efforts so far: 
- Originally I used [the ready-to-go product from mRo](https://store.mrobotics.io/USB-FTDI-Serial-to-JST-GH-p/mro-ftdi-jstgh01-mr.htm) and encountered this problem. Then I switched to another FTDI chip with TX/RX indicators integrated for debugging. I soldered the cables/pins according to the Pixhawk pinout manual. I believe the cable is working (though the soldered one can have large noises) because 
  - The RX indicator is flickering when I connect the FTDI cable on Pixhawk (regardless of the USB end connected or not). 
  - The TX indicator is flickering when the USB end is connected to onboard computer AND mavros is running (regardless of the 6-pin JST end connected or not).
  - When cable is properly connected on both end (and even mavros is not running), I can see symbols/streaming by running `cat /dev/ttyPixhawk` and RX indicator is flickering. The Python serial tool `miniterm.py` can also output symbols/streaming.
- In PX4 firmware, MAV_1_CONFIG parameter has been set to `TELEM2`; other parameters such as mode and baudrate are also set properly. (If MAV_1_CONFIG is disabled, the RX indicator will not flash any more.)
- In the launch file, `fcu_url` is set to `/dev/ttyPixhawk:921600`, and `tgt_system` and `tgt_component` do match the `MAV_SYS_ID` and `MAV_COMP_ID` parameters in PX4 firmware.

Potential influence:
- I have plugged FTDI cable into the TELEM2 port with 5V TTL a few times, before I realized it should be 3.3V and changed the settings on both FTDI chips. I don't expect this will damage the board though. 
- The aircraft had crashed into a tree some time ago (at a very close distance), but no parts were damaged.

At this point I am still not entirely sure if this is a hardware issue (e.g., damaged FCU), or a software issue (e.g., incompatible/mismatched version). **It seems to me that the computer and FCU are indeed exchanging data, bi-directionally, but somehow they don't recognize each other**. (The baud rate is also matched, BTW). Went through existing issues but didn't get any more inspiration. Any hints or testing suggestions are appreciated. 

### MAVROS version and platform

Mavros: 1.8.0  (May 2021)
ROS: Melodic 
Ubuntu: 18.04
Onboard computer: Jetson Xavier NX
Autopilot: Pixhawk 4 
Connection: Onboard computer and Pixhawk is directly connected via a FTDI chip (USB-JST)

### Autopilot type and version

[X] PX4

Version: 1.11.3  (Jan 2021)

### Node logs

```

... logging to /home/arcs/.ros/log/e06f65a2-e69c-11eb-868f-204ef6c5ef8f/roslaunch-xavier-nx-9241.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://xavier-nx:45997/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
[... 516 lines cut ...]
<launch>
    
    

    <arg name="fcu_url" default="/dev/ttyPixhawk:921600" />
    <arg name="gcs_url" default="" />
    <arg name="tgt_system" default="1" />
    <arg name="tgt_component" default="1" />
    <arg name="log_output" default="screen" />
    <arg name="fcu_protocol" default="v2.0" />
    <arg name="respawn_mavros" default="false" />

    <include file="$(find mavros)/launch/node.launch">
        <arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
        <arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />
        <arg name="fcu_url" value="$(arg fcu_url)" />
        <arg name="gcs_url" value="$(arg gcs_url)" />
        <arg name="tgt_system" value="$(arg tgt_system)" />
        <arg name="tgt_component" value="$(arg tgt_component)" />
        <arg name="log_output" value="$(arg log_output)" />
        <arg name="fcu_protocol" value="$(arg fcu_protocol)" />
        <arg name="respawn_mavros" default="$(arg respawn_mavros)" />
    </include>

</launch>

vooon, 2021-07-19:
From diagnostics i see that something coming in at relatively high speed, but only 2 messages ok.
From checkid i see that it is having some complete garbage messages (checkid do not verify framing status, which likely indicating CRC errors).

From that facts i can suspect that you're having problem with hardware link. Please verify that you properly connected Rx-Tx and Grounds, and that you do not have ground loops, which can cause a lot of noise.
If you could, i suggest to use some simple oscilloscope to check signal conditions. Most of the pins on STM32 are 5V tolerant, so probably you hadn't damaged it. Anyway you can try to use TELEM1.

hanzheteng, 2021-07-22:
Thank you for the constructive feedback. It ends up with the case that both cables I used are problematic. 

The mRo USB FTDI Serial to JST-GH adapter is not a ready-to-go product (though recommended by PX4 docs in [this section](https://docs.px4.io/master/en/peripherals/companion_computer_peripherals.html#ftdi-devices)). I have to swap the TX and RX cables (cut them off and solder again). As for the second adapter I used, the chip is not a proper one and the noise level is too high (soldering was bad).

After fixing the hardware link, I made it work on the TELEM2 port. Out of curiosity, I also tested it with the FTDI chip in 5V mode, and it worked as well (the red power pin is not connected anyway). As you pointed out that most pins are 5V tolerant, I didn't really damage the port. Also found it in [this section](https://docs.px4.io/master/en/companion_computer/pixhawk_companion.html#hardware-setup) in PX4 docs, 

> All Pixhawk serial ports operate at 3.3V and are 5V level compatible.

Thanks again for the help. I am closing this issue.

closed 2021-07-22 by hanzheteng