mavlink/mavros #1593: mavROS unable to talk to BlueROV2's Pixhawk

Firestarss, 2021-07-19:
----

### Issue details

So I just recently downloaded mavROS in order to be able to use ROS to interface with my [BlueROV2](https://bluerobotics.com/store/rov/bluerov2/). I got the mavROS node setup and running but every time I try to issue a command (such as `rosservice call /mavros/cmd/arming "value: true"`) I get the following errors:
```
success: False
result: 4
```
This is returned for just about every single command I run.

### MAVROS version and platform

Mavros: Unknown... let me know where to find this and I'll add it in
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

[  ] ArduPilot
[ X] PX4 (Onboard Pi running arduSub. Pi takes commands from a topside computer and sends mavlink commands to the PX4)

Version: Unknown... let me know where to find this and I'll add it in

### Node logs

```
bluerov@bluerov2:~$ roslaunch mavros px4.launch fcu_url:=udp://:14555@192.168.2.1:14555
... logging to /home/bluerov/.ros/log/0ec3e51a-e8da-11eb-a6c9-5f9ad241a2cf/roslaunch-bluerov2-13889.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://bluerov2:35281/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
[... 402 lines cut ...]
        value: "1626735553.488679886"
      - 
        key: "Estimated time offset (s)"
        value: "9223372036.854703903"
```

### Check ID

```
ERROR. I got 1 addresses, but not your target 1:1

---
Received 330 messages, from 1 addresses
sys:comp   list of messages
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/timer.py", line 234, in run
    self._callback(TimerEvent(last_expected, last_real, current_expected, current_real, last_duration))
  File "/opt/ros/noetic/lib/mavros/checkid", line 82, in timer_cb
    for address, messages in self.message_sources.iteritems():
AttributeError: 'dict' object has no attribute 'iteritems'

```

Firestarss, 2021-07-19:
So uhh... I'm stupid and was providing the wrong port. The command should be `roslaunch mavros px4.launch fcu_url:=udp://:14550@192.168.2.1:14550` apparently. I can now issue commands. Apologies for potentially wasting anyone's time

vooon, 2021-07-20:
NOTE: ArduSub is a variation of ArduPilot (ArduRover), not PX4.

closed 2021-07-20 by vooon