mavlink/mavros #1597: libmavconn: thread race in callback API

asherikov, 2021-07-26:
### Issue details

Callback API of `MAVConnInterface` class is not thread safe, which is pretty much obvious from the code -> https://github.com/mavlink/mavros/blob/ros2/libmavconn/include/mavconn/interface.hpp#L210, and is detected by thread sanitizers.

### MAVROS version and platform

I guess all.

### Comment

It is also impossible to work around the issue in user code -- connection is established in the constructor.

I see two potential fixes that don't break API:
- Add 'delay_connection' flag to the constructor and `connect()` method, so that users can register their callbacks before connection. This leaves some room for mistake, but IMO constructor must not connect anyway.
- Mark callback pointers as deprecated and introduce methods for setting them in a safe way with a mutex, calls to callback methods should be adjusted accordingly.

vooon, 2021-08-03:
Well, yes, that lib needs overhaul, but it's unlikely to have any significant changes in ROS1 code.

asherikov, 2021-08-03:
We (and many others) are going to stick to ROS 1 for quite a while. I might find some time to fix this, but which approach should be taken?

vooon, 2021-08-03:
I suppose that you can introduce connect method, so you could setup pointers after `open_url()`.
I'm not quite sure that you really need to have lock, since that pointers were changed once.

closed 2021-11-26 by vooon