mavlink/mavros #1598: tf_prefix issue when using mavros with multiple robots

sammar80, 2021-07-27:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

Hi , I am using multiple robots each one runs a mavros node I added a tf_prefix to all robot's frames the problem is although I changed the frames in the apm_config.yaml from 'map' >> 'robot_2/map' and 'base_link' >> 'robot_2/base_link' I still find these frames ( base_link,base_link_frd ,odom,odom_ned,map and map_ned) were published to the tf topic any idea of what else I should change or what's I'm missing , Thanks in advance

### MAVROS version and platform

Mavros: master
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[ X ] ArduPilot
[ ] PX4

Version: ArduRover V4.0.0 (0e52bafa)

### Node logs
roslaunch functions apm_multi_robot.launch
... logging to /home/optimus/.ros/log/3765039c-eeba-11eb-bb8d-d03745187cad/roslaunch-Autobot-11457.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Autobot:37433/

SUMMARY
========

CLEAR PARAMETERS
 * /robot_2/mavros/

PARAMETERS
 * /robot_2/mavros/cmd/use_comp_id_system_control: False
 * /robot_2/mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /robot_2/mavros/conn/heartbeat_rate: 1.0
 * /robot_2/mavros/conn/system_time_rate: 1.0
 * /robot_2/mavros/conn/timeout: 10.0
 * /robot_2/mavros/conn/timesync_rate: 10.0
 * /robot_2/mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /robot_2/mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /robot_2/mavros/distance_sensor/rangefinder_pub/id: 0
 * /robot_2/mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /robot_2/mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /robot_2/mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /robot_2/mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /robot_2/mavros/distance_sensor/rangefinder_sub/id: 1
 * /robot_2/mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /robot_2/mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /robot_2/mavros/fake_gps/eph: 2.0
 * /robot_2/mavros/fake_gps/epv: 2.0
 * /robot_2/mavros/fake_gps/fix_type: 3
 * /robot_2/mavros/fake_gps/geo_origin/alt: 408.0
 * /robot_2/mavros/fake_gps/geo_origin/lat: 47.3667
 * /robot_2/mavros/fake_gps/geo_origin/lon: 8.55
 * /robot_2/mavros/fake_gps/gps_id: 4
 * /robot_2/mavros/fake_gps/gps_rate: 5.0
 * /robot_2/mavros/fake_gps/horiz_accuracy: 0.5
 * /robot_2/mavros/fake_gps/mocap_transform: False
 * /robot_2/mavros/fake_gps/mocap_withcovariance: False
 * /robot_2/mavros/fake_gps/satellites_visible: 6
 * /robot_2/mavros/fake_gps/speed_accuracy: 0.0
 * /robot_2/mavros/fake_gps/tf/child_frame_id: fix
[... 446 lines cut ...]
      - 
        key: "Last RTT (ms)"
        value: "6.656577"
      - 
        key: "Mean RTT (ms)"
        value: "172.460749"
      - 
        key: "Last remote time (s)"
        value: "1345.355940000"
      - 
        key: "Estimated time offset (s)"
        value: "1627376833.152102947"
---

### Check ID

```
rosrun mavros checkid
```
OK. I got messages from 1:1.

---
Received 1166 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 65, 24, 150, 152, 27, 29, 30, 33, 34, 163, 36, 165, 42, 178, 182, 62, 193, 194, 74, 87, 111, 241, 116, 125

sammar80, 2021-07-27:
[
[frames.pdf](https://github.com/mavlink/mavros/files/6884469/frames.pdf)
](url)

vooon, 2021-08-03:
Some of static transforms are hardcoded into the node, some just requires config changes.
Please note that tf_prefix param support dropped while ago from tf lib.

sammar80, 2021-08-04:
Thank you @vooon do you have any idea where else I can change the configurable frames because I changed all the frames in  apm_config.yaml

closed 2026-10-31 by github-actions[bot]