mavlink/mavros #1599: /mavros/local_position/pose does not update drone's position

daothanh2011, 2021-07-28:
[image]
 
I have a problem with fake_gps configuration.
When I public the topic /mavros/vision_pose/pose to the PX4, it works successfully, but the topic /mavros/local_position/pose is not act like vision_pose. And the drone is not moved in QGC (it is only roted).
Could you pls help me this problem? Thank you in advance!
here is my configurations:
#px4_config.yaml
fake_gps:

use_mocap: false
mocap_transform: false
use_vision: true

#px4 setup
EKF2_AID_MASK : 24
EKF2_HGT_MODE: vision
MAV_USE_HILGPS: 1

closed 2021-08-04 by daothanh2011

Smark2019, 2022-07-25:
Hi,

I have been facing this kind of problem. Can I ask how you overcome the problem?

Btw, in my problem; during a 1 meter takeoff mission, my drone is arming but not taking off because local pose data especially z data is not accurate and although my drone is armed, z data is growing.

daothanh2011, 2022-07-26:
The depth camera makes it a little difficult to get accurate z data , so it is recommended to calibrate the camera and fly under good lighting conditions .