mavlink/mavros #1601: Topics can echo data with "ros2 topic echo" but not with "rclpy or rclcpp"

hexray-newbee, 2021-08-04:
### Issue details

`ros2 topic echo /mavros/mavros/raw/fix` will echo NavSatFix.
But, the following sample code cannot sub data.
```python
# refer: https://docs.ros.org/en/foxy/Tutorials/Writing-A-Simple-Py-Publisher-And-Subscriber.html#write-the-subscriber-node
import rclpy
from rclpy.node import Node

from sensor_msgs.msg import NavSatFix

class MinimalSubscriber(Node):

    def __init__(self):
        super().__init__('minimal_subscriber')
        self.subscription = self.create_subscription(
            NavSatFix,
            '/mavros/mavros/raw/fix',
            self.listener_callback,
            10)
        self.subscription  # prevent unused variable warning

    def listener_callback(self, msg):
        self.get_logger().info('I heard: "%f"' % msg.altitude)

def main(args=None):
    rclpy.init(args=args)

    minimal_subscriber = MinimalSubscriber()

    rclpy.spin(minimal_subscriber)

    # Destroy the node explicitly
    # (optional - otherwise it will be done automatically
    # when the garbage collector destroys the node object)
    minimal_subscriber.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

### MAVROS version and platform

Mavros: 2.0.3
ROS: Foxy
Ubuntu: 20.04.2 (docker pull ros:foxy)

### Autopilot type and version

[x] ArduPilot
[  ] PX4

Version: 4.1.0-beta3

### Node logs

```
[INFO] [launch]: All log files can be found below /root/.ros/log/2021-08-04-07-35-09-268724-0ed951b0de46-19250
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mavros_node-1]: process started with pid [19252]
[mavros_node-1] [INFO] [1628062509.325559163] [mavros]: Starting mavros_node container
[mavros_node-1] [INFO] [1628062509.325624314] [mavros]: FCU URL: udp://:12345@
[mavros_node-1] [INFO] [1628062509.325634029] [mavros]: GCS URL: 
[mavros_node-1] [INFO] [1628062509.325640487] [mavros]: UAS Prefix: /uas1
[mavros_node-1] [INFO] [1628062509.325646294] [mavros]: Starting mavros router node
[mavros_node-1] [WARN] [1628062509.325670581] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[mavros_node-1] [INFO] [1628062509.328972878] [mavros]: Built-in SIMD instructions: SSE, SSE2
[mavros_node-1] [INFO] [1628062509.329007862] [mavros]: Built-in MAVLink package version: 2021.6.6
[mavros_node-1] [INFO] [1628062509.329016146] [mavros]: Known MAVLink dialects: common ardupilotmega ASLUAV all development icarous matrixpilot paparazzi standard uAvionix ualberta
[... 285 lines cut ...]
- level: "\0"
  name: 'mavros: Heartbeat'
  message: Normal
  hardware_id: uas:///uas1
  values:
  - key: Heartbeats since startup
    value: '1732'
  - key: Frequency (Hz)
    value: '0.999999'
  - key: Vehicle type
    value: Quadrotor
  - key: Autopilot type
    value: ArduPilot
  - key: Mode
    value: GUIDED
  - key: System status
    value: STANDBY
---
```

### Check ID

```
No executable found
```

vooon, 2021-08-05:
Do you sure of that double `/mavros/mavros`? Could you please list nodes?
Btw sometimes i sees problems with topics/services, but don't know why they are happening.
Maybe it's on DDS layer, maybe i done something wrong in C++ part.

dayjaby, 2021-08-19:
I have also the problem of this `/mavros/mavros` topics being created:

```
mavros/actuator_control
/mavros/altitude
/mavros/battery
/mavros/estimator_status
/mavros/extended_state
/mavros/global_position/global
/mavros/home_position/home
/mavros/landing_target/raw
/mavros/local_position/pose
/mavros/mavros/accel
/mavros/mavros/attitude
/mavros/mavros/cmd_vel
/mavros/mavros/cmd_vel_unstamped
/mavros/mavros/compass_hdg
/mavros/mavros/control
/mavros/mavros/data
/mavros/mavros/data_raw
/mavros/mavros/desired
/mavros/mavros/diff_pressure
/mavros/mavros/event
/mavros/mavros/fences
/mavros/mavros/global
/mavros/mavros/global_to_local
...
```

ros2 echo works:
```
$ ros2 topic echo /mavros/state
header:
  stamp:
    sec: 1629378205
    nanosec: 686848266
  frame_id: ''
connected: true
armed: false
guided: true
manual_input: false
mode: AUTO.LOITER
system_status: 3
---
```

even for the `/mavros/mavros` topics:

```
ros2 topic echo /mavros/mavros/raw/fix
header:
  stamp:
    sec: 1629378235
    nanosec: 371334293
  frame_id: base_link
status:
  status: 0
  service: 1
latitude: 47.3977419
longitude: 8.545594
altitude: 535.0689155639569
position_covariance:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
position_covariance_type: 2
---
```

Node list looks like
```
$ ros2 node list
WARNING: Be aware that are nodes in the graph that share an exact name, this can have unintended side effects.
/camera_plugin
/gazebo
/mavros
/mavros
/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
/mavros/mavros
```

vooon, 2021-08-21:
@dayjaby that's looks like there some trouble with sub-nodes of plugins because usual path would look like `/<uas-node-name>/<plugin-node-name>/<some-topic>`, e.g. `/mavros/global_position/global` -> UAS's name is `mavros`, Plugin name is `global_position`, and published topic is global.

vooon, 2021-08-21:
@dayjaby here is example from by build:
```
root@c169f39972ff:/# ros2 node list
/mavros
/mavros/actuator_control
/mavros/adsb
/mavros/altitude
/mavros/cmd
/mavros/ftp
/mavros/geofence
/mavros/global_position
/mavros/home_position
/mavros/imu
/mavros/landing_target
/mavros/local_position
/mavros/manual_control
/mavros/mission
/mavros/param
/mavros/rallypoint
/mavros/rc
/mavros/setpoint_accel
/mavros/setpoint_attitude
/mavros/setpoint_position
/mavros/setpoint_raw
/mavros/setpoint_trajectory
/mavros/setpoint_velocity
/mavros/sys
/mavros/tdr_radio
/mavros/time
/mavros/wind
/mavros_node
/mavros_router
/transform_listener_impl_55cee143bf80
root@c169f39972ff:/# ros2 topic list -v
Published topics:
 * /diagnostics [diagnostic_msgs/msg/DiagnosticArray] 2 publishers
 * /mavros/adsb/vehicle [mavros_msgs/msg/ADSBVehicle] 1 publisher
 * /mavros/altitude [mavros_msgs/msg/Altitude] 1 publisher
 * /mavros/battery [sensor_msgs/msg/BatteryState] 1 publisher
 * /mavros/estimator_status [mavros_msgs/msg/EstimatorStatus] 1 publisher
 * /mavros/extended_state [mavros_msgs/msg/ExtendedState] 1 publisher
 * /mavros/geofence/fences [mavros_msgs/msg/WaypointList] 1 publisher
 * /mavros/global_position/compass_hdg [std_msgs/msg/Float64] 1 publisher
 * /mavros/global_position/global [sensor_msgs/msg/NavSatFix] 1 publisher
 * /mavros/global_position/gp_lp_offset [geometry_msgs/msg/PoseStamped] 1 publisher
 * /mavros/global_position/gp_origin [geographic_msgs/msg/GeoPointStamped] 1 publisher
 * /mavros/global_position/local [nav_msgs/msg/Odometry] 1 publisher
 * /mavros/global_position/raw/fix [sensor_msgs/msg/NavSatFix] 1 publisher
 * /mavros/global_position/raw/gps_vel [geometry_msgs/msg/TwistStamped] 1 publisher
 * /mavros/global_position/raw/satellites [std_msgs/msg/UInt32] 1 publisher
 * /mavros/global_position/rel_alt [std_msgs/msg/Float64] 1 publisher
 * /mavros/home_position/home [mavros_msgs/msg/HomePosition] 1 publisher
 * /mavros/imu/data [sensor_msgs/msg/Imu] 1 publisher
 * /mavros/imu/data_raw [sensor_msgs/msg/Imu] 1 publisher
 * /mavros/imu/diff_pressure [sensor_msgs/msg/FluidPressure] 1 publisher
 * /mavros/imu/mag [sensor_msgs/msg/MagneticField] 1 publisher
 * /mavros/imu/static_pressure [sensor_msgs/msg/FluidPressure] 1 publisher
 * /mavros/imu/temperature_baro [sensor_msgs/msg/Temperature] 1 publisher
 * /mavros/imu/temperature_imu [sensor_msgs/msg/Temperature] 1 publisher
 * /mavros/local_position/accel [geometry_msgs/msg/AccelWithCovarianceStamped] 1 publisher
 * /mavros/local_position/odom [nav_msgs/msg/Odometry] 1 publisher
 * /mavros/local_position/pose [geometry_msgs/msg/PoseStamped] 1 publisher
 * /mavros/local_position/pose_cov [geometry_msgs/msg/PoseWithCovarianceStamped] 1 publisher
 * /mavros/local_position/velocity_body [geometry_msgs/msg/TwistStamped] 1 publisher
 * /mavros/local_position/velocity_body_cov [geometry_msgs/msg/TwistWithCovarianceStamped] 1 publisher
 * /mavros/local_position/velocity_local [geometry_msgs/msg/TwistStamped] 1 publisher
 * /mavros/lt_marker [geometry_msgs/msg/Vector3Stamped] 1 publisher
 * /mavros/manual_control/control [mavros_msgs/msg/ManualControl] 1 publisher
 * /mavros/mission/reached [mavros_msgs/msg/WaypointReached] 1 publisher
 * /mavros/mission/waypoints [mavros_msgs/msg/WaypointList] 1 publisher
[... 29 lines cut ...]
 * /mavros/home_position/set [mavros_msgs/msg/HomePosition] 1 subscriber
 * /mavros/landing_target/pose [geometry_msgs/msg/PoseStamped] 1 subscriber
 * /mavros/local_position/pose [geometry_msgs/msg/PoseStamped] 1 subscriber
 * /mavros/manual_control/send [mavros_msgs/msg/ManualControl] 1 subscriber
 * /mavros/rc/override [mavros_msgs/msg/OverrideRCIn] 1 subscriber
 * /mavros/setpoint_accel/accel [geometry_msgs/msg/Vector3Stamped] 1 subscriber
 * /mavros/setpoint_attitude/cmd_vel [geometry_msgs/msg/TwistStamped] 1 subscriber
 * /mavros/setpoint_attitude/thrust [mavros_msgs/msg/Thrust] 1 subscriber
 * /mavros/setpoint_position/global [geographic_msgs/msg/GeoPoseStamped] 1 subscriber
 * /mavros/setpoint_position/global_to_local [geographic_msgs/msg/GeoPoseStamped] 1 subscriber
 * /mavros/setpoint_position/local [geometry_msgs/msg/PoseStamped] 1 subscriber
 * /mavros/setpoint_raw/attitude [mavros_msgs/msg/AttitudeTarget] 1 subscriber
 * /mavros/setpoint_raw/global [mavros_msgs/msg/GlobalPositionTarget] 1 subscriber
 * /mavros/setpoint_raw/local [mavros_msgs/msg/PositionTarget] 1 subscriber
 * /mavros/setpoint_trajectory/local [trajectory_msgs/msg/MultiDOFJointTrajectory] 1 subscriber
 * /mavros/setpoint_velocity/cmd_vel [geometry_msgs/msg/TwistStamped] 1 subscriber
 * /mavros/setpoint_velocity/cmd_vel_unstamped [geometry_msgs/msg/Twist] 1 subscriber
 * /mavros/statustext/send [mavros_msgs/msg/StatusText] 1 subscriber
 * /parameter_events [rcl_interfaces/msg/ParameterEvent] 30 subscribers
 * /tf [tf2_msgs/msg/TFMessage] 1 subscriber
 * /tf_static [tf2_msgs/msg/TFMessage] 1 subscriber
 * /uas1/mavlink_sink [mavros_msgs/msg/Mavlink] 1 subscriber
 * /uas1/mavlink_source [mavros_msgs/msg/Mavlink] 1 subscriber

```

dayjaby, 2021-08-21:
Are you on ros2 galactic or foxy [[ I use foxy ]]

vooon, 2021-08-23:
@dayjaby galactic, but foxy should be exactly same.

esharet, 2021-10-11:
Your subscriber should work with the same QoS as it's publisher. For example, if you are trying to subscribe to the topic /mavros/global_position/raw/fix:
```
from mavros.base import SENSOR_QOS

self.global_position_sub = self.create_subscription(mavros.global_position.NavSatFix, '/mavros/global_position/raw/fix', self.global_position_cb, qos_profile=SENSOR_QOS)

def global_position_cb(self, topic):
        # self.get_logger().info("x %f: y: %f, z: %f" % (topic.latitude, topic.longitude, topic.altitude))
        pass
```

It is working for me with ros2 foxy, ubuntu 20.04.

weicong96, 2021-11-25:
Hi can i check if there is an updated solution for setting set_stream_rate and having no response on MAVROS ros2 foxy?
I am facing same issue as described in one of the replies above:

> I am running ROS2 in a Jetson Nano, connected to a Kakute F7 flight controller. I was able to start the mavros_node and echo the `/mavros/state`, but some of the topics are not publishing data. I currently need at least:
> 
> ```
> /mavros/global_position/rel_alt
> /mavros/rc/in
> /mavros/global_position/local
> ```
> 
> I also tried calling the set rate service, but I never get a response:
> 
> `ros2 service call /mavros/set_stream_rate mavros_msgs/srv/StreamRate "{stream_id: 0, message_rate: 30, on_off: true}"`

Details:
Operating System: Ubuntu 20.04
Package: ros-foxy-mavros 2.0.3
On top of that I'm also seeing "No satellites found" in my /diagnostics output.

I have the same QOS for the subscribers.

vooon, 2022-01-17:
Hi folks, @dayjaby , @weicong96 , @hexray-newbee , @aa-calvo 
Can you please try current ros2 master? I do not get silent messages anymore, but don't really know what is changed.
If all ok, i'm going to release 2.1 as i don't really want to leave broken version in repos.

Rainerino, 2022-02-20:
> Hi folks, @dayjaby , @weicong96 , @hexray-newbee , @aa-calvo Can you please try current ros2 master? I do not get silent messages anymore, but don't really know what is changed. If all ok, i'm going to release 2.1 as i don't really want to leave broken version in repos.

I just test with the current `mavros/ros2` branch, encountered the same problem. My `node list` look just like https://github.com/mavlink/mavros/issues/1601#issuecomment-901898300 (ocean of `mavros/mavros`), and I can't subscribe to any nodes but can echo them with `roscli`. (I think sometimes some of the nodes are missing for some reason). On my `rqt_graph`, most of the nodes are missing.

**MAVROS version and platform**
Mavros: 2.1.0
ROS: Foxy focal
OS: Pop-OS 20.04 5.15.15

**Autopilot type and version**
[x] ArduPilot 4.1.2
[x] PX4 v1.12.3
(both on sitl)

`mavros.launch.py`
```python

from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
       Node(
            package='mavros',
            executable='mavros_node',
            name='mavros',
            output='log',
            emulate_tty=True,
            # respawn=False,
            # respawn_delay=1,
            parameters=[
                {'fcu_url': 'udp://:14550@'}
            ]
        )
    ])
```

subscriber call: 
```python
     self.subscription = self.create_subscription(
            mavros.global_position.NavSatFix,
            '/mavros/global_position/raw/fix',
            self.listener_callback,
            qos_profile=SENSOR_QOS)
        self.subscription  # prevent unused variable warning
```

Any pointers would be greatly appreciated!

Student865, 2023-09-26:
Hello, have you sovled the problem? I'm also using mavros with ROS2 foxy. I can only subscribe the topic "/mavros/state", but other topics such as "/mavros/local_position/pose" and "/mavros/altitude" cannot be subscribed. I can use "ros2 topic echo" to get these topics, and I use "rqt_graph" and find that my node is really subscribing these topics. 
Besides, I found that if I start the node with launch file, there will be a lot of topics named "/mavros/mavros" and "/mavros/mavros/xxxxx", but if I start using "ros2 run",  the topics' name will be "mavros/xxxxx".

clydemcqueen, 2023-09-26:
Have you checked to make sure that your nodes are subscribing with the correct reliability setting? A lot of mavros topics are published "best_effort". If the reliability setting doesn't match, the messages will not be received.

+1 to upgrading to a supported version of ROS2, the ros2 tooling has been improved, which also might help.

closed 2026-10-31 by github-actions[bot]