mavlink/mavros #1602: Can't use mavsetp with ROVER (OFFBOARD mode forced, but not supported in Rover)

v-to, 2021-08-13:
Hello,

We try to use mavsetp with ArduRover.
Current implementation force enabling OFFBOARD mode:
https://github.com/mavlink/mavros/blob/29db67693a3fecadfec973a6879930645c2ee9d1/mavros/scripts/mavsetp#L29

OFFBOARD is however not supported by Rover https://ardupilot.org/rover/docs/rover-control-modes.html 
https://github.com/ArduPilot/ardupilot_wiki/blob/master/rover/source/docs/rover-control-modes.rst

In this situation mavsetp crushes while wtrying to switch to OFFBOARD.
I suggest one of the following solutions:

1. Switch back to this version https://github.com/mavlink/mavros/blob/d73b81903c3bb05eb31fcc5d5bc1da23d4abb201/mavros/scripts/mavsetp

2. Enable guided mode in case of Rover in code

All the best 

Witold

vooon, 2021-08-13:
Hi,

That script is more like an example of PX4 OFFBOARD.
You likely need to write your own controller node, so just be sure to use GUIDED mode, which is APM's equivalent.

v-to, 2021-08-15:
Hi Voon,

Great. Thanks for answer - everything is clear now.
In fact we decided to use MAV_CMD_NAV_SET_YAW_SPEED which seems to be easiest opiton in our case.

All the best!

WItold

closed 2021-08-18 by vooon