mavlink/mavros #1610: Iq Tutorials MAVROS multiple drone connection issue

saifullahsohail123, 2021-08-30:
Hi There! I have been following Intelligent Quads IQ tutorials recently. I have done almost all of it. Now I am facing problem with MAVROS while connecting multiple drones with MAVROS. It states the connection is started but the ROS code is still waiting for FCU  connections. What could be the problem, kindly guide me in this regard.

I have attachecd the main files involved.
[square.txt](https://github.com/mavlink/mavros/files/7075662/square.txt)
[multi_square_sol.txt](https://github.com/mavlink/mavros/files/7075668/multi_square_sol.txt)
[multi-apm.txt](https://github.com/mavlink/mavros/files/7075671/multi-apm.txt)
[ardupilot.txt](https://github.com/mavlink/mavros/files/7075678/ardupilot.txt)
[multi_drone.txt](https://github.com/mavlink/mavros/files/7075681/multi_drone.txt)

ROS:  Melodic  
Ubuntu: 18.04

Results

[image]
[image]
[image]

The program is still Waiting for the FCU connection despite the connection established as seen in the following screenshot:

[image]

vooon, 2021-08-30:
What do you see in state topic?

saifullahsohail123, 2021-08-31:
I do see MAVROS topics of drone1, drone2 and drone3.[
[image]
[image]
[image]
](url)

vooon, 2021-08-31:
I mean to use `rostopic echo`.

closed 2026-10-31 by github-actions[bot]