mavlink/mavros #1629: CON: Lost connection, HEARTBEAT timed out

Ayush1285, 2021-10-03:
### Issue details

I am running px4 sitl on gazebo using command 
```
roslaunch px4 mavros_posix_sitl.launch
```
and getting error **CON: Lost connection, HEARTBEAT timed out** 

### MAVROS version and platform

Mavros: 1.9.0
ROS: Noetic
Ubuntu: 20.04 (WSL 2)

### Autopilot type and version

PX4

### Node logs

```
started roslaunch server http://LAPTOP-U2IK80BP:45503/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
[... 524 lines cut ...]
        value: "29.396000000"
      -
        key: "Estimated time offset (s)"
        value: "167975404.299634695"
---
```

### Check ID

```
OK. I got messages from 1:1.

---
Received 300 messages, from 1 addresses
sys:comp   list of messages
Exception in thread Thread-6:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/timer.py", line 234, in run
    self._callback(TimerEvent(last_expected, last_real, current_expected, current_real, last_duration))
  File "/home/ayush/catkin_ws/src/mavros/mavros/scripts/checkid", line 82, in timer_cb
    for address, messages in self.message_sources.iteritems():
AttributeError: 'dict' object has no attribute 'iteritems'
```

vooon, 2021-10-03:
From node log disconnection event occurs too fast. Maybe timeout setting changed?

Ayush1285, 2021-10-04:
> From node log, disconnection event occurs too fast. Maybe timeout setting changed?

This is my CON settings. 
# sys_status & sys_time connection options
conn:
  heartbeat_rate: 1.0    # send hertbeat rate in Hertz
  timeout: 10.0          # hertbeat timeout in seconds
  timesync_rate: 10.0    # TIMESYNC rate in Hertz (feature disabled if 0.0)
  system_time_rate: 1.0  # send system time to FCU rate in Hertz (disabled if 0.0)

vooon, 2021-10-04:
Looks correct.

Ayush1285, 2021-10-04:
> Looks correct.

yeah, but still I'm getting the same error

vooon, 2021-10-04:
Currently i do not have any idea why that was happening. But i never ever tried WSL, so maybe it's worth to try setup VM with Ubuntu and setup mavros there.

gssammy, 2021-11-09:
i'm seeing this and i'm not using WSL: https://github.com/mavlink/mavros/issues/1645

boris-gu, 2021-11-17:
I started getting the same error, I am using docker.
[image]

vooon, 2021-11-20:
Just tried to reproduce on AArch64. But it works.
[image]

vooon, 2021-11-20:
Hmm. I see a pattern in timestamps:
```
[ INFO] [1633286829.192927400, 2.616000000]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1633286829.195145200, 2.616000000]: IMU: High resolution IMU detected!
[ INFO] [1633286829.196590100, 1633286829.192759000]: HP: requesting home position
[ WARN] [1633286829.196759700, 1633286829.192759000]: CON: Lost connection, HEARTBEAT timed out.
```

What is that second number is? Perhaps it somehow related to boot_time_ms? Or it is simulated time?
Anyway if timer is not based on wall time, it certainly will expire on that huge bump.

vooon, 2021-11-20:
Try to change config to have:
```yaml
time:
  publish_sim_time: false

gssammy, 2021-11-23:
setting `publish_sim_time` to `false` worked for me. Thanks!

vooon, 2021-11-24:
@gssammy great! So timers is not wall-based. Good to know.

closed 2021-11-24 by vooon

SamLin55587, 2021-12-13:
@gssammy  I have same problem ,but I don't know where I can setting publish_sim_time can you tell me?

gssammy, 2021-12-13:
@SamLin55587: I edited the `time` plugin config in the px4_config: https://github.com/mavlink/mavros/blob/16103a93444af8aa908a0218b8dc9c6f20b8cb33/mavros/launch/px4_config.yaml#L21. It should look something like below:
```
# sys_time
time:
  time_ref_source: "fcu"  # time_reference source
  timesync_mode: MAVLINK
  timesync_avg_alpha: 0.6 # timesync averaging factor
  publish_sim_time: false
  ```
It's also weird that the developers choose not to talk about this particular parameter in the plugin description here: http://wiki.ros.org/mavros#mavros.2FPlugins.sys_time

ythare, 2024-07-31:
> I started getting the same error, I use docker. [image]

hallo, i am getting the same err in docker, did you work out it?

commit deaa9438 (2021-11-24): plugin: sys: Use wall timers for connection management

Fixes #1629