mavlink/mavros #1642: Issue with ros1_bridge with mavros

weicong96, 2021-11-04:
### Issue details
I was originally trying to use mavros on Ubuntu 18.04 using ros1 melodic and ros2 eloquent. This had failed result in an error regarding static assertion failed regarding ros1 and ros2 message sizes when trying to build ros1_bridge from source. Originally i had thought was an issue with message compatibility from ros-melodic-mavros-msgs and mavros on master branch.

I went ahead to test out debian packages `ros-foxy-mavros-msgs` and `ros-noetic-mavros-extras` and trying to build them, still have the same error(details below)

Background:
I have other ros2 nodes, trying to get them to communicate to mavros on ros1 via ros1_bridge. 

### MAVROS version and platform
Previous attempted configuration that had worked
ros-melodic-mavros-msgs: 1.8.0
ros-eloquent-mavros-msgs: no such package was avaliable, used 2.0.3 from master.

ros-noetic-mavros-msgs: 1.9.0
ros-foxy-mavros-msgs: 2.0.3
ROS: foxy and noetic
Ubuntu: 20.04

### Node logs
Logs from building error
```
--- stderr: ros1_bridge
/srv/bridge/build/ros1_bridge/generated/mavros_msgs__msg__OverrideRCIn__factories.cpp: In static member function ‘static void ros1_bridge::Factory<ROS1_T, ROS2_T>::convert_1_to_2(const ROS1_T&, ROS2_T&) [with ROS1_T = mavros_msgs::OverrideRCIn_<std::allocator<void> >; ROS2_T = mavros_msgs::msg::OverrideRCIn_<std::allocator<void> >]’:
/srv/bridge/build/ros1_bridge/generated/mavros_msgs__msg__OverrideRCIn__factories.cpp:61:32: error: static assertion failed: destination array not large enough for source array
   61 |     (ros2_msg.channels.size()) >= (ros1_msg.channels.size()),
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/ros1_bridge.dir/build.make:3803: CMakeFiles/ros1_bridge.dir/generated/mavros_msgs__msg__OverrideRCIn__factories.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:260: CMakeFiles/ros1_bridge.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< ros1_bridge [2min 40s, exited with code 2]
```

vooon, 2021-11-04:
Messages package synchronized in 1.10.0 and 2.0.4. Can you please test?

weicong96, 2021-11-05:
Hi, this issue seems fixed with 1.10.0 and 2.0.4. Thanks.

closed 2021-11-05 by weicong96