mavlink/mavros #1643: LandingTarget position_valid in python

Y-H-Park, 2021-11-06:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
We are simulated LandingTarget using mavros, python in Gazebo. Other variables of LandingTarget are used, but, there is no part about the "position_valid".
Inputs the position and pose of LandingTarget.pose are good, but it is not working. I wonder if position_valid in python is not suppoted in LandingTarget.
I research Auto Drone Station. so, it is very very important for me. 
Any other solution ??
Thank you!

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[O] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```

### Diagnostics

```
```
header: 
  seq: 24962
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "map"
target_num: 0
frame: 9
angle: [0.47360944747924805, -0.17104771733283997]
distance: 5.01578426361
size: [1.0, 1.0]
pose: 
  position: 
    x: 2.57066499141
    y: -0.866404584033
    z: 5.01578449959
  orientation: 
    x: 0.0524738328068
    y: -0.00854498640532
    z: 0.985223288658
    w: 0.162814469762
type: 2
"position_valid" : where (?) 
---
```

vooon, 2021-11-06:
https://mavlink.io/en/messages/common.html#LANDING_TARGET
https://github.com/mavlink/mavros/blob/ros2/mavros_msgs/msg/LandingTarget.msg

As you can see position_valid if one of mavlink 2 extensions. Likely it does not exist at the moment of plugin development, so it does not exist in ros msg.
You should make a PR with that change.

Y-H-Park, 2021-11-06:
I wanna use LandingTarget position_valid, I know that in the mavros there is /mavlink/to and /mavlink/from. so can I use the mavlink 2.0 protocol in the mavros ?
Thank for quick answer!

vooon, 2021-11-06:
It much harder to write decoder around those topics than addition of a few lines of code to existing plugin.

Y-H-Park, 2021-11-06:
When using other plugins supported by mavros(vision_pose, setting yaw plugins etc..), are they available at the same time ?
Thank you SSSSSSOOOOOOOOOO much!!!!!!

vooon, 2021-11-06:
That's the point to have plugins - easiness of addition of custom translators. Plugin may be loaded from third party package, just need to meet some requirements.

Y-H-Park, 2021-11-06:
I have a last question... Can I use dronekit with marvos ?

vooon, 2021-11-07:
Umm, for what purpose? You can use gcs_url to provide a connection for dronekit.